zephyrproject-rtos/zephyr #88670: [RFC] PWM API: add support for enabling DMA requests

SiViSur, 2025-04-15:
## Introduction

When using PWM to generate signals (e.g. sine waves or sound), it is useful to use DMA to update the duty cycle on the fly without occupying the CPU ([example](https://community.st.com/t5/stm32-mcus/custom-signal-generation-using-pwm-and-dma/ta-p/49809)). For this, it is required to enable DMA request generation on the timer/PWM peripheral.

### Problem description

The PWM peripheral can be used to generate complex signals such as sine waves or sounds from .wav files. To achieve this, the peripheral needs to be in PWM output mode and the duty cycle needs to change once per period, also depending on the signal. The best way to achieve this without overloading the CPU is to use DMA requests triggered by the end of the PWM cycle.

On STM32 microcontrollers, this cannot be done with the current PWM and DMA APIs. STM32 microcontrollers have a TIMx_DIER with CCyDE fields which need to be set to enable the triggering of DMA requests. Other microcontrollers such as the [nrf54L](https://docs.nordicsemi.com/bundle/ps_nrf54L15/page/pwm.html#ariaid-title25), [Microchip PIC32CM](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU32/ProductDocuments/DataSheets/PIC32CM-JH00-JH01-Family-Data-Sheet-DS60001632.pdf), and [TI CC13x2, CC26x2](https://www.ti.com/lit/ug/swcu185g/swcu185g.pdf?ts=1744717065656&ref_url=https%253A%252F%252Fwww.google.com%252F) also need some kind of subscription/enable to allow these DMA requests to happen.

### Proposed change

Add optional functions to the PWM API which allow to enable and disable DMA requests triggered by a given PWM channel.

## Detailed RFC

### Proposed change (Detailed)

I propose to add a Kconfig parameter `PWM_WITH_DMA` which extends the Zephyr PWM API with the ability to trigger DMA requests from PWM channels. When enabled, this provides two functions: `pwm_enable_dma(const struct device *dev, uint32_t channel)` and `pwm_disable_dma(const struct device *dev, uint32_t channel)`.

For STM32, the `pwm_enable_dma` function sets the appropriate CCyDE field of the TIMx_DIER register where y is the given channel and x can be derived from the given device. The `pwm_disable_dma` function clears this field. On STM32 timers, these fields can be set for channels that support Input Capture/Output Compare. However, the DMA requests can be used for other purposes as mentioned above. Therefore, I propose the functions to support channels 1, 2, 3 and 4, as is the default case for the `pwm_stm32_enable/disable_capture` functions. I also believe the functions need to be independent of `CONFIG_PWM_CAPTURE`.

For other architectures/vendors, the functions should call the publish/subscribe API (nordic) or set the appropriate registers (Microchip, Texas Instruments). However, for some of them (e.g. Microchip), these registers are located on the DMA itself, not on the timer.

### Dependencies

This change affects the PWM API. For some vendors, it might affect the DMA API.

### Concerns and Unresolved Questions

On STM32, for the DMA request to be actually triggered by PWM cycles, the TIMx_DIER_CCyDE field needs to be set. For other vendors/architectures, I have only read documentation. I have not confirmed how it works there.

On STM32, there's a correlation between the Input Capture/Output Compare functionality and DMA for PWM, despite that DMA can be used for other purposes (as confirmed by their own [TIM_DMA](https://github.com/STMicroelectronics/STM32CubeF0/tree/3c780f555e7ce3f48deb8ce0ee1623320e2e7aee/Projects/STM32F091RC-Nucleo/Examples/TIM/TIM_DMA) example).

## Alternatives

The use case I have in mind is specifically focused on using DMA to change the duty cycle (TIMx_CCRy register). The alternative to adding enable and disable DMA functions, is to add a function which does both the DMA and PWM setup. For that, that function needs to take care of starting PWM in output mode on a given channel, configure a DMA request with given in-memory array of data (possibly using a struct like `dma_config`), enable DMA request triggering by the given PWM channel and starting the DMA.

SiViSur, 2025-04-15:
Added a possible solution in #88671

TheArkadiuszGrzelka, 2026-08-19:
Coming to this from nRF, where the enable/disable form cannot be implemented at
all, with a driver behind the argument rather than a preference. The revert of
#88671 does not seem to have been discussed here, so this may also be useful as
a way forward rather than only as a second opinion.

**The revert reason looks like a property of the split rather than of the
implementation.** #88671 was reverted in `5ac8a0e2aed` with "Proposed API change
result in vendor specific details of the DMA leaking into application code,
which prevents usage with portable code". `pwm_enable_dma()` only arms the
request line, so the transfer itself still has to be built by the caller: the
peripheral register address, the request line, the transfer width and the burst
behaviour are all chosen outside the driver. Those are vendor facts, so a
portable caller branches per SoC however the enable call is spelled.

**That also puts pwm at odds with how the rest of the tree handles DMA.** Every
other class that uses DMA hides it behind the driver: the driver takes a `dmas`
phandle from devicetree and calls `dma_config()` itself, and the application
never sees the DMA controller. Counting files that call `dma_config()` directly:
spi 25, serial 15, i2s 9, i2c 7, adc 7, then flash, video, mipi_dbi, mspi, usb.
`pwm_enable_dma()` would have been the first API to invert that, which may be
the deeper reason the leak was hard to avoid.

**On nRF there is nothing to pair `pwm_enable_dma()` with.** Zephyr has no dma
driver-class implementation for nRF: `drivers/dma/` carries no `dma_nrf*` and
`dts/bindings/dma/` no Nordic binding. nRF has no DMA controller with
allocatable channels; EasyDMA lives inside each peripheral and is reachable only
through that peripheral's own nrfx driver. So on the whole nRF family the pair
has nothing to arm, while the underlying capability, streaming a buffer of
compare values with the CPU uninvolved, is present and in use
(`nrfx_pwm_simple_playback()`).

**Which points back at the alternative this RFC already names.** One call that
takes the buffer and performs the setup keeps the vendor detail inside the
driver, in the same shape as spi, i2s and the rest. Roughly:

```c
int pwm_set_sequence(const struct device *dev, uint32_t channel,
                     const uint16_t *values, size_t count, uint32_t flags);
```

On STM32 that is what #88671 already implemented, with the DMA half moved behind
the driver boundary instead of in front of it: same CCyDE bit, same DMA-to-CCR
transfer, set up from a `dmas` phandle the way `spi_ll_stm32` and `i2s_ll_stm32`
do. On nRF it maps onto `nrfx_pwm_simple_playback()`, no dma driver required.

**The element type is the part I would want reviewed first,** because getting it
wrong reintroduces exactly the portability problem above. nRF sequence words
carry the output polarity in bit 15, while STM32 wants plain CCR values.

I measured what that bit actually does on an nRF5340, driving PWM0 at 16 MHz
with COUNTERTOP 16000 and a compare of 4000, and sampling the driven pin
through its own input buffer:

| sequence word | `pin_inverted` | measured high time |
| --- | --- | --- |
| 0x0fa0 | false | 75.0 % |
| 0x8fa0 | false | 25.0 % |
| 0x0fa0 | true  | 75.0 % |
| 0x8fa0 | true  | 25.0 % |

Two things follow. Bit 15 inverts the waveform, so with it clear the compare
field is the low time and with it set it is the high time. And `pin_inverted`
does not move the waveform at all: it sets the idle level, nothing more. So
polarity cannot be pushed into the pin configuration, and a portable buffer of
plain compare counts costs nRF a transform of the caller's buffer inside the
driver.

That cost is smaller than it sounds, because the driver already owns that bit.
`drivers/pwm/pwm_nrfx.c` composes every sequence word as
`compare_value | (inverted ? 0 : POLARITY_MASK)`, so today's callers never see
it. Keeping it that way for a buffer call is consistent, not new.

Completion signalling is the other open question: callback, blocking wait, or
both, and whether repeat and end-delay belong in the call or in the flags.

**Where this comes from.** I have just submitted a WS2812 led_strip backend for
nRF (#116678) that streams a frame through nrfx directly, because the pwm class
cannot express it. It is vendor-specific by construction for that reason alone,
and it would collapse into a generic "WS2812 over any PWM" backend the moment a
call like the above existed. Its framing is verified on a logic analyzer rather
than by eye: 250 ns and 750 ns symbols on a 2000 ns period, with the symbol
counts matching the requested pattern exactly across 18 consecutive frames. So I
can prototype the nRF side of such an API if the direction is acceptable.