zephyrproject-rtos/zephyr #88686: modem: Cellular modem sample fails with RAK5010 board (quectel bg95 modem)

olalonde, 2025-04-16:
**Describe the bug**

It seems there is an issue with the quectel bg95 modem driver at the `AT+CREG?` command. It gets stuck there and times out.

**To Reproduce**

Build and flash the [cellular modem](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/cellular_modem/src/main.c) sample on a RAK5010-M-BG95 board.

I'm using a valid Claro (Dominican Republic) SIM card in an area with EGPRS signal.

**Expected behavior**

L4 should connect in time.

**Impact**

showstopper

**Logs and console output**

```
SEGGER J-Link V7.94i - Real time terminal output
SEGGER J-Link (unknown) V2.0, SN=802000342
Process: JLinkExe
*** Booting nRF Connect SDK v2.9.1-60d0d6c8d42d ***
*** Using Zephyr OS v3.7.99-ca954a6216c9 ***
Powering on modem
Bring up network interface
Waiting for L4 connected
[00:00:11.916,015] <dbg> modem_chat: modem_chat_script_start: running script: quectel_bg95_init_chat_script
[00:00:11.916,046] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 0
[00:00:11.916,076] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
[00:00:21.916,137] <wrn> modem_chat: quectel_bg95_init_chat_script: timed out
[00:00:25.338,958] <dbg> modem_chat: modem_chat_on_unknown_command_received: RDY
[00:00:27.911,743] <dbg> modem_chat: modem_chat_on_unknown_command_received: APP RDY
[00:00:31.877,288] <dbg> modem_chat: modem_chat_on_unknown_command_received: RDY
[00:00:33.416,473] <dbg> modem_chat: modem_chat_script_start: running script: quectel_bg95_init_chat_script
[00:00:33.416,534] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 0
[00:00:33.416,564] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
[00:00:33.437,744] <dbg> modem_chat: modem_chat_on_unknown_command_received: ATE0
[00:00:33.437,805] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.437,835] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 1
[00:00:33.437,866] <dbg> modem_chat: modem_chat_script_next: sending: AT+CFUN=4
[00:00:33.572,387] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.572,418] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 2
[00:00:33.572,448] <dbg> modem_chat: modem_chat_script_next: sending: AT+CMEE=1
[00:00:33.594,146] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.594,207] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 3
[00:00:33.594,238] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG=1
[00:00:33.615,936] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.615,966] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 4
[00:00:33.615,997] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGREG=1
[00:00:33.637,695] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.637,725] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 5
[00:00:33.637,756] <dbg> modem_chat: modem_chat_script_next: sending: AT+CEREG=1
[00:00:33.659,454] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.659,484] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 6
[00:00:33.659,515] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:00:33.681,091] <dbg> modem_chat: modem_chat_log_received_command: +CREG:  1 0
[00:00:33.681,182] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.681,213] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 7
[00:00:33.681,243] <dbg> modem_chat: modem_chat_script_next: sending: AT+CEREG?
[00:00:33.702,911] <dbg> modem_chat: modem_chat_log_received_command: +CEREG:  1 4
[00:00:33.702,972] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.703,002] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 8
[00:00:33.703,033] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGREG?
[00:00:33.724,731] <dbg> modem_chat: modem_chat_log_received_command: +CGREG:  1 0
[00:00:33.724,792] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:33.724,822] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 9
[00:00:33.724,853] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGSN
[... 82 lines cut ...]
[00:00:37.753,997] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 12)
[00:00:37.781,829] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:37.781,951] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 12)
[00:00:37.782,012] <dbg> modem_ppp: modem_ppp_process_received_byte: Receiving PPP frame
[00:00:37.782,104] <dbg> modem_ppp: modem_ppp_process_received_byte: Received PPP frame (len 9)
[00:01:07.678,649] <dbg> modem_chat: modem_chat_script_start: running script: quectel_bg95_periodic_chat_script
[00:01:07.678,680] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_periodic_chat_script: step: 0
[00:01:07.678,710] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:01:11.678,802] <wrn> modem_chat: quectel_bg95_periodic_chat_script: timed out
[00:01:41.678,985] <dbg> modem_chat: modem_chat_script_start: running script: quectel_bg95_periodic_chat_script
[00:01:41.679,046] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_periodic_chat_script: step: 0
[00:01:41.679,077] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:01:45.679,138] <wrn> modem_chat: quectel_bg95_periodic_chat_script: timed out
L4 was not connected in time
```

**Environment (please complete the following information):**

 - OS: Macos
 - Toolchain: Zephyr SDK / nordic vscode extension
 - Commit SHA or Version used: Zephyr OS v3.7.99-ca954a6216c9

**Additional context**

See discussion: https://github.com/zephyrproject-rtos/zephyr/discussions/88678

olalonde, 2025-04-16:
After analyzing the logs with ChatGPT:

> Earlier in the init and periodic scripts, you can see this:
> 
> ```
> +CREG:  1,0  → registered to home network (OK for CS)
> +CGREG: 1,0  → registered for GPRS (OK for PS)
> +CEREG: 1,4  → **not registered, and searching** (critical)
> +CEREG: 1,4 means EPS (LTE/4G) registration is still in progress or failing.
> ```
> 
> This suggests the modem:
> 
> Hasn't fully attached to the LTE network
> 
> Likely cannot establish a data session when the PPP is triggered.

I know that there is not LTE-m or nb-iot in Dominican Republic so I expect the modem to connect to a 2G network. Does the driver only support connecting to a LTE network?

0x447a, 2025-04-17:
@olalonde the cellular network supported are defined in an enum and are delcared as such:

`enum cellular_access_technology {
	CELLULAR_ACCESS_TECHNOLOGY_GSM = 0,
	CELLULAR_ACCESS_TECHNOLOGY_GPRS,
	CELLULAR_ACCESS_TECHNOLOGY_UMTS,
	CELLULAR_ACCESS_TECHNOLOGY_EDGE,
	CELLULAR_ACCESS_TECHNOLOGY_LTE,
	CELLULAR_ACCESS_TECHNOLOGY_LTE_CAT_M1,
	CELLULAR_ACCESS_TECHNOLOGY_LTE_CAT_M2,
	CELLULAR_ACCESS_TECHNOLOGY_NB_IOT,
};`

As you can see, only LTE alongside the other LTE variants that you already cannot use are defined in `zephyr/include/zephyr/drivers/cellular.h`

[https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/cellular.h](url)

olalonde, 2025-04-17:
I am not sure what you mean... The enum you copy pasted includes non LTE technologies like GSM. Where do you see that only LTE is supported?

n3rd4i, 2025-06-03:
> To the dial-script commands try adding `AT+CGACT?` and `AT+CGDCONT?` after the `AT+CFUN=1` command is executed.

Hi. I have the same issue described by this ticket, I would like to report the output of these commands:
```
[00:00:02.090,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_dial_chat_script: step: 2
[00:00:02.090,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CFUN=1
[00:00:02.170,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:02.170,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_dial_chat_script: step: 3
[00:00:02.170,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGACT?
[00:00:02.230,000] <dbg> modem_chat: modem_chat_on_unknown_command_received: +CGACT: 1,0
[00:00:02.230,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:02.230,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_dial_chat_script: step: 4
[00:00:02.230,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGDCONT?
[00:00:02.290,000] <dbg> modem_chat: modem_chat_on_unknown_command_received: +CGDCONT: 1,"IP","eciot","0.0.0.0",0,0,0
[00:00:02.290,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:02.290,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_dial_chat_script: step: 5
[00:00:02.290,000] <dbg> modem_chat: modem_chat_script_next: sending: ATD*99***1#
[00:00:02.290,000] <dbg> modem_chat: modem_chat_script_stop: quectel_bg95_dial_chat_script: complete
[00:00:02.350,000] <dbg> modem_ppp: modem_ppp_is_byte_expected: Dropping byte 0x0d because 0x7e was expected.
[00:00:02.350,000] <dbg> modem_ppp: modem_ppp_is_byte_expected: Dropping byte 0x0a because 0x7e was expected.

```

I would also like to help on debugging this issue. regards.

n3rd4i, 2025-06-09:
Update:

After removing this section from the `sample/net/cellular_modem`, the sample started working for me:
* https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0/samples/net/cellular_modem/src/main.c#L307-L313

```
*** Booting nRF Connect SDK v2.7.0-5cb85570ca43 ***
*** Using Zephyr OS v3.6.99-97e91ed9ce1b ***
Powering on modem
[00:00:00.000,000] <dbg> modem_chat: modem_chat_script_start: running script: quectel_bg95_init_chat_script
[00:00:00.000,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 0
[00:00:00.000,000] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
Bring up network interface
Waiting for L4 connected
[00:00:00.090,000] <dbg> modem_chat: modem_chat_on_unknown_command_received: ATE0
[00:00:00.090,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.090,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 1
[00:00:00.090,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CFUN=4
[00:00:00.410,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.410,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 2
[00:00:00.410,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CMEE=1
[00:00:00.470,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.470,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 3
[00:00:00.470,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG=1
[00:00:00.530,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.530,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 4
[00:00:00.530,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGREG=1
[00:00:00.590,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.590,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 5
[00:00:00.590,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:00:00.650,000] <dbg> modem_chat: modem_chat_log_received_command: +CREG:  1 0
[00:00:00.650,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.650,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 6
[00:00:00.650,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CEREG?
[00:00:00.710,000] <dbg> modem_chat: modem_chat_log_received_command: +CEREG:  0 0
[00:00:00.710,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.710,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 7
[00:00:00.710,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGREG?
[00:00:00.770,000] <dbg> modem_chat: modem_chat_log_received_command: +CGREG:  1 4
[00:00:00.770,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.770,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 8
[00:00:00.770,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CPIN?
[00:00:00.830,000] <dbg> modem_chat: modem_chat_on_unknown_command_received: +CPIN: READY
[00:00:00.830,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.830,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 9
[00:00:00.830,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGSN
[00:00:00.890,000] <dbg> modem_chat: modem_chat_log_received_command:  ***************
[00:00:00.890,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 10
[00:00:00.890,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.890,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 11
[00:00:00.890,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGMM
[00:00:00.950,000] <dbg> modem_chat: modem_chat_log_received_command:  BG95-M3
[00:00:00.950,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 12
[00:00:00.950,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:00.950,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 13
[00:00:00.950,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGMI
[00:00:01.010,000] <dbg> modem_chat: modem_chat_log_received_command:  Quectel
[00:00:01.010,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 14
[00:00:01.010,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:01.010,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 15
[00:00:01.010,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+QGMR
[00:00:01.070,000] <dbg> modem_chat: modem_chat_log_received_command:  BG95M3LAR02A03_01.012.01.012
[00:00:01.070,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 16
[00:00:01.070,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:01.070,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 17
[00:00:01.070,000] <dbg> modem_chat: modem_chat_script_next: sending: AT+CIMI
[00:00:01.130,000] <dbg> modem_chat: modem_chat_log_received_command:  ***************
[00:00:01.130,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 18
[00:00:01.130,000] <dbg> modem_chat: modem_chat_log_received_command: OK
[00:00:01.130,000] <dbg> modem_chat: modem_chat_script_next: quectel_bg95_init_chat_script: step: 19
[... cut ...]
 MODEM_CHAT_SCRIPT_DEFINE(quectel_bg95_init_chat_script, quectel_bg95_init_chat_script_cmds,
 			 abort_matches, modem_cellular_chat_callback_handler, 10);
@@ -1634,6 +1634,7 @@ MODEM_CHAT_SCRIPT_CMDS_DEFINE(quectel_bg95_dial_chat_script_cmds,
 			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGDCONT=1,\"IP\","
 							 "\""CONFIG_MODEM_CELLULAR_APN"\"",
 							 ok_match),
+			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+QCFG=\"nwscanseq\",0201", ok_match),
 			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CFUN=1", ok_match),
 			      MODEM_CHAT_SCRIPT_CMD_RESP_NONE("ATD*99***1#", 0),);
```

closed 2025-08-24 by github-actions[bot]