zephyrproject-rtos/zephyr #88692: shell: build failure due to definition conflicts

hakehuang, 2025-04-16:
**Describe the bug**

```
303   \| #define _POSIX2_COLL_WEIGHTS_MAX (2) \|   /opt/toolchains/zephyr-sdk-0.17.1-rc1/arm-zephyr-eabi/picolibc/include/sys/syslimits.h:105:   note: this is the location of the previous definition 105 \| #define   _POSIX2_COLL_WEIGHTS_MAX 2 \| /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/posix/posix_features.h:304:   error: "_POSIX2_EXPR_NEST_MAX" redefined [-Werror] 304 \| #define   _POSIX2_EXPR_NEST_MAX (32) \|   /opt/toolchains/zephyr-sdk-0.17.1-rc1/arm-zephyr-eabi/picolibc/include/sys/syslimits.h:106:   note: this is the location of the previous definition 106 \| #define   _POSIX2_EXPR_NEST_MAX 32 \|   /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/include/zephyr/posix/posix_features.h:305:   error: "_POSIX2_LINE_MAX" redefined [-Werror] 305 \| #define   _POSIX2_LINE_MAX (2048) \|   /opt/toolchains/zephyr-sdk-0.17.1-rc1/arm-zephyr-eabi/picolibc/include/sys/syslimits.h:107:   note: this is the location of the previous definition 107 \| #define   _POSIX2_LINE_MAX 2048 \| cc1: all warnings being treated as errorsninja: build   stopped: subcommand failed.

```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - any platform
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 -

**To Reproduce**

scripts/twister -p mimxrt1180_evk/mimxrt1189/cm7 -j 12 -T tests -T samples --build-only --inline-logs -M all 

**Expected behavior**

Build pass

**Impact**

shell

**Logs and console output**

see above

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows) ALL
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-2465-gb2b7a468eadd

mmahadevan108, 2025-04-16:
@stephanosio can you help take a look at this.

nashif, 2025-04-16:
why are you submitting bugs against an unreleased sdk? This is because we did not merge the needed picolibc patches into Zephyr yet. you need those for this to work.

hakehuang, 2025-04-17:
> why are you submitting bugs against an unreleased sdk? This is because we did not merge the needed picolibc patches into Zephyr yet. you need those for this to work.

@nashif, the sdk is deploy id the docker, my system pull the latest zephyr docker.

danieldegrasse, 2025-04-22:
@hakehuang, going to close this as testing shouldn't be done using pre-release SDK revisions. Rather, the same docker image should be used as is used in upstream CI.

closed 2025-04-22 by danieldegrasse

hakehuang, 2025-04-23:
> [@hakehuang](https://github.com/hakehuang), going to close this as testing shouldn't be done using pre-release SDK revisions. Rather, the same docker image should be used as is used in upstream CI.

Thanks, just try to set sentry alert for the coming release, as we always pull the dock-src to build image, so once this update, our test image is updated as well.