zephyrproject-rtos/zephyr #88697: hostap: tls: Implement server name verification

krish2718, 2025-04-16:
**Describe the bug**

MbedTLS 3.6.3 now mandates servername checking unless explicitly disabled and this causes TLS handshake to fail, see https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.3.

A temporary WAR https://github.com/zephyrproject-rtos/zephyr/pull/88679 to disable this has been raised, but this is a security loophole, we need to implement this in the hostap. Which already supports https://github.com/zephyrproject-rtos/hostap/blob/main/wpa_supplicant/wpa_supplicant.conf#L1329 (`domain_suffic_match`) option and this needs to be plumbed down till MbedTLS.

**To Reproduce**

Steps to reproduce the behavior:
1. Build and Run `wifi-enterprise` snippet in `samples/net/wifi/shell` on a nRF7002DK board
2. Test EAP-TLS
3. Handshake fails without the above PR (WAR)

**Expected behavior**
TLS should pass but also in a secure way (not with the WAR)

**Impact**
Security vulnerability.

**Logs and console output**
NA

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Latest main 86293ebe114

**Additional context**
NA

d3zd3z, 2025-04-29:
"security loophole" is a very gentle way to put this. This effectively removes the point in using TLS at all, since it becomes trivial to mount a MitM attack.

My suggestion would be to revert #88679, until an actual fix can be found.

tomi-font, 2025-04-30:
> So, looking further, this doesn't disable hostname checking, it just makes it "work again", and as far as I can tell, hostname checking has never been present?

Correct, the previous default behavior in Mbed TLS was to not check the hostname if `mbedtls_ssl_set_hostname()` wasn't called.

cfriedt, 2025-08-05:
@jukkar @krish2718 @sachinthegreen @ceolin - do you know if this can be closed, or is this still an open issue for 4.3.0?

cfriedt, 2025-10-28:
@krish2718 - can you give a status update? Is there a PR that can be linked?

krish2718, 2025-10-28:
> [@krish2718](https://github.com/krish2718) - can you give a status update? Is there a PR that can be linked?

There is a PR https://github.com/zephyrproject-rtos/zephyr/pull/98190 raised but there seems to be some confusion as it only has revert not the implementation, I will check internally and update.

closed 2025-11-14 by nashif