zephyrproject-rtos/zephyr #88704: tests: spi: spi_loopback fails with test_spi_null_rx_buf_set

hakehuang, 2025-04-16:
**Describe the bug**

test_spi_null_rx_buf_set fails on several NXP platforms

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - frdm_ke17z/frdm_ke17z512/frdm_k64f
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
b983b0441b4ad6f11edfb678c09dc7237570c5e4 is the first bad commit
commit b983b0441b4ad6f11edfb678c09dc7237570c5e4
Author: Piotr Krzyzanowski <piotr.krzyzanowski@nordicsemi.no>
Date:   Mon Mar 24 22:58:15 2025 +0100

    tests: drivers: spi: Add NULL spi_buf_set tests

    Add test cases where spi_transceive gets NULL spi_buf_set
    once for tx and once for rx.

    Signed-off-by: Piotr Krzyzanowski <piotr.krzyzanowski@nordicsemi.no>

 tests/drivers/spi/spi_loopback/src/spi.c | 24 ++++++++++++++++++++++++
 1 file changed, 24 insertions(+)

```

**To Reproduce**

```
scripts/twister --device-testing --hardware-map map.yaml -T tests/drivers/spi/spi_loopback/
```

**Expected behavior**

test PASS

**Impact**

spi over flexspi

**Logs and console output**

```
*** Booting Zephyr OS build v4.1.0-2445-g246f02b09460 ***
SPI test on buffers TX/RX 0x20004380/0x20004360, frame size = 8, DMA enabled (without CONFIG_NOCACHE_MEMORY)
Polling...Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.002 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.002 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: SLOW
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_async_call
 PASS - test_spi_async_call in 0.235 seconds
===================================================================
START - test_spi_complete_large_transfers
 PASS - test_spi_complete_large_transfers in 0.232 seconds
===================================================================
START - test_spi_complete_loop
 PASS - test_spi_complete_loop in 0.001 seconds
===================================================================
START - test_spi_complete_multiple
 PASS - test_spi_complete_multiple in 0.002 seconds
===================================================================
START - test_spi_null_rx_buf_set
E: Transfer could not start
E: Timeout waiting for transfer complete

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:109: spi_loopback_transceive: (ret is true)
SPI transceive failed, code -116
 FAIL - test_spi_null_rx_buf_set in 0.221 seconds
===================================================================
START - test_spi_null_tx_buf
 PASS - test_spi_null_tx_buf in 0.001 seconds
===================================================================
START - test_spi_null_tx_buf_set
 PASS - test_spi_null_tx_buf_set in 0.001 seconds
===================================================================
START - test_spi_rx_bigger_than_tx
 PASS - test_spi_rx_bigger_than_tx in 0.001 seconds
===================================================================
START - test_spi_rx_every_4
 PASS - test_spi_rx_every_4 in 0.001 seconds
===================================================================
START - test_spi_rx_half_end
 PASS - test_spi_rx_half_end in 0.001 seconds
===================================================================
START - test_spi_rx_half_start
 PASS - test_spi_rx_half_start in 0.001 seconds
===================================================================
TESTSUITE spi_loopback failed.
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: FAST
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_async_call
 PASS - test_spi_async_call in 0.144 seconds
===================================================================
START - test_spi_complete_large_transfers
 PASS - test_spi_complete_large_transfers in 0.141 seconds
===================================================================
START - test_spi_complete_loop
 PASS - test_spi_complete_loop in 0.001 seconds
===================================================================
START - test_spi_complete_multiple
 PASS - test_spi_complete_multiple in 0.002 seconds
===================================================================
START - test_spi_null_rx_buf_set
E: Transfer could not start
E: Timeout waiting for transfer complete

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:109: spi_loopback_transceive: (ret is true)
SPI transceive failed, code -116
 FAIL - test_spi_null_rx_buf_set in 0.221 seconds
===================================================================
START - test_spi_null_tx_buf
 PASS - test_spi_null_tx_buf in 0.001 seconds
===================================================================
START - test_spi_null_tx_buf_set
 PASS - test_spi_null_tx_buf_set in 0.001 seconds
===================================================================
START - test_spi_rx_bigger_than_tx
 PASS - test_spi_rx_bigger_than_tx in 0.001 seconds
===================================================================
START - test_spi_rx_every_4
 PASS - test_spi_rx_every_4 in 0.001 seconds
===================================================================
START - test_spi_rx_half_end
 PASS - test_spi_rx_half_end in 0.001 seconds
===================================================================
START - test_spi_rx_half_start
 PASS - test_spi_rx_half_start in 0.001 seconds
===================================================================
TESTSUITE spi_loopback failed.

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [spi_extra_api_features]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.002 seconds
 - PASS - [spi_extra_api_features.test_spi_lock_release] duration = 0.002 seconds

SUITE FAIL -  91.67% [spi_loopback]: pass = 11, fail = 1, skip = 0, total = 12 duration = 0.698 seconds
 - PASS - [spi_loopback.test_nop_nil_bufs] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_async_call] duration = 0.235 seconds
 - PASS - [spi_loopback.test_spi_complete_large_transfers] duration = 0.232 seconds
 - PASS - [spi_loopback.test_spi_complete_loop] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_complete_multiple] duration = 0.002 seconds
 - FAIL - [spi_loopback.test_spi_null_rx_buf_set] duration = 0.221 seconds
 - PASS - [spi_loopback.test_spi_null_tx_buf] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_null_tx_buf_set] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_bigger_than_tx] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_every_4] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_half_end] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_half_start] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: c7ffdb2648f21a877a72ffdc41103a86
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-2494-g86293ebe114

kasjer, 2025-05-12:
This affects other platforms as well
```c
ZTEST(spi_loopback, test_spi_null_rx_buf_set)
{
	struct spi_dt_spec *spec = loopback_specs[spec_idx];
	const struct spi_buf_set tx = spi_loopback_setup_xfer(tx_bufs_pool, 1,
							      NULL, BUF_SIZE); // <-- third parameter should not be NULL, code requests to have transmission of 18 bytes without providing rx or tx buffer

	spi_loopback_transceive(spec, &tx, NULL);
}
```

fnadeau, 2025-05-23:
On the FRDM-RW612, I have the following output

```
*** Booting Zephyr OS build v4.1.0-3734-g7f3e336ee4fd ***
SPI test on buffers TX/RX 0x10044b00/0x10044ae0, frame size = 8
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: SLOW
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_complete_large_transfers
 PASS - test_spi_complete_large_transfers in 0.132 seconds
===================================================================
START - test_spi_complete_loop
 PASS - test_spi_complete_loop in 0.001 seconds
===================================================================
START - test_spi_complete_multiple
 PASS - test_spi_complete_multiple in 0.001 seconds
===================================================================
START - test_spi_null_rx_buf_set
ASSERTION FAIL [0] @ WEST_TOPDIR/deps/zephyr/lib/libc/picolibc/assert.c:27
E: r0/a1:  0x00000004  r1/a2:  0x0000001b  r2/a3:  0x00000006
E: r3/a4:  0x00000004 r12/ip:  0x00000008 r14/lr:  0x18003545
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x18007ec2
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x10044b18 (test_spi_null_rx_buf_set)
E: Halting system
```

As @kasjer mention, third argument shouldn't be NULL. I would guess that test `test_spi_null_rx_buf_set` should be same-ish as `test_spi_null_tx_buf_set`. Changing third argument to `buffer_tx` fixed the issue.

hakehuang, 2025-06-05:
and in mimxrt1170 platfrom:
```
*** Booting Zephyr OS build v4.1.0-5242-gf2f3e1797f10 ***
SPI test on buffers TX/RX 0x80004160/0x80004140, frame size = 8, DMA enabled
Polling...Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.002 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.001 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_extra_api_features
===================================================================
START - test_spi_hold_on_cs
 PASS - test_spi_hold_on_cs in 0.002 seconds
===================================================================
START - test_spi_lock_release
 PASS - test_spi_lock_release in 0.001 seconds
===================================================================
TESTSUITE spi_extra_api_features succeeded
Running TESTSUITE spi_loopback
===================================================================
Testing loopback spec: SLOW
START - test_nop_nil_bufs
 PASS - test_nop_nil_bufs in 0.001 seconds
===================================================================
START - test_spi_async_call
 PASS - test_spi_async_call in 0.137 seconds
===================================================================
START - test_spi_complete_large_transfers
 PASS - test_spi_complete_large_transfers in 0.136 seconds
===================================================================
START - test_spi_complete_loop_mode_0
 PASS - test_spi_complete_loop_mode_0 in 0.001 seconds
===================================================================
START - test_spi_complete_loop_mode_1
 PASS - test_spi_complete_loop_mode_1 in 0.001 seconds
===================================================================
START - test_spi_complete_loop_mode_2
 PASS - test_spi_complete_loop_mode_2 in 0.001 seconds
===================================================================
START - test_spi_complete_loop_mode_3
 PASS - test_spi_complete_loop_mode_3 in 0.001 seconds
===================================================================
START - test_spi_complete_multiple
 PASS - test_spi_complete_multiple in 0.002 seconds
===================================================================
START - test_spi_complete_multiple_timed
Transfer took 962 us vs theoretical minimum 864 us
Latency measurement: 98 us
 PASS - test_spi_complete_multiple_timed in 0.008 seconds
===================================================================
START - test_spi_concurrent_transfer_different_spec
SPI concurrent transfer data mismatch, spec 0x80008bfc
SPI concurrent transfer data mismatch, spec 0x80008bcc
SPI concurrent transfer data mismatch, spec 0x80008be4

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:776: test_spi_concurrent_transfer_helper: atomic_get(&thread_test_fails) not equal to 0

 FAIL - test_spi_concurrent_transfer_different_spec in 0.032 seconds
===================================================================
START - test_spi_concurrent_transfer_same_spec
SPI concurrent transfer data mismatch, spec 0x80008028
SPI concurrent transfer data mismatch, spec 0x80008028
SPI concurrent transfer data mismatch, spec 0x80008028

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:776: test_spi_concurrent_transfer_helper: atomic_get(&thread_test_fails) not equal to 0

[... 166 lines cut ...]
 - PASS - [spi_loopback.test_spi_complete_multiple_timed] duration = 0.008 seconds
 - FAIL - [spi_loopback.test_spi_concurrent_transfer_different_spec] duration = 0.032 seconds
 - FAIL - [spi_loopback.test_spi_concurrent_transfer_same_spec] duration = 0.032 seconds
 - PASS - [spi_loopback.test_spi_null_rx_buf_set] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_null_tx_buf] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_null_tx_buf_set] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_bigger_than_tx] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_every_4] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_half_end] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_rx_half_start] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_same_buf_cmd] duration = 0.001 seconds
 - PASS - [spi_loopback.test_spi_word_size_16] duration = 0.001 seconds
 - FAIL - [spi_loopback.test_spi_word_size_24] duration = 0.022 seconds
 - PASS - [spi_loopback.test_spi_word_size_32] duration = 0.004 seconds
 - PASS - [spi_loopback.test_spi_word_size_7] duration = 0.001 seconds
 - FAIL - [spi_loopback.test_spi_word_size_9] duration = 0.020 seconds
 - PASS - [spi_loopback.test_spi_write_back] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 7c21a69d98f1c4962b548119657718db
PROJECT EXECUTION FAILED

```

decsny, 2025-06-10:
I am not able to reproduce on RT1170 and the message above from issue author also shows this test case does not fail, so i will change the original message with strikethrough