zephyrproject-rtos/zephyr #88733: RTIO: default to the fallback function when driver doesnt support RTIO

VynDragon, 2025-04-16:
**Description**
When using a driver that uses RTIO (for example sensor async api provider), it will not default to the i2c_iodev_submit_fallback function if it has not been implemented by the driver.

**Solution**
Implement automatic default fallback.

ubieda, 2025-04-16:
Thanks for capturing the issue.

IIRC - There was a reason why the auto-fallback was not implemented for bus-drivers. I think the run-time check was something we wanted to stay away from (it would apply both to those having it). However, I see for I2C and I3C we do have a run-time check that resolves in an error:

https://github.com/zephyrproject-rtos/zephyr/blob/e1a0350d23706fa1201d5f37190c77141abe8197/include/zephyr/drivers/i2c.h#L1053-L1064

https://github.com/zephyrproject-rtos/zephyr/blob/e1a0350d23706fa1201d5f37190c77141abe8197/include/zephyr/drivers/i3c.h#L2481-L2492

This is not consistent with SPI: 

https://github.com/zephyrproject-rtos/zephyr/blob/e1a0350d23706fa1201d5f37190c77141abe8197/include/zephyr/drivers/spi.h#L1234-L1241

To move forward, we should standardize this pattern, by either:

1. Auto-fallback should the submit API is not implemented, or
2. Throw a compile-time error indicating that a particular bus-driver does not implement it, while the corresponding Kconfig flags are enabled (e.g: CONFIG_I2C_RTIO and CONFIG_SENSOR_ASYNC_API).

I would rather prefer No. 2, but TBD how that would be implemented.

ubieda, 2025-04-16:
@teburd @yperess feel free to chime in here

teburd, 2025-04-16:
Compile time error is the right option here imho

closed 2026-09-26 by github-actions[bot]