zephyrproject-rtos/zephyr #88771: Wi-Fi certificate re-distribution

jukkar, 2025-04-17:
This is related to discussion in #87656 and the question is if it is allowed to re-distribute Wi-Fi Alliance certificates that are currently present in `samples/net/wifi/test_certs` directory.

The certificates show this information in them
```
$ openssl x509 -in client.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            97:d4:07:ec:a6:05:15:13
        Signature Algorithm: sha384WithRSAEncryption
        Issuer: C = US, L = Santa Clara, O = Wi-Fi Alliance, CN = Suite B RSA 3k Root CA
        Validity
            Not Before: Aug 27 00:22:56 2019 GMT
            Not After : Aug 24 00:22:56 2029 GMT
...
```
They were merged as part of this PR #74847

jukkar, 2025-04-17:
cc: @kartben @carlescufi @dleach02

MaochenWang1, 2025-04-21:
Hi @krish2718 how did you generate the RSA2k certs and key files? any steps?

krish2718, 2025-04-21:
Please see the note https://docs.zephyrproject.org/latest/connectivity/networking/api/wifi.html here, I used freeradius raddb scripts.

MaochenWang1, 2025-04-21:
@krish2718  I still have a question, why need to keep both `samples/net/wifi/test_certs/rsa2k` and `samples/net/wifi/test_certs/rsa3k`?
Since `samples/net/wifi/test_certs/rsa3k` are Wi-Fi Alliance certificates, why not just move the certs from `samples/net/wifi/test_certs/rsa2k` into `samples/net/wifi/test_certs/`, and remove `rsa3k and rsa2k`?

krish2718, 2025-04-21:
Different platforms have varying crypto abilities so had made variants of certs ( I was planning to submit another rsa2k variant soon). Of course these are just for reference to verify quickly using golden certs, users can always provide their own.