zephyrproject-rtos/zephyr #88772: xtensa: Shared memory between different privilege threads leads to MultiHit/LoadStorePrivilege exception

ldmstm, 2025-04-17:
**Describe the bug**
When targeting the Xtensa architecture with MMU & userspace enabled, accessing shared memory between an unprivileged (user mode) thread and a privileged (kernel) thread can lead to a MultiHit or LoadStorePrivilege exception due to the ring levels being inconsistent between them.

The userspace thread will access memory at ring 2 (and the DTLB will only hit when the RASID matches), but the kernel thread will do so at ring 0 (the pagetable mapping says to do so), so a DTLB entry will be mapped with ASID = 0x01, which is always present at ring 0. Under certain circumstances, there could be mappings with ASID = 0x01 and ASID = <ring2 of the userspace thread>, leading to a multihit tlb exception when accessing the memory from a privileged context, like an exception handler (syscall, interrupt, window overflow/underflow, debug, etc).

The current workaround is to also change the kernel pagetable mappings to make this memory mapping into a ring 2 mapping at the privileged level.

This issue was first observed after upgrading from Zephyr v3.6.0 to v3.7.0, but it's unknown if this was present before.

**To Reproduce**
1. Download the attached test case [tlbexample.zip](https://github.com/user-attachments/files/19796179/tlbexample.zip)
2. Compile and run for the `qemu_xtensa/dc233b/mmu` board
3. Should get a LoadStorePrivilegeException or a MultiHitTLBException, depending on the macro expansion in `src/priv.c:14`

If you want to observe the other exception, the macro can be toggled.

**Expected behavior**
The program runs infinitely, bouncing between the `priv` and the `user` prints (can be observed when building and running for `qemu_x86/atom/virt`.

**Impact**
Very high, we allocate memory in the user thread stack that will then be accessed (read-only) from a privileged thread, but the two accesses cause this issue in some conditions. When this happens in a window exception, the RTOS hangs completely and enters a triple fault, without printing any error messages.

**Logs and console output**
See the attached logs (obtained from `qemu-system-xtensa`)
[loadstorepriv_log.txt](https://github.com/user-attachments/files/19796112/loadstorepriv_log.txt)
[multihit_log.txt](https://github.com/user-attachments/files/19796111/multihit_log.txt)
[loadstorepriv_infotlb.txt](https://github.com/user-attachments/files/19796114/loadstorepriv_infotlb.txt)
[multihit_infotlb.txt](https://github.com/user-attachments/files/19796113/multihit_infotlb.txt)

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 24.04 on Windows 10 WSL)
 - Toolchain: Zephyr SDK, xtensa-dc233c_zephyr-elf
 - Zephyr v3.7.1

**Additional context**

dcpleung, 2025-04-18:
Your example relies on something that is highly discouraged: directly accessing other thread's stack. `str_ptr` is simply a pointer to the string on the user stack, and you are trying to access it via the `main` thread. Even though the `main` thread is running under privileged mode, it still should not be accessing other threads' stacks directly. Conceptually, they can be considered to running in different memory domains. If you try to access another user thread's stack in another domain, it would raise load/store privilege exception. However, between kernel and user threads, Xtensa hardware provides an indirect mechanism to catch this undesired behavior, via multi-hit TLB exception. To be honest, I don't think this needs to be fixed as the exception discourages undesired behavior.

ldmstm, 2025-04-30:
Here's a simplified example that shows off the issue without any custom syscalls, using the `k_msgq` instead.

[tlbexample.zip](https://github.com/user-attachments/files/19976524/tlbexample.zip)

> Your example relies on something that is highly discouraged: directly accessing other thread's stack.

There's an optimization in the `k_msgq` code -- when a thread:
 - wants to get a message from the msgq
 - is able to wait `K_FOREVER`
 - the msgq is currently empty

then whoever puts a message first to that msgq will bypass the queue and instead push it directly into the waiting thread's buffer. If that buffer is located in the stack, you'd encounter a very similar situation to what we see in the first example, and it seems like a reasonable use of the system. This is only one example of this happening, but there could be many others that work on other architectures but break under very specific conditions in Xtensa, because of the very particular MMU the systems have.

We couldn't find this "don't access another thread's stack" anywhere in the documentation or the code, and in our opinion "discouraged" shouldn't mean "fault-triggering". We are aware that the buffer's contents could change under our noses and get "rugpulled", and this is accepted.

We still believe this should be fixed, and memory that can be accessed by specific threads in different privileges (not necessarily shared by all, so not Ring3) should be marked as Ring2 in all mappings. Sharing TLBs is possible using Ring1, but that's way beyond the scope of this fix (and Zephyr in general).

PR #89425 (2025-05-28): xtensa: allow flushing auto-refill DTLBs on page table swap
This adds a new kconfig and corresponding code to allow flushing auto-refill data TLBs when page tables are swapped (e.g. during context switching). This is mainly used to avoid multi-hit TLB exception raised by certain memory access pattern. If memory is only marked for user mode access but not inside a memory domain, accessing that page in kernel mode would result in a TLB being filled with kernel ASID. When going back into user mode, access to the memory would result in another TLB being filled with the user mode ASID. Now there ar [... cut]
commit d31ee53b (2025-05-28): xtensa: allow flushing auto-refill DTLBs on page table swap

This adds a new kconfig and corresponding code to allow flushing
auto-refill data TLBs when page tables are swapped (e.g. during
context switching). This is mainly used to avoid multi-hit TLB
exception raised by certain memory access pattern. If memory is
only marked for user mode access but not inside a memory domain,
accessing that page in kernel mode would result in a TLB being
filled with kernel ASID. When going back into user mode, access
to the memory would result in another TLB being filled with
the user mode ASID. Now there a [... cut]
closed 2025-05-28 by kartben