zephyrproject-rtos/zephyr #88812: Sensor decoder context should not be reused in Sensor Read and Decode example

ymleung314, 2025-04-19:
See: https://docs.zephyrproject.org/latest/hardware/peripherals/sensor/read_and_decode.html

In the Polling Read section, `temp_decoder` is reused for all `sensor_decode`. The frame iterator `temp_decoder.fit` is not reset after each read-decode cycle, later call to `sensor_decode` will just return 0.

Probably the context variable should be named more clearly too.

danieldegrasse, 2025-04-22:
@ymleung314 is there a specific bug you have observed with the sample here? Or are these suggestions for code improvement?

If there is a bug here, please fill out the bug template given here: https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md

ymleung314, 2025-04-23:
It believe it's a bug but because it's a code snippet in the docs, I am not sure if it is suitable to use the ordinary bug template here.

danieldegrasse, 2025-04-25:
> It believe it's a bug but because it's a code snippet in the docs, I am not sure if it is suitable to use the ordinary bug template here.

I'd still prefer you did- even if the change is just a doc fix, it is still an issue

closed 2025-08-12 by github-actions[bot]