zephyrproject-rtos/zephyr #88814: west debugserver fails if DEBUG_THREAD_INFO=y

timkruse, 2025-04-19:
**Describe the bug**

Launching the debug-server (openocd) on the Raspberry Pi Pico W fails when the project has `CONFIG_DEBUG_THREAD_INFO` turned on.

Another - but related - problem is when having an OpenOCD debug session ongoing, gdb is able to recognize the threads but it is not able to set breakpoints correctly. I guess this should be a separate bug report ?

**Environment:**

- OS: Windows10
- Toolchain Zephyr SDK 0.17.0
- Zephyr repo `main` (`e41909a32ce` at the time of writing)
- Board: Raspberry Pi Pico W

**To Reproduce**
Tested with the hello_world project

    cd zephyr/samples/hello_world
    west build -p -b rpi_pico/rp2040/w --build-dir build -- -DOPENOCD=/path/to/openocd.exe -DRPI_PICO_DEBUG_ADAPTER=cmsis-dap -DCONFIG_DEBUG_THREAD_INFO=y
    west debugserver

Now OpenOCD shows the error:

> can't read "_TARGETNAME": no such variable

**Workaround**

Start the OpenOCD server manually without west and replace `_TARGETNAME` with `rp2040.core0` in the openocd command.

**Logs and console output**
```
west debugserver
-- west debugserver: rebuilding
[1/10] Performing build step for 'second_stage_bootloader'
ninja: no work to do.
[3/3] Completed 'second_stage_bootloader'
-- west debugserver: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; thread info enabled
Open On-Chip Debugger 0.12.0 (2024-09-16) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 d52e355daa09f17ce64819122cb067b8a2ee0d4b
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
set_adapter_speed_if_not_set
DEPRECATED! use 'tcl port' not 'tcl_port'
DEPRECATED! use 'gdb port', not 'gdb_port'
Warn : Transport "swd" was already selected
Info : Hardware thread awareness created
Info : Hardware thread awareness created
checking adapter speed...
Error: BUG: unknown adapter clock mode
adapter speed: 2000 kHz
can't read "_TARGETNAME": no such variable
FATAL ERROR: command exited with status 1: openocd.exe -s 'D:/documents/programming/cpp-embedded/zephyrproject/zephyr/boards/raspberrypi/rpi_pico\support' -s 'C:\Program Files/OpenOCD/share/openocd/scripts' -f 'D:/documents/programming/cpp-embedded/zephyrproject/zephyr/boards/raspberrypi/rpi_pico\support\openocd.cfg' -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' -c 'source [find interface/cmsis-dap.cfg]' -c 'transport select swd' -c 'source [find target/rp2040.cfg]' -c 'set_adapter_speed_if_not_set 2000' -c '$_TARGETNAME configure -rtos Zephyr' '-c init' '-c targets' -c 'reset init'
```

soburi, 2025-04-19:
It seem not correctly connect to debug adapter.

First, Please try 

```
west debugserver --openocd /usr/local/bin/openocd # or path to your openocd that supports rp2350
```

soburi, 2025-04-19:
The DEBUG_THREAD_INFO is a kconfig property, so `-DCONFIG_DEBUG_THREAD_INFO=y` can't be set correctly.

Please add

```
CONFIG_DEBUG_THREAD_INFO=y
```

to the prj.conf and retry it?

timkruse, 2025-04-20:
Originally, I used the prj.conf to set the variable. Same result. I wrote `-DCONFIG_DEBUG_THREAD_INFO=y` to simplify the issue description, but I didn't think about those being different systems. Sorry for that.

To your earlier comment to use a suitable OpenOCD version: I tried with the latest binary from the official OpenOCD (shown above) as well as a version from the Raspberry Pi Version (built by myself). Both versions are from September 2024. I can imagine a newer version is required. Maybe I have to switch over to linux, too ...

From your comment I presume that debugging with `CONFIG_DEBUG_THREAD_INFO=y` should work ? If so it's clearly a problem with my setup.

soburi, 2025-04-27:
@timkruse 

Could you please try specifying the following options to the west debugserver?

`--target-handle _TARGETNAME_0`

timkruse, 2025-05-07:
Sorry for the late reply. Now it works with `--target-handle _TARGETNAME_0`

Thank you!

closed 2025-05-07 by soburi

power-max, 2026-05-07:
I am experiencing the exact same bug, but with Pi Pico 2. Now as I understand, the official zephyr support for both cores on the RP2350 is lacking at of the time of this writing. I don't know whether that's a relavent detail.

Annoyingly with this option enabled, I could still upload/flash! No obvious errors during build other than some irrelevent warnings about me sloppily forgetting to disable previously used kconfigs.

The UART bridge example would build and I could successfully flash it, but didn't seem to run. The very first couple of printk statements did not work so it didn't seem to be even reaching main(). And in an attempt to get debugger to work I discovered this issue.

With `CONFIG_DEBUG_THREAD_INFO=n` the build succeeded and flashed and the debugger launched and those printk statements worked correctly.