zephyrproject-rtos/zephyr #88818: display: Writing too fast to SDL display with native_sim causes display to stay blank

VynDragon, 2025-04-19:
**Describe the bug**
Writing too fast to SDL display with native_sim causes display to stay blank

**To Reproduce**
Attached file contains a app that reproduces the bug.
Uncommenting the sleep at the start fixes the bug (display goes black instead of staying blank forever)

**Expected behavior**
Display goes black without the sleep.

**Impact**
Annoyance, trap if unknown.

**Environment (please complete the following information):**

 - OS: Arch linux latest
 - Toolchain: Zephyr SDK 17.0
 - Commit 3c2d2a6ee586f2d72d7dbb553c6fa9cecf2d6fb1

[zephyr_sdl_native_sim_race.tar.gz](https://github.com/user-attachments/files/19821908/zephyr_sdl_native_sim_race.tar.gz)

aescolar, 2025-04-22:
CC @Finomnis @faxe1008 @pdgendt @kartben

Finomnis, 2025-04-22:
Does @faxe1008's PR fix your issue?

Finomnis, 2025-04-22:
It's this is thread priority issue? Does the SDL thread have a lower priority than main?

aescolar, 2025-04-22:
> I also tried to change CONFIG_SDL_DISPLAY_THREAD_PRIORITY to 25 and 75 and that didnt fix it either.

The other way around:
https://docs.zephyrproject.org/latest/kernel/services/threads/index.html#thread-priorities
Smaller numbers mean higher priority

VynDragon, 2025-04-22:
ah mb i thought it was centered on 50 for some reason.
It works with -25 with and without the PR.

aescolar, 2025-04-22:
@faxe1008  @Finomnis I think we should just increase the default CONFIG_SDL_DISPLAY_THREAD_PRIORITY to 0
(Note this discussion https://github.com/zephyrproject-rtos/zephyr/pull/87883#discussion_r2023211796 )

Finomnis, 2025-04-22:
How about `CONFIG_SDL_DISPLAY_THREAD_PRIORITY=14`? Does this also fix the issue?

VynDragon, 2025-04-22:
> How about `CONFIG_SDL_DISPLAY_THREAD_PRIORITY=14`? Does this also fix the issue?

It does not, it starts working with `CONFIG_SDL_DISPLAY_THREAD_PRIORITY=-1`

Finomnis, 2025-04-22:
While this feels like a fix for you, I suspect it's actually more of a workaround. Why would the display thread not run at all, even at lower priority?

I mean, if the main thread has a higher priority than the display thread (which is sane imo, display usually is less real time critical than the actual control logic of a system), then of course the display thread does not do anything while main is active. I do not consider this a bug.

On the other hand, if you have the display thread be a very high priority, maybe even a coop priority, then every other task, no matter how low it's own priority is, has the ability to

VynDragon, 2025-04-22:
Or put differently, device_is_ready is false because it is not okay to write to display and breaks the display when done, but true is returned anyway.

VynDragon, 2025-04-22:
> sleeping or yielding in main thread

Actually it doesnt update there either with -1 priority, only when main thread is exited.

Finomnis, 2025-04-23:
> it does not initialize properly and the display write (and future display writes) are not taken into account _at all_.

Yes, exactly. That's why I don't think it's a thread priority issue but a bug somewhere else. Because if thread priority was the issue, the writes would all happen after main is done.

> device_is_ready is false

That sounds like the real issue.

faxe1008, 2025-04-23:
Yes the issue is that the display drivers thread needs to run at least once before reaching the main.
The fix for this is to yield inside of the driver init and change the priority - aside from that there is no other means of
forcing a thread to be scheduled right?

Finomnis, 2025-04-23:
Well you can wait with a semaphore or similar inside of the driver init.

faxe1008, 2025-04-23:
Adjusted the PR, with all the changes.

Finomnis, 2025-04-23:
@VynDragon Does the new version of #88886 fix your problem?

VynDragon, 2025-04-23:
yes it does.

closed 2025-04-25 by kartben