zephyrproject-rtos/zephyr #88819: benchmark.thread_metric.synchronization results deviation observed on mec172xevb_assy6906

golowanow, 2025-04-19:
**Describe the bug**

`mec172xevb_assy6906/mec172x_nsz` board shows results for `benchmark.thread_metric.synchronization` with significantly larger deviation in comparison to what is observed for other platforms.
For example, the benchmark daily probe results for a mec17 board and another board with rather typical statistics for comparison (in the attachments).

**To Reproduce**
Run `tests/benchmarks/thread_metric` several times (10-20) to collect results; each benchmark run records 3 measurements by default. Calculate standard deviation for each run and compare deviations between the runs. The difference between these deviations on mec172 might be quite large e.g. 94% in case of 14/Apr vs. 15/Apr.

**Expected behavior**
A small difference (<10%) between `benchmark.thread_metric.synchronization` run's deviations.

**Impact**
unknown

**Logs and console output**

mec172xevb_assy6906 -
[image]

some other board -
[image]

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v4.1.0-2642-g0648e49bf0f5

peter-mitsis, 2025-04-21:
A preliminary analysis has not yielded anything yet that appears to be obviously wrong in the code that would lead to such results.

One thing that I note is that the timer on this board uses a 32768 Hz clock. I am suspicious of the accuracy of this clock.

peter-mitsis, 2025-05-06:
I've done some more digging, and I am certain that the reported deviation is _not_ related to the thread_metric benchmark. That is, I am certain that it is related to something else that the thread benchmark depends upon.

There are 3 primary parts in play during the benchmark: the threads giving/taking the semaphore, the timer driver and the serial driver. As mentioned earlier, the timer driver is a potential source of variation, but it has been reported since that the board does not appear to show deviation there. Another potential source is the serial driver, which incidentally is based on polling for the board in question (as opposed to leveraging the UART interrupts as seems to be the case for the disco_l475_iot1). Another potential source over which the benchmark has no control/influence is that of caching of data/instructions within the processor--but we simply do not have the visibility within the processor to say how much of an impact this might be.

Recommending to close this issue.

closed 2025-05-06 by peter-mitsis