zephyrproject-rtos/zephyr #88821: Stepper: Callbacks not called when not using a dedicated timer

jbr7rr, 2025-04-19:
**Describe the bug**

When using the stepper API for DRV84xx (and possible others) without the use of a dedicated timer, the callback is not called when the stepper motor has finished turning

**To Reproduce**

In file ``tests/drivers/stepper/drv84xx/api/boards/native_sim.overlay``
Remove or comment out the following line:
```
		// counter = <&counter0>;
```

Run the test:
```
west twister -T tests/drivers/stepper/drv84xx/api/ 
```

The test will fail on the callbacks

**Expected behavior**

Callbacks called in the same manner as when using a dedicated timer.

**Impact**

ESP32 does not have a suitable timer for the stepper API, so on that platform there is no way to make the stepper motor work with the current API

**Logs and console output**

[handler.log](https://github.com/user-attachments/files/19822356/handler.log)

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

OS: Manjaro
Toolchain: Zephyr SDK
Main branch @ 0648e49bf0f565f24f31eb5b61ac69e971f38d28

**Additional context**

jilaypandya, 2025-04-19:
> ESP32 does not have a suitable timer for the stepper API, so on that platform there is no way to make the stepper motor work with the current API

there is an alternative work_q based approach which is already part of the driver subsystem, it should be used per default if counter is not used. We will look into it.

Ping @faxe1008

jbr7rr, 2025-04-20:
> there is an alternative work_q based approach which is already part of the driver subsystem, it should be used per default if counter is not used. We will look further into it.

Yeah, but when using the work_q based approach the callback doesn't get called, I had a look and there is a problem that the work does not get rescheduled.

This fixes the issue:

```
diff --git a/drivers/stepper/step_dir/step_dir_stepper_common.c b/drivers/stepper/step_dir/step_dir_stepper_common.c
index bd4eac4b996..b6b7566f07f 100644
--- a/drivers/stepper/step_dir/step_dir_stepper_common.c
+++ b/drivers/stepper/step_dir/step_dir_stepper_common.c
@@ -145,11 +145,11 @@ static void position_mode_task(const struct device *dev)
 		(void)step_dir_stepper_perform_step(dev);
 	}
 
-	update_remaining_steps(dev->data);
-
 	if (config->timing_source->needs_reschedule(dev) && data->step_count != 0) {
 		(void)config->timing_source->start(dev);
 	}
+
+	update_remaining_steps(dev->data);
 }
 
 static void velocity_mode_task(const struct device *dev)
```

I'll try to write a proper PR