zephyrproject-rtos/zephyr #88822: Code duplication between drivers

zbas, 2025-04-19:
How should it be handled? Is there a location where code common to multiple drivers should go?

I am asking because get_tim_clk in the stm32 drivers is duplicated in the counter and pwm drivers and it has been ackowledged that this should be addressed.

etienne-lms, 2025-05-27:
Since this is purely related to STM32 clock management,maybe a helper function declared in **include/zephyr/drivers/clock_control/stm32_clock_control.h** and defined in a new source file **drivers/clock_control/clock_stm32_tim.c** could do the job?

alexapostolu, 2025-05-28:
The only difference is `counter_ll_stm32_timer.c` uses `DT_HAS_COMPAT_STATUS_OKAY(st_stm32f0_rcc)` while the other file `pwm_stm32.c` uses `!defined(CONFIG_SOC_SERIES_STM32C0X) && !defined(CONFIG_SOC_SERIES_STM32F0X) &&  !defined(CONFIG_SOC_SERIES_STM32G0X) && !defined(CONFIG_SOC_SERIES_STM32U0X)` in their `#if` statements.

Are those two the same condition?