zephyrproject-rtos/zephyr #88831: STM32WLE5 UART not able receiving data when enabled CONFIG_PM

shiwzhi, 2025-04-20:
**Describe the bug**

On RAK3172 target, when enable CONFIG_PM configs, Interrupt-driven UART  can not receive data properly.

**To Reproduce**

```
CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
```

```
/ {
	aliases {
		pms-uart = &usart1;
	};
};

&cpu0 {
	device_type = "cpu";
	compatible = "arm,cortex-m4";
	reg = <0>;
	cpu-power-states = <&stop0 &stop1>;
};

&usart1 {
	current-speed = <9600>;
	status = "okay";
	pinctrl-0 = <&usart1_tx_pb6 &usart1_rx_pb7>;
	pinctrl-names = "default";
};
```
```
uint8_t rx_buf[100];
uint8_t rx_len = 0;
void uart_callback(const struct device *dev, void *user_data)
{
    uint8_t c;

    if (!uart_irq_update(dev))
    {
        return;
    }

    if (!uart_irq_rx_ready(dev))
    {
        return;
    }

    while (uart_fifo_read(dev, &c, 1) == 1)
    {
        if (rx_len < 50)
        {
            rx_buf[rx_len] = c;
            rx_len++;
        }
    }
}

int uart_init()
{
    int ret;
    uart_dev = DEVICE_DT_GET(DT_ALIAS(uart1));

    if (NULL == uart_dev)
    {
        return -1;
    }
    if (!device_is_ready(uart_dev))
    {
        return -1;
    }

    ret = uart_irq_callback_user_data_set(uart_dev, uart_callback, NULL);
    if (0 != ret)
    {
        return -1;
    }

    uart_irq_rx_enable(uart_dev);

    return 0;
}
```

**Expected behavior**

The UART can work normally when using STM32 STOP1 or STOP2 mode

**Impact**

This prevent me to use UART based sensor in low power mode.

**Logs and console output**

Without low power mode 
```
[00:25:11.127,000] <dbg> Sensor: uart_out: UART TX:
                              42 4d e2 00 00 01 71                             |BM....q

[00:25:13.132,000] <dbg> Sensor: uart_out: UART RX:
                              42 4d 00 1c 00 09 00 13  00 18 00 09 00 13 00 18 |BM...... ........
                              02 e3 02 82 00 6d 00 00  00 00 00 00 13 00 02 fc |.....m.. ........
```
After enable
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
```
[00:01:22.612,000] <dbg> PMS: pms_uart_out: PMS TX: 
                              42 4d e2 00 00 01 71                             |BM....q          
[00:01:24.618,000] <dbg> PMS: pms_uart_out: PMS RX: 
```

**Environment (please complete the following information):**

 - OS: Mac OS
 - Toolchain Zephyr 4.1.0
 - Commit SHA or Version used 7823374e872145b5bd018bfe447839eb36042611

**Additional context**

mickael868, 2025-04-24:
Did you take a look at the `boards/st/power_mgmt/serial_wakeup` sample? In this sample, the usart1 dts node features the `wakeup-source` and `fifo-enable` properties and the `main.c` enables the wakeup feature.

shiwzhi, 2025-04-25:
> Did you take a look at the `boards/st/power_mgmt/serial_wakeup` sample? In this sample, the usart1 dts node features the `wakeup-source` and `fifo-enable` properties and the `main.c` enables the wakeup feature.

```
&usart1 {
    current-speed = <9600>;
    status = "okay";
    pinctrl-0 = <&usart1_tx_pb6 &usart1_rx_pb7>;
	pinctrl-names = "default";

	clocks = <&rcc STM32_CLOCK(APB2, 14U)>,
                 <&rcc STM32_SRC_HSI USART1_SEL(2)>;
	wakeup-source;
	fifo-enable;
};

&clk_hsi {
	/* Make sure HSI is enabled */
	status = "okay";
};
```
```
    bool ret = pm_device_wakeup_is_capable(uart_dev);
    if (!ret)
    {
        printk("Device is not wakeup capable\n");
    }
    else
    {
        printk("Device is wakeup capable\n");

        ret = pm_device_wakeup_enable(uart_dev, true);
        if (!ret)
        {
            printk("Could not enable wakeup source\n");
        }
        else
        {
            printk("Wakeup source enable ok\n");
        }

        ret = pm_device_wakeup_is_enabled(uart_dev);
        if (!ret)
        {
            printk("Wakeup source not enabled\n");
        }
        else
        {
            printk("Wakeup source enabled\n");
        }
    }
```
I tried this, now I can receive data but only the first 8 bytes
```
[00:10:26.593,000] <dbg> Sensor: uart_out: UART TX: 
                              42 4d e2 00 00 01 71                             |BM....q          
[00:10:28.593,000] <dbg> Sensor: uart_out: UART RX: 
                              42 4d 00 1c 00 00 00 00                          |BM......  
```

erwango, 2025-04-25:
Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/87859 has an impact on your issue ?

shiwzhi, 2025-04-25:
> Can you check if recently merged [#87859](https://github.com/zephyrproject-rtos/zephyr/pull/87859) has an impact on your issue ?

Sorry, I'm not sure how to set up DMA and use Async UART
But It seems I can use this to prevent low power mode while waiting for the RX data.
```
pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, PM_ALL_SUBSTATES);
k_msleep(1000);
pm_policy_state_lock_put(PM_STATE_SUSPEND_TO_IDLE, PM_ALL_SUBSTATES);
```

closed 2025-04-25 by shiwzhi