zephyrproject-rtos/zephyr #88850: tests: boot: test_mcuboot: fail on spi nor devices platfrom

hakehuang, 2025-04-21:
**Describe the bug**

tests/boot/test_mcuboot fails on all NXP spi not like platforms

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - all NXP spi nor flash platforms
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
```
efd4f93045789f1fc7a82aa1fc5a3fed6d0b2e03 is the first bad commit
commit efd4f93045789f1fc7a82aa1fc5a3fed6d0b2e03
Author: Jamie McCrae <jamie.mccrae@nordicsemi.no>
Date:   Wed Apr 9 09:40:19 2025 +0100

    west.yml: MCUboot synchronization from upstream

    Update Zephyr fork of MCUboot to revision:
      20f98e0a975c24864872e0df5701eb1082e9c957

      - 20f98e0a Updates for 2.2.0-rc1 release
      - 7a33bcaa boot: bootutil: loader: Add include for flash
        function
      - 8a9627d9 boot/zephyr/boards: nrf54h20dk 'iron' board
        configuration
      - de7a9dc5 zephyr: Disable SPI_NOR by default for
        nrf7002dk/nrf5340/cpuapp
      - 5aed4245 swap_move: correct appsize calcs and warning
      - 454c033f bootutil: swap-scratch: Fix conflicts after support
        of devices w/o erase
      - 572b8fb2 boot: zephyr: flash_map_extended: Add pointless
        workaround for clang
      - 091af82e boot/zephyr: nrf54h20dk adaptations
      - ed434f33 boot: bootutil: Fix clash of STRUCT_PACKED definition
      - 33ad4973 boot: zephyr: add support for mcx_n9xx_evk
      - 881608eb boot/zephyr: Use MCUBOOT_RAM_LOAD instead of
        CONFIG_BOOT_RAM_LOAD
      - 993c2ff8 boot/bootutil/ram_load: Add flash area pointer to
        bootloader state
      - 8b7c7cc9 zephyr: flash map: fix comparison signedness
      - b92d4dd7 boot: zephyr: Only supply MBEDTLS config file path
        if set
      - 9fa0c6b5 zephyr: Fix translation of PSA Kconfig to MCUboot
        config
      - 2d681da4 bootutil: Fix ed25519 pure signature verification

    Signed-off-by: Jamie McCrae <jamie.mccrae@nordicsemi.no>

 west.yml | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

```

**To Reproduce**

```
scripts/twister --device-testing --hardware-map /home/ubuntu/nxp/mimxrt1040_evk/map.yaml -T tests/boot/test_mcuboot/ --west-flash
```

**Expected behavior**

Run pass

**Impact**

mcuboot

**Logs and console output**

```
*** Booting MCUboot v2.1.0-r1-294-g20f98e0a975c ***
*** Usng Zephyr OS build v4.1.0-2646-e41909a32ce9 ***
c: Starting bootloader
*** Booting MCUboot v2.1.0-r1-294-g20f98e0a975c ***
*** Usng Zephyr OS build v4.1.0-2646-e41909a32ce9 ***
c: Starting bootloader
*** Booting MCUboot v2.1.0-r1-294-g20f98e0a975c ***
*** Usng Zephyr OS build v4.1.0-2646-e41909a32ce9 ***
c: Starting bootloader
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-2654-gf087aa264e9

hakehuang, 2025-04-21:
@butok

henrikbrixandersen, 2025-04-21:
CC: @nordicjm

butok, 2025-04-22:
Not verified yet, but this might be caused by: https://github.com/mcu-tools/mcuboot/commit/5aed424531d178ad5a678ee0a760e8bc2c234a0e

katgiadla, 2025-05-08:
Issue does not occurs more after fix on NRF platforms.

butok, 2025-05-13:
> Issue does not occur more after fix on NRF platforms.

Hi @katgiadla 
Could you point to the NRF fix?

katgiadla, 2025-05-13:
> [mcu-tools/mcuboot#2266 (comment)](https://github.com/mcu-tools/mcuboot/pull/2266#discussion_r2055691842) => To be fixed by [mcu-tools/mcuboot@efa3039](https://github.com/mcu-tools/mcuboot/commit/efa30399324b51fa8defe587b9c220a0b56a9cdb), but not fixed.

like below ^

closed 2025-05-19 by kartben