zephyrproject-rtos/zephyr #88882: posix: Without configuring POSIX_API, it is impossible to configure POSIX_TIMER

Freey0, 2025-04-22:
**Describe the bug**

Without configuring POSIX_API, it is impossible to configure POSIX_TIMER,  
Error: 'CLOCK_REALTIME' undeclared.

**To Reproduce**

```
#include <time.h>

int main(void)
{
    struct timespec tp;
    clock_gettime(CLOCK_REALTIME, &tp);
	return 0;
}
```
```
# CONFIG_POSIX_API=y
CONFIG_POSIX_AEP_CHOICE_PSE51=y
CONFIG_POSIX_TIMERS=y
```

**Expected behavior**

**Impact**

**Logs and console output**

/home/zephyr/src/main.c: In function 'main':
/home/zephyr/src/main.c:12:5: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
   12 |     clock_gettime(CLOCK_REALTIME, &tp);
      |     ^~~~~~~~~~~~~
/home/zephyr/src/main.c:12:19: error: 'CLOCK_REALTIME' undeclared (first use in this function)
   12 |     clock_gettime(CLOCK_REALTIME, &tp);
      |                   ^~~~~~~~~~~~~~
/home/zephyr/src/main.c:12:19: note: each undeclared identifier is reported only once for each function it appears in

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows) linux
 - Toolchain (e.g Zephyr SDK, ...)  zephyr 0.17.0 
 - Commit SHA or Version used v4.1.0-2646-ge41909a32ce9

**Additional context**

cfriedt, 2025-04-22:
When using CONFIG_POSIX_API, the zephyr/posix/time.h header is pulled in first, which declares CLOCK_REALTIME, clock_gettime, etc unconditionally.

My initial take is that the build is picking up  the time.h from picolibc here.

https://github.com/picolibc/picolibc/blob/main/newlib%2Flibc%2Finclude%2Ftime.h

The picolibc time.h header has an internal preprocessor macro `__POSIX_VISIBLE` that is set based on the existence of application conformance macros, IIRC.

cfriedt, 2025-04-22:
There is another Kconfig option that should be added to include Zephyr POSIX headers on the default search path. That can also just be done with cmake in the application. Another option is to define application conformance macros, so that picolibc declares POSIX definitions.

There was a PR to do that in a systematic way over a year ago that received a lot of abusive comments and hasn't been revisited in a while.

cfriedt, 2025-04-22:
I think it would be nice to deprecate POSIX_API shortly though, and this will need to be addressed.

Just updating the priority - there is already work in progress.

cfriedt, 2025-06-18:
I think my preference here is to move the POSIX `time.h` declarations to the minimal libc (with a bit of a rewrite), and use Kconfig / CMake to declare the standard and necessary `_POSIX_TIMERS` feature test macro when it has been appropriately configured.

cfriedt, 2025-06-18:
Oh, actually - I think that this has probably already been fixed (in a different way). @Freey0 - perhaps you would prefer to use `sys_clock_gettime()` in #90096 ?

It's a bit of a conundrum to want to use POSIX calls without wanting to configure POSIX. Newlib, picolibc, minimal libc, etc, all intermix usage of posix and non-posix code. For the simple use case of getting monotonic or wall-clock time, #90096 added `sys_clock*()` apis that would remove the dependency cycle, so I would suggest using that instead.