zephyrproject-rtos/zephyr #88899: Retrive net_stats for ethernet vlan cause assert trap in Arm target

linuxemb, 2025-04-22:
**Describe the bug**

**Logs and console output**
<ASSERTION FAIL [iface] @ WEST_TOPDIR/zephyr/subsys/net/ip/net_stats.h:38
**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (Zephyr 3.7
 - Commit SHA or Version used

linuxemb, 2025-04-22:
Repro steps:
1 Enable config in prj file 

#define CONFIG_NET_STATISTICS
#define CONFIG_NET_STATISTICS_ETHERNET
#define CONFIG_NET_MGMT
#define CONFIG_NET_MGMT_EVENT
2 call this in c file 

	if(net_if_l2(iface) == &NET_L2_GET_NAME(ETHERNET)) {
		net_mgmt(NET_REQUEST_STATS_GET_ETHERNET, iface, &stats, sizeof(stats));
	}
3 --tested on my Fpga board running on A55

carlescufi, 2025-05-30:
@linuxemb is this still an issue? Could you please help out with a reproducer?