zephyrproject-rtos/zephyr #88902: stm32: support qdec for all stm32 devices

nono313, 2025-04-22:
**Is your enhancement proposal related to a problem? Please describe.**
Currently only the stm32f4 has qdec peripheral in its timers. This enhancement is to add it to all timers that support it on all stm32 mcus.

**Describe the solution you'd like**
Add 'st,stm32-qdec' node on timers that support it on all stm32 platforms

**Describe alternatives you've considered**
I tried adding the node manually for the stm32h753 I have on my devicetree overlay and it works, but I still think it should be present (as disabled) by default on mcu's devicetrees.

**Additional context**
N/A

erwango, 2025-04-23:
@nono313 Thanks for raising this issue, but we're currently busy with multiple other topics. But if you managed to do it on your end, don't hesitate to share you work and push it as a PR, we'll be glad to help you in the process.

erwango, 2025-11-19:
@Majistor Sure, you can complete the work that was started with https://github.com/zephyrproject-rtos/zephyr/pull/95624.
So remains: STM32C0/F0/F4/G4/H5/H7RS/N6/U0/U3/WB/WBA/WB0/WL/MPX.
Do you have the HW required to test those series?

mt-tass, 2026-03-04:
@erwango  can i take this up for stm32f4

akansh-sinha, 2026-03-24:
@erwango I put together a draft PR (#106158) for STM32U0 QDEC support. I only enabled it for TIM1-TIM3 since the reference manual (RM0503) shows those are the only ones with the hardware interface. Let me know if this approach works or if I need to change anything!