zephyrproject-rtos/zephyr #88925: JEDEC: devicetree cannot describe valid flash chip sizes

JordanYates, 2025-04-23:
**Is your enhancement proposal related to a problem? Please describe.**

The compile time JEDEC size parameter in devicetree is specified in bits:
https://github.com/zephyrproject-rtos/zephyr/blob/6463c68bc394e11beeb7ef0db9afcead9fb3aed2/dts/bindings/mtd/jedec%2Cjesd216.yaml#L25-L27
Devicetree properties are limited to unsigned 32 bits numbers:
https://docs.zephyrproject.org/latest/build/dts/intro-syntax-structure.html#properties
Attempts to specify a value greater than or equal to `2**32` results in errors like the following:
```
devicetree error: /home/jordan/code/workspace/folder/boards/board.dtsi:160 (column 11): parse error: 4294967296 does not fit in 32 bits
```
Finally, SPI flash devices tend to come in sizes that are powers of two, e.g. 64 Mbit, 128 Mbit, etc.

Combining these limitations together, we end up with a hard limit on the size of a flash chip that can be described.
Max size in bits = `2**31` (`2**32` errors out)
Max size in Mbits = `2**31/1024/1024` = 2048 Mbit

This is substantially smaller than the size supported by the flash API:
 * `flash_get_size` returns a size in `uint64_t` bytes
 * `flash_read/flash_write` take a `off_t` byte address (`uint32_t` on a Cortex-M33)

So while the flash API can handle flash chips up to 32 Gbit in size, the devicetree binding is limited to 2 Gbit.
This is only *just* large enough to support the largest current SPI NOR flash chips:
https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash?__locale=en

[image]

**Describe the solution you'd like**

It should be possible for the JEDEC bindings to describe flash chips all the way up to `2**32` bytes large (`uint32_t` address space).

de-nordic, 2025-05-08:
Nordic QSPI binding has this:
https://github.com/zephyrproject-rtos/zephyr/blob/af77efb77e912e109e8a9326c0b522c5058d9218/dts/bindings/mtd/nordic%2Cqspi-nor.yaml#L25-L28

I guess we could do something like that or maybe store exponent.

Regarding API returned values: we need system wide storage type that can be switched between 32 and 64 bit, depending on whether user is aware of devices that need larger size and not by the architecture register size; I understand that using 64bit on 32bit arch is not super efficient, but you can still connect memory of size > 4GiB to device with 32bit registers. I think @cfriedt tried to provide such definitions, but the number of changes has spawn huge PR and somehow that stalled the transition.

Note also that I am quite not happy with the `off_t` used by any storage API that is not file system call, as literally none of device can handle negative values, and they literally loose half of the type range.
File system types are something else, as you may be willing to seek into negative range of your current position, but still you will not write or read from negative range of `off_t` (you probably could, but API does not support this).

JordanYates, 2025-09-29:
> Since the range of uint32 goes from `0 - 4294967295`, maybe you need `int64`.

Good idea in theory, but:

> Devicetree properties are limited to unsigned 32 bits numbers:
https://docs.zephyrproject.org/latest/build/dts/intro-syntax-structure.html#properties

cfriedt, 2025-09-29:
@de-nordic - storing the exponent is a good idea. I've generally never seen a storage device that isn't a power of two in size. Maybe it could happen though.

Couldn't we also use a property like `#address-cells` to determine if a storage node has 32-bits or 64-bits of addressing bits?

Still, it's subtly annoying that spi nor capacities are described in bits. I can understand from a manufacturing standpoint, because that's a standard, but from a consumer perspective people might prefer bytes.