zephyrproject-rtos/zephyr #88929: [Kernel][Initialization] MSP and PSP Stack Conflict in z_interrupt_stacks Causes Memory Corruption

FixJA, 2025-04-23:
**Describe the bug**
During early Zephyr initialization, the MSP (Main Stack Pointer) is set to the top of z_interrupt_stacks. However, the PSP (Process Stack Pointer) used by z_cstart already consumes part of z_interrupt_stacks. When an interrupt is triggered during z_sys_init_run_level(INIT_LEVEL_PRE_KERNEL_2), the MSP stack overwrites the PSP stack content, corrupting local variables in z_sys_init_run_level. This leads to memory out-of-bounds access and a chip exception.

here is my debug infomation

[image]
bufferIndex initial value is 1
[image]
An interrupt triggers the use of an MSP stack pointer, and the MSP pointer points to something that is already in use by the PSP pointer, causing the PSP stack to be corrupted
[image]
bufferIndex value is charged
[image]

[image]

**Environment (please complete the following information):**

 - OS: linux
 - Toolchain Zephyr SDK 0.17  
 - Zephyr OS build v4.1.0-2729-g785d0bbfb3fc
 - board: mimxrt1020_evk

**Additional context**
prj.conf
```
# Copyright (c) 2021 Nordic Semiconductor ASA
# SPDX-License-Identifier: Apache-2.0
#
# This file contains selected Kconfig options for the application.

# zephyr printf
CONFIG_SERIAL=y
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_PRINTK=y
CONFIG_INIT_STACKS=y
CONFIG_SHELL=y
CONFIG_CONSOLE_SUBSYS=y
CONFIG_SHELL_BACKEND_SERIAL=y
CONFIG_REQUIRES_FULL_LIBC=y
CONFIG_POSIX_API=y

CONFIG_GPIO=y
CONFIG_BLINK=y

#network
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_ARP=y
CONFIG_NET_TCP=y
CONFIG_NET_UDP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_DHCPV4_OPTION_CALLBACKS=y
CONFIG_NET_CONTEXT_RCVTIMEO=y
CONFIG_NET_CONTEXT_SNDTIMEO=y
CONFIG_DNS_RESOLVER=y
CONFIG_NET_TCP_MAX_RECV_WINDOW_SIZE=5120

CONFIG_NET_PKT_TX_COUNT=6
CONFIG_NET_BUF_RX_COUNT=64

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_CONNECTION_MANAGER_MONITOR_STACK_SIZE=2048
CONFIG_SLIP_STATISTICS=n

# shell
CONFIG_NET_SHELL=y
CONFIG_SHELL_PROMPT_UART="rt1020:~$ "
CONFIG_SHELL_BACKEND_SERIAL_TX_RING_BUFFER_SIZE=128
CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE=256

CONFIG_MAIN_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_NET_TCP_WORKQ_STACK_SIZE=4096
CONFIG_NET_RX_STACK_SIZE=4096
CONFIG_NET_SOCKETS_SERVICE_STACK_SIZE=2048
CONFIG_NET_MGMT_EVENT_STACK_SIZE=2048
CONFIG_ISR_STACK_SIZE=4096

```
debug.conf
```
# compiler
CONFIG_NO_OPTIMIZATIONS=y

# logging
CONFIG_LOG=y
CONFIG_NET_LOG=y
CONFIG_APP_LOG_LEVEL_DBG=y
CONFIG_GPIO_LOG_LEVEL_DBG=y
CONFIG_ETHERNET_LOG_LEVEL_DBG=y
#debug
CONFIG_THREAD_MONITOR=y

CONFIG_STACK_SENTINEL=y
CONFIG_ASSERT=y

CONFIG_TRACING=y
CONFIG_PERCEPIO_TRACERECORDER=y
CONFIG_PERCEPIO_TRC_START_MODE_START_FROM_HOST=y
CONFIG_PERCEPIO_TRC_CFG_STREAM_PORT_RTT=y
CONFIG_PERCEPIO_TRC_CFG_CTRL_TASK_STACK_SIZE=4096
```

danieldegrasse, 2025-04-29:
@FixJA do you have any other platforms running Zephyr that also exhibit this issue?

FixJA, 2025-05-03:
I only have iMXRT series boards on hand, and this is my first time using Zephyr. I’ve noticed that this issue occurs when multiple component features are enabled.

danieldegrasse, 2025-05-12:
> I only have iMXRT series boards on hand, and this is my first time using Zephyr. I’ve noticed that this issue occurs when multiple component features are enabled.

I see you're using Ozone as a debugger- can you reproduce the issue outside of Ozone? I know there are have previously been issues with the default reset strategy Ozone uses that did not initialize stack pointers properly

FixJA, 2025-05-13:
I checked the Ozone configuration file, and the initial stack settings were problematic. I reviewed the MAP file to set the correct stack pointer, but the issue persisted. When I downloaded the program and reset the chip without connecting the debugger, the program still did not work properly.

[image]

dleach02, 2025-06-11:
@andyross @danieldegrasse 

I did a bit of digging on this since it was assigned to me. @FixJA has identified an issue. Here is the flow from reset:

- reset.S: 
  - msp = z_main_stack
  - psp = z_interrupt_stacks
  - processor is switched to use psp for Thread mode
  - z_prep_c:
    - bss_zero/data_copy
    - z_arm_interrupt_init
    - z_cstart:
      - z_sys_init_run_level(INIT_LEVEL_EARLY)
      - arch_kernel_init()
        - z_arm_interrupt_stack_setup
          - **msp = z_interrupt_stack  <-- now both msp and psp are pointing to the same stack area**
      - z_sys_init_run_level(INIT_LEVEL_PRE_KERNEL_1)
      - z_sys_init_run_level(INIT_LEVEL_PRE_KERNEL_2)
      - switch_to_main_thread(prepare_multithreading()) <--- sets PSP to thread stack which is actually z_main_stack

After the call to _**z_arm_interrupt_stack_setup**_ we find the system in a state of msp & psp both pointing to the z_interrupt_stacks. The looping calls of z_sys_init_run_level() for PRE_KERNEL_1 and PRE_KERNEL_2 now have their psp based stack exposed to interrupt handling.

Couple of questions:

1. Is there an expectation of interrupt enable status in Zephyr's initialization sequence?
    - I say this because I see we have an explicit call to arch_irq_unlock_outlined() in arch_switch_to_main_thread() so I'm wondering if we expect no interrupts to occur in the system until we get to this point (which would mean something in the PRE_KERNEL_1/2 init enabled interrupts). If it is the case that we don't expect interrupts to be enabled during this phase then we need to investigate each of the initialization routines in LEVEL_EARLY/PRE_KERNEL_1/2 to see if any of them enable interrupts.

2. Couldn't the psp be initialized in reset.S to z_main_stack? The z_main_stack won't be reused until switch_to_main_thread() after the prepare_multithreading() sets it to bg_thread_main thread. Code execution is not expected to return to the code after the call to prepare_multithreading() so call stack at this point is not needed. 

> I changed the location of the interrupt stack initialization code, and the program started working properly.

@FixJA patch above moving the z_arm_interrupt_stack_setup() to before the call of switch_to_main_thread() works in this situation because it allowed the interrupt handling to use the pre-configured msp z_main_stack, which is unused until switch_to_main_thread(). But the system is still exposed to the issue because you have to make a call to prepare_multithreading() inside the call to switch_to_main_thread().

dleach02, 2025-06-12:
We can't setup psp to z_main_stack because in the prepare_multithread() it would do some initialization of the stack area if **CONFIG_INIT_STACKS** is enabled. That will blow away the return stack information and is probably one of the reasons the initialization flow configures the msp/psp stacks as such.

Per @FixJA debug information the interrupt occurring is the cortex_m_systick ISR. Of course it is firing before the Kernel is even running. We could consider changing the CONFIG_SYSTEM_CLOCK_INIT_PRIORITY to 255 to force it to be the last PRE_KERNEL_2 module initialized. This would be a Band-Aid because potential problem is still there but window reduced.

Another option is to create a new stack that is only used for the initialization flow.

wearyzen, 2025-06-18:
Hi All,

- As per below line the expectation is that all IRQs with configurable priority should be masked during iqr lock: https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/doc/hardware/arch/arm_cortex_m.rst?plain=1#L282 
and irq lock is what is done during initialization.
 
- interrupts are locked (masked) here during initialization with BASEPRI set to _EXC_IRQ_DEFAULT_PRIO (which is Z_EXC_PRIO(_IRQ_PRIO_OFFSET)) (32 in most cases): https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/arch/arm/core/cortex_m/reset.S#L157

- interrupts are unlocked (unmasked) here during initialization by setting BASEPRI to 0: https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/arch/arm/core/cortex_m/thread.c#L550

- priority of SysTick is set to _IRQ_PRIO_OFFSET which is same as the BASEPRI mask _EXC_IRQ_DEFAULT_PRIO (32): https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/drivers/timer/cortex_m_systick.c#L526
**So ideally the systick interrupt should have been masked.**

- But between the original position of z_arm_interrupt_stack_setup and the one in kernel/init.c that fixed the issue for @FixJA, there is a lot of code with soc hooks etc. which could have either changed priority of the systick irq to higher than the default BASEPRI mask or changed the BASEPRI itself. In all the images show the BASEPRI is shown as 0x1000 and changes to 0x2000 in the timer isr and this is different from what I expected.
@FixJA could you please add a breakpoint at below line and confirm what is the value of BASEPRI after this?https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/arch/arm/core/cortex_m/reset.S#L158 
and if possible could you also confirm what was the priority set for the systick irq in your setup?

Now coming to the issue of MSP and PSP pointing to the same stack. This could be an issue in such cases and if there is a fault that needs to be handled during the initialization phase so for this we can try below:
1. Reduce the amount of time the stack is shared by moving z_arm_interrupt_stack_setup() to arch_switch_to_main_thread() [here](https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/arch/arm/core/cortex_m/thread.c#L546) just before arch_irq_unlock_outlined(). Both the functions are internal to arm so logic stays in arch specific files and the shared time would be very small. The con here is that since MSP is pointing to the main stack, if there is an interrupt before the switch is done, it will corrupt the 0xaa init pattern (done as part of the CONFIG_INIT_STACK) of the main stack.

or
 
2. As @dleach02 suggested, not sharing the stack and setting PSP to a new_initialization_stack to be used just during the initialization phase. This would mean we waste some memory since the new_initialization_stack will never be used later but this would also simplify the code and make it easier to debug the next time. Further, if CONFIG_INIT_STACKS is enabled PSP could be set to new_initialization_stack and if not it can be set directly to z_main_stack.

FixJA, 2025-06-23:
@wearyzen Hi, this is my debug infomation
> - priority of SysTick is set to _IRQ_PRIO_OFFSET which is same as the BASEPRI mask _EXC_IRQ_DEFAULT_PRIO (32): https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/drivers/timer/cortex_m_systick.c#L526

[image]

> - But between the original position of z_arm_interrupt_stack_setup and the one in kernel/init.c that fixed the issue for @FixJA, there is a lot of code with soc hooks etc. which could have either [...]
@FixJA could you please add a breakpoint at below line and confirm what is the value of BASEPRI after this?https://github.com/zephyrproject-rtos/zephyr/blob/a626d865c0b27d583b92ce359ad484d86aabdce1/arch/arm/core/cortex_m/reset.S#L158
and if possible could you also confirm what was the priority set for the systick irq in your setup?

[image]

wearyzen, 2025-06-23:
Hi @FixJA thank you for the info.
The screenshot shows the BASEPRI set in reset.S is 0x1000, this is a bit confusing because BASEPRI should only be having 8 bit values on CortexM7.
If for some reason your IDE shows 0x10 as 0x1000 then its fine (confusing but fine) because then this matches with the priority set in the systick driver for SysTick_IRQn. But if BASEPRI is actually set to 0x1000 then it basically means the interrupt masking is disabled because 0-7 bits of BASEPRI are 0.

The other issue seen from your earlier screenshot is that the BASEPRI is 0x2000 at the entry of the interrupt handler which means BASEPRI was changed between reset.S and the interrupt handler by some unknown code. Going by the earlier assumption that 0x1000 was 0x10 and 0x2000 is 0x20, if BASEPRI is change to 0x20 then all the interrupts having priority lower than this will be enabled during kernel initialization (including the systick) and can cause the corruption you reported.

I would suggest to temporarily remove the hack you have and capture the value of BASEPRI at reset.S and at systick interrupt handler, if they are not same then further debug which part of code changed this value.
Also, it would be helpful to confirm if BASEPRI is actually 0x1000 or if the IDE is incorrectly showing 0x10 as 0x1000.

peter-mitsis, 2025-08-12:
@FixJA - Any updates on @wearyzen's suggested next steps?

mathieuchopstm, 2025-09-19:
How about reserving part of the interrupt stack (configurable via Kconfig) as available during boot for interrupts? Such that:
https://github.com/zephyrproject-rtos/zephyr/blob/27fa72163203fa3e2d883cdbfcc1fb7ba469e6ed/arch/arm/core/cortex_m/reset.S#L115

would become
```armasm
 ldr r0, =z_interrupt_stacks + CONFIG_ISR_STACK_SIZE + MPU_GUARD_ALIGN_AND_SIZE - CONFIG_INIT_ISR_STACK_SAFEZONE
```
leaving `CONFIG_INIT_ISR_STACK_SAFEZONE` bytes free in case an interrupt is taken.

Another thing that could be done is, for platforms that have `BASEPRI`, to still use `PRIMASK` to protect from interrupts during initialization. This doesn't protect from code that clears `PRIMASK`, but such a thing would have to be done through ASM/CMSIS because there still wouldn't be any Zephyr API that would do that (i.e., `irq_{un}lock` continues to use `BASEPRI`). This could potentially allow the kernel to survive until right before interrupts would be unmasked, and we could check that `BASEPRI` is as expected, warning/`panic`ing if not. The `_isr_wrapper` could also `panic()` if an interrupt is taken when `k_is_pre_kernel()`, though that would make ISR latency worse and thus kinda sucks.

But really, this here seems to be a programming error somewhere: the best we can do is add band-aids, but that won't fix the underlying issue (which isn't to say having a first-aid kit - gated behind Kconfig - to help diagnose such issues is bad, just that it shouldn't be considered as the "main solution" imo)