zephyrproject-rtos/zephyr #89031: PM device refactoring

bjarki-andreasen, 2025-04-24:
## Introduction
This issue documents and tracks the changes needed to refactor PM device. The aim is to move from a complex mix of strategies, namely:
* PM_DEVICE only
* PM_DEVICE_SYSTEM_MANAGED
* PM_DEVICE_RUNTIME
* PM_DEVICE_SYSTEM_MANAGED + PM_DEVICE_RUNTIME

to:

* PM_DEVICE_RUNTIME
* PM_DEVICE_SYSTEM_MANAGED

and lastly to:

* PM_DEVICE_RUNTIME

## Key changes to user experience

The following key APIs will be hidden from the user:

* `pm_device_action_run()`
* `pm_device_state_get()`
* `pm_device_runtime_enable()`
* `pm_device_busy_set()`

These APIs will solely be called by either PM_DEVICE_RUNTIME or PM_DEVICE_SYSTEM_MANAGED.

Users will instead call:

* `pm_device_runtime_get()`
* `pm_device_runtime_put()`
* `pm_device_runtime_put_async()`

if PM_DEVICE_RUNTIME is selected. In case of PM_DEVICE_SYSTEM_MANAGED, devices are managed by the system state alone.

Most importantly, the devicetree property `zephyr,pm-device-runtime-auto` will be removed, users will no longer need to add this property manually to devices to change their behaviors individually. if PM_DEVICE_RUNTIME is enabled, this will apply to all devices, no need for the flag, while achieving consistent behavior.

## Key changes to driver developer experience

The following key APIs will be hidden from the developer:

* `pm_device_action_run()`
* `pm_device_init_suspended()`

and the following API will be removed entirely:

* `pm_device_runtime_enable()`

Device drivers will only need to implement the pm action hook, and call `pm_device_driver_init()` as the last thing before leaving the device driver init function. Which state the device enters on boot, will be entirely managed by PM device calling the pm action hook implementation.

Device drivers which depend on other devices, will call the same APIs as users, `pm_device_runtime_get()` and friends. There will be no `#ifdef PM_DEVICE_RUNTIME` anywhere in the driver.

ceolin, 2025-04-25:
@bjarki-andreasen thanks for creating it. From my understanding it is missing creating power domains for internal devices, otherwise those devices won't never know when they are powered off. Other than that is looking great.

bjarki-andreasen, 2025-04-28:
> [@bjarki-andreasen](https://github.com/bjarki-andreasen) thanks for creating it. From my understanding it is missing creating power domains for internal devices, otherwise those devices won't never know when they are powered off. Other than that is looking great.

Right! that is step actually step 1 :)

ceolin, 2025-04-29:
> Right! that is step actually step 1 :)

Ah cool, I thought it was just referring to current ones.

nashif, 2025-04-30:
> 7\. Remove inline self get/put calls from device drivers, like often present in `can_start()`/`can_stop()` or `i2c_transfer()`. These are no longer needed as applications/subsystems are mandated to call these APIs before and after using the device.

I am not following this change, this is a pillar and a fundemntal deisgn of runtime device power management. Need more information and details about this change and what does it fix/provide.

I think I understand some of the other items, but there need to be some more details and probably we need an overview of the overall power management architecture, why those changes are needed and where we will end up and if anything else is going to follow or if those changes get us where want to be.

bjarki-andreasen, 2025-04-30:
> I am not following this change, this is a pillar and a fundemntal deisgn of runtime device power management. Need more information and details about this change and what does it fix/provide.
>
> I [...]

See the parent issue #84049 for details. This issue just describes the how, not the why :)

decsny, 2025-04-30:
Would the [busy status indication](https://docs.zephyrproject.org/latest/services/pm/device.html#busy-status-indication) have any use or be removed along with system managed device PM ?

bjarki-andreasen, 2025-05-01:
> Would the [busy status indication](https://docs.zephyrproject.org/latest/services/pm/device.html#busy-status-indication) have any use or be removed along with system managed device PM ?

Given device states now dictate what system states can be entered, the busy flag is not needed. The system simply can't change into a state which would conflict with the state of the device :)

bjarki-andreasen, 2025-05-01:
> > > [@bjarki-andreasen](https://github.com/bjarki-andreasen) thanks for creating it. From my understanding it is missing creating power domains for internal devices, otherwise those devices won't never know when they are powered off. Other than that is looking great.
> > 
> > 
> > Right! that is step actually step 1 :)
> 
> do you think it would it be feasible to have a generic driver or at least some common code for the concept of an "internal SOC domain"?

That is what I'm working towards with step 1. In short, an "internal SOC domain" is simply a power domain which constrains system states, like any device can today :)

```
soc_pd: soc-pd {
        compatible = "power-domain";
        zephyr,disabling-power-states = <&soc_sleep, &soc_deep_sleep>;
};

my-dev {
        compatible = "my-dev";
        power-domains = <&soc_pd>;
};
```

The driver for "power-domain" would just be an "empty" device which defines a PM_DEVICE. I'm moving all logic required for power domains up to PM_DEVICE_RUNTIME, so power domain drivers just have to implement PM_DEVICE_ACTION_SUSPEND and PM_DEVICE_ACTION_RESUM. Calling power domain children will be done automatically :)

> BTW, I am new to the power management area of zephyr, but I noticed NXP has this type of thing in a platform DT:
> 
> [zephyr/dts/arm/nxp/nxp_rw6xx_common.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/96f51bf797bb42cddab9429502082dde79276cb8/dts/arm/nxp/nxp_rw6xx_common.dtsi#L86-L90)
> 
> Lines 86 to 90 in [96f51bf](/zephyrproject-rtos/zephyr/commit/96f51bf797bb42cddab9429502082dde79276cb8)
> 
>  power_mode3_domain: power_mode3_domain { 
>  	compatible = "power-domain-soc-state-change"; 
>  	onoff-power-states = <&standby>; 
>  	#power-domain-cells = <0>; 
>  }; 
> 
> with [this binding ](https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/power-domain/power-domain-soc-state-change.yaml) and [this implementation](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/power_domain/power_domain_soc_state_change.c). It seems like it's a power domain representing an SOC system power mode/state instead of a power region in the SOC. Is this wrong / does it need to be changed ? Because on this platform for example there are different domains in the SOC and they are affected differently by the different power states, so having a power domain driver for a power mode seems backwards.
> Is there some other mechanism to indicate to devices that the power domain will be turned on or off from a system power mode transition, or is it supposed to be the other way around with something like where devices indicate to domains that they are busy and domains block the system power mode transition?

Exactly this: _"is it supposed to be the other way around with something like where devices indicate to domains that they are busy and domains block the system power mode transition?"_

users call pm_device_runtime_get() on what they need, say an I2C controller, PM_DEVICE_RUNTIME checks if the controller is on one or more power domains, it then resumes the power domains, some of which may prevent specific sleep states from being transitioned to, then resumes the I2C controller itself :)

bjarki-andreasen, 2025-05-01:
@decsny Here's the "internal SOC"/default power domain after the refactor https://github.com/zephyrproject-rtos/zephyr/pull/89372/commits/646281af3a285ed2b9bd124597d701f4f7170e64 

Its empty, given all power domain and pm policy constraints are part of PM_DEVICE_RUNTIME :)

decsny, 2025-05-29:
I recently helped someone understand configuration for the new proposal by drawing for them this table with this example device:

--------------------
possible SOC power states (example): mode_0 (active), mode_1 (idle), mode_2 (standby)

example device DT:
```
dev@x {
    reg = <x>;
    compat = "some_dev";
    zephyr,disabling-states = <&mode_2>;
};
```
| `CONFIG_PM` | `CONFIG_PM_DEVICE` | Initial Device State | Lowest possible SOC mode | 
|---|---|---|---|
| n | n | Active <sup>*</sup> | mode_0 |
| y | n | Active <sup>*</sup> | mode_1 | 
| n | y | Suspend | mode_0 |
| y | y | Suspend | mode_2<sup>+</sup> |

<sup>*</sup> will never be suspended if CONFIG_PM_DEVICE=n
<sup>+</sup> can only enter mode_2 if dev is suspended (device constraint)

--------------------

My question first of all is, is the explanation accurate?

Second, if it is, can I propose that we combine the two configs into one choice config? Since the implications of this are kind of confusing otherwise to people I think. My suggestion for the choices are like this for a hypothetical `CONFIG_PM_MODE` or similar:

Old configs -> new choice
* CONFIG_PM=n, CONFIG_PM_DEVICE=n --> "disabled" (because power management is disabled, obviously)
* CONFIG_PM=y, CONFIG_PM_DEVICE=n --> "simple" (documented as the mode where no domains ever lose power but system can idle?)
* CONFIG_PM=n, CONFIG_PM_DEVICE=y --> "domains" (the domains can be powered off but the system never idles, niche)
* CONFIG_PM=y, CONFIG_PM_DEVICE=y --> "full" (this is the mode where the system design potentially gets the most complex)

Thoughts?

carlescufi, 2025-06-10:
Architecture WG:

- @bjarki-andreasen explains the steps outlined in this issue
- @nashif suggests that we need to discuss this more before actually jumping to the implementation. Suggestion to revive the existing power management meeting
- @nashif also disagrees with the fact that 
- @bjarki-andreasen responds that many have already contributed to the discussion that has led to the curren proposal
- @bjarki-andreasen also comments that this is aligning the current implementation, no reworking everything
- @bjarki-andreasen also mentions that sometimes the application is the only one that has all the information required to process the details of power-management. Drivers can try to do what they can, but they are often lacking knowledge required to implement the full picture scenario

nashif, 2025-06-18:
> * Clean up all device drivers. This includes the following:
>   
>   * Use `pm_device_driver_init()` instead of calling `pm_device_runtime_enable()`, `pm_device_init_suspended()` etc.
>   * Use pm_device_runtime_ APIs for resuming device dependencies like busses, pwms, etc, from device drivers.
>   
>   note: we will not remove inline calls to self get/put in `can_start()` or `i2c_transfer()` at this time.
> *

and then

> 7\. Remove inline self get/put calls from device drivers, like often present in `can_start()`/`can_stop()` or `i2c_transfer()`. These are no longer needed as applications/subsystems are mandated to call these APIs before and after using the device.

So this is conflicting.

A user should be able to get/put a driver and that is fine, but we should not remove inline self get/put from drivers and we should leave this as is, power management shall happen at the lowest level, i.e. the driver itself which still can cover many usecases. We need to look at adding autosuspend APIs and other methods to optimize things at that level, i.e. to avoid repeatitive suspend/resume operations, but power management in general happens at the driver level and we should not push PM one level up as the only way.

bjarki-andreasen, 2025-06-19:
> and then
>
>
> So this is conflicting.

I meant "at this step", corrected to "until step 7" to be unambiguous

> A user should be able to get/put a driver and that is fine, but we should not remove inline self get/put from drivers and we should leave this as is, power management shall happen at the lowest [...]

All device PM can be done one layer above devices themselves, some can be done within device drivers at the cost of unneccesary resume/suspend cycles and increased and unpredrictable latencies. This alone, not even considering "autosuspend" leads to increasen power consumption. The only justification for these negatives I know of is that in some cases, users and subsystems don't have to power manage their devices. Are there others?

closed 2026-09-25 by github-actions[bot]