zephyrproject-rtos/zephyr #89056: STM32U0: PM and UART: PM lock is taken and never released

mickael868, 2025-04-24:
**Describe the bug**

While trying to add the PM feature to the STM32U0 target in this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/88963), I saw that the device never went to the sleep mode because it is locked by the UART driver.

**To Reproduce**

Using the following [MWE](https://github.com/mickael868/stm32u0-uart-pm-lock), build and flash with the typical commands:

- `west build -b example -- -DBOARD_ROOT=. `
- `west flash`

The current consumption is 1.75mA as is.
If `pm_policy_state_lock_get()` and `pm_policy_state_lock_put()` are commented out in `drivers/serial/uart_stm32.c`, the current consumption drops to 440 µA. In both cases, the debug probe is attached and consumes few hundreds of µA.

**Expected behavior**

The current consumption is low (ie. the system goes to a low power mode) without manually disabling the pm lock in `uart_stm32.c`.

**Impact**

The system does not reach the expected lifetime when battery-powered.

**Logs and console output**

The lock is get by `uart_poll_out()` at the first `printk()` (the boot banner or if disabled, my own app `printk()`). Subsequent `printk()` do not get or put the lock anymore.

```
Breakpoint 1, uart_poll_out (dev=<optimized out>, out_char=42 '*') at app/build/zephyr/include/generated/zephyr/syscalls/uart.h:108
108             z_impl_uart_poll_out(dev, out_char);
(gdb) backtrace
#0  uart_poll_out (dev=<optimized out>, out_char=42 '*') at app/build/zephyr/include/generated/zephyr/syscalls/uart.h:108
#1  console_out (c=42) at app/zephyr/drivers/console/uart_console.c:102
#2  0x080004a4 in char_out (c=<optimized out>, ctx_p=<optimized out>) at app/zephyr/lib/os/printk.c:98
#3  0x08004216 in vfprintf ()
#4  0x080004de in vprintk (fmt=0x8007689 "*** Booting Zephyr OS build v4.1.0-2646-ge41909a32ce9 ***\n", fmt@entry=0x1 <error: Cannot access memory at address 0x1>, ap=..., ap@entry=...) at app/zephyr/lib/os/printk.c:134
#5  0x0800560c in printk (fmt=0x8007689 "*** Booting Zephyr OS build v4.1.0-2646-ge41909a32ce9 ***\n") at app/zephyr/lib/os/printk.c:198
#6  0x08003ff0 in boot_banner () at app/zephyr/kernel/banner.c:46
#7  0x08002cc0 in bg_thread_main (unused1=<optimized out>, unused2=<optimized out>, unused3=<optimized out>) at app/zephyr/kernel/init.c:551
#8  0x08000508 in z_thread_entry (entry=0x8002cad <bg_thread_main>, p1=0x0, p2=0x0, p3=0x0) at app/zephyr/lib/os/thread_entry.c:48
#9  0x0800108c in arch_switch_to_main_thread (main_thread=0x8002cad <bg_thread_main>, stack_ptr=0x0, _main=0x8002cad <bg_thread_main>) at /home/mika/dev/embedded/zephyr/common/zephyr/arch/arm/core/cortex_m/thread.c:546
#10 0x20000608 in z_idle_threads ()
```

**Environment (please complete the following information):**

- OS: Linux
- Toolchain: Zephyr SDK: 0.16.3
- Zephyr: [PR](https://github.com/zephyrproject-rtos/zephyr/pull/88963) based on the git commit 5fb47cd97ff780e906243b285ed9461538718899

**Additional context**

The lock is taken by a *poll*  function but `CONFIG_UART_INTERRUPT_DRIVEN=y` is defined.

erwango, 2025-04-25:
@JordanYates Can you have a look ?

JordanYates, 2025-04-25:
@mickael868 can you verify if the `uart_stm32_isr` handler is being called?

mickael868, 2025-04-25:
@JordanYates No, `uart_stm32_isr()` is not called. 
`uart_stm32_irq_config_func_##index` gets called twice (2 lpuart ports are defined in the application dts). 
I will not be able to investigate on this issue for a week.

mickael868, 2025-05-30:
I'm closing this issue because I will not continue investigating on it.

- No TC ISR on LPUART1: LPUART1 was only exposed during the development and debug process of my application,  it is not exposed in the production device.
- Issues with the interrupt driven mode : I'm happy with the async mode. Here are an example UART wrapper if anyone stumbles on the same question:

uart.c

```c
/** SPDX-License-Identifier: CC0-1.0 */

#include "uart.h"

#include <zephyr/drivers/uart.h>
#include <zephyr/logging/log.h>
#include <zephyr/pm/device.h>

#define DMA_RX_BLOCK_SIZE 128
#define RX_FIFO_SIZE      512
#define TX_FIFO_SIZE      512

/* Signal RX availability by giving this semaphore, typically waited in an
 * event loop.
 * TODO adapt to your code */
extern struct k_sem _isr_sem;

LOG_MODULE_DECLARE(uart, CONFIG_UART_LOG_LEVEL);

static uint32_t _rx_timeout;

static bool _tx_pending;

/* double buffering used for RX. Index of the currently used buffer */
static int _rx_buf_idx;
static uint8_t _dma_rx_buf[2][DMA_RX_BLOCK_SIZE];

static uint8_t _dma_tx_buf[TX_FIFO_SIZE];

static uint8_t _rx_buf[RX_FIFO_SIZE];
static struct ring_buf _rx;

static uint8_t _tx_buf[TX_FIFO_SIZE];
static struct ring_buf _tx;

static int enable_rx_dma(const struct device* dev)
{
    return uart_rx_enable(
        dev,
        _dma_rx_buf[_rx_buf_idx],
        DMA_RX_BLOCK_SIZE,
        _rx_timeout);
}

static void
uart_cb(const struct device* dev, struct uart_event* evt, void* user_data)
{
    switch (evt->type)
    {
    case UART_RX_BUF_REQUEST:
        _rx_buf_idx = 1 - _rx_buf_idx;
        uart_rx_buf_rsp(dev, _dma_rx_buf[_rx_buf_idx], DMA_RX_BLOCK_SIZE);
        ring_buf_put(
            &_rx,
            evt->data.rx.buf + evt->data.rx.offset,
            evt->data.rx.len);
        k_sem_give(&_isr_sem);
        break;

    case UART_RX_RDY:
        ring_buf_put(
            &_rx,
            evt->data.rx.buf + evt->data.rx.offset,
            evt->data.rx.len);
        k_sem_give(&_isr_sem);
        break;

    case UART_TX_DONE:
        if (ring_buf_is_empty(&_tx))
        {
            _tx_pending = false;
        }
        else
        {
            uint32_t size = ring_buf_get(&_tx, _dma_tx_buf, TX_FIFO_SIZE);
            uart_tx(dev, _dma_tx_buf, size, SYS_FOREVER_US);
        }
        break;

    case UART_RX_DISABLED:
        /* re-enable */
        enable_rx_dma(dev);
        break;

    default:
        break;
    }
}

void uart_init(const struct device* dev, uint32_t bps)
{
    /* timeout = 1 byte */
    /* 8 bits + start + stop = 10. given in µs */
    _rx_timeout = 10e6 / bps;

    ring_buf_init(&_rx, sizeof(_rx_buf), _rx_buf);
    ring_buf_init(&_tx, sizeof(_tx_buf), _tx_buf);

    if (!device_is_ready(dev))
    {
        LOG_ERR("%s is not ready\n", dev->name);
        return;
    }

#ifdef CONFIG_PM
    bool ret;

    ret = pm_device_wakeup_is_capable(dev);
    if (!ret)
    {
        LOG_ERR("Device is not wakeup capable\n");
    }
    else
    {
        ret = pm_device_wakeup_enable(dev, true);
        if (!ret)
        {
            LOG_ERR("Could not enable wakeup source\n");
        }

        ret = pm_device_wakeup_is_enabled(dev);
        if (!ret)
        {
            LOG_ERR("Wakeup source not enabled\n");
        }
    }
#endif

    const struct uart_config uart_cfg = {
        .baudrate  = bps,
        .parity    = UART_CFG_PARITY_NONE,
        .stop_bits = UART_CFG_STOP_BITS_1,
        .data_bits = UART_CFG_DATA_BITS_8,
        .flow_ctrl = UART_CFG_FLOW_CTRL_NONE};

    int err = uart_configure(dev, &uart_cfg);
    if (err != 0)
    {
        LOG_ERR("uart config error %d\n", err);
    }

    err = uart_callback_set(dev, uart_cb, NULL);
    if (err != 0)
    {
        LOG_ERR("uart set cb error %d\n", err);
    }

    err = enable_rx_dma(dev);
    if (err != 0)
    {
        LOG_ERR("uart rx en error %d\n", err);
    }
}

struct ring_buf* uart_rx()
{
    return &_rx;
}

void uart_write(const struct device* dev, const uint8_t* data, size_t size)
{
    /* discard whole data block on overflow */
    if (ring_buf_space_get(&_tx) >= size)
    {
        ring_buf_put(&_tx, data, size);
        if (!_tx_pending)
        {
            uint32_t size = ring_buf_get(&_tx, _dma_tx_buf, TX_FIFO_SIZE);
            uart_tx(dev, _dma_tx_buf, size, SYS_FOREVER_US);
            _tx_pending = true;
        }
    }
}
```

uart.h

```c
/** SPDX-License-Identifier: CC0-1.0 */

#ifndef UART_H
#define UART_H

#include <zephyr/device.h>
#include <zephyr/kernel.h>
#include <zephyr/sys/ring_buffer.h>

typedef void (*UartRxCb)(void);

void uart_init(const struct device* dev, uint32_t bps);
struct ring_buf* uart_rx();
void uart_write(const struct device* dev, const uint8_t* data, size_t size);

#endif /* UART_H */
```

dts extract (to be adapted to your application)

```c
&lpuart3 {
  clocks = <&rcc STM32_CLOCK(APB1, 12U)>,
           <&rcc STM32_SRC_HSI LPUART3_SEL(2)>;
  wakeup-source;
  fifo-enable;
  pinctrl-0 = <&lpuart3_tx_pa4 &lpuart3_rx_pa5>;
  pinctrl-1 = <&lpuart3_tx_pa4 &lpuart3_rx_pa5>;
  pinctrl-names = "default", "sleep";
  current-speed = <115200>;
  dmas = <&dmamux1 0 35 (STM32_DMA_PERIPH_TX | STM32_DMA_MEM_8BITS | STM32_DMA_PERIPH_8BITS)>,
         <&dmamux1 1 34 (STM32_DMA_PERIPH_RX | STM32_DMA_MEM_8BITS | STM32_DMA_PERIPH_8BITS | STM32_DMA_MODE_CYCLIC)>;
  dma-names = "tx", "rx";
  status = "okay";
};
```

closed 2025-05-30 by mickael868