zephyrproject-rtos/zephyr #89071: BUG: Unable to mount a filesystem on disco_l475_iot1

Stanislas-MARTIN, 2025-04-25:
**Describe the bug**

I am using the `disco_l475_iot1` and trying to mount a littlefs file system on the storage partition. Using GDB, I was able to found out that the macro `FLASH_SIZE` define in `flash_stm32.h` is equal to 0 when it should 1MB in my case.

**To Reproduce**

Steps to reproduce the behavior:
1. Create an app.overlay with the following content
```
&quadspi {
        status = "disabled";
};

&mx25r6435f {
        status = "disabled";
};
```
2. Create a prj.conf with the following content
```
```

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

FRASTM, 2025-06-05:
Hello,

I took the sample you described and get the following log output:
```
Mounting fs
[00:00:00.001,000] <inf> littlefs: LittleFS version 2.11, disk version 2.1
[00:00:00.001,000] <inf> littlefs: FS at flash-controller@40022000:0xf8000 is 16 0x800-byte blocks with 512 cycle
[00:00:00.001,000] <inf> littlefs: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.001,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1386: Corrupted dir pair at {0x0, 0x1}
[00:00:00.001,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:00.023,000] <err> os: ***** MPU FAULT *****
[00:00:00.023,000] <err> os:   Stacking error (context area might be not valid)
[00:00:00.023,000] <err> os:   Data Access Violation
[00:00:00.023,000] <err> os:   MMFAR Address: 0x20002360
[00:00:00.023,000] <err> os: r0/a1:  0x18db3701  r1/a2:  0x2a004152  r2/a3:  0x3bfff10b
[00:00:00.023,000] <err> os: r3/a4:  0xe9cddaf9 r12/ip:  0xe7b5320c r14/lr:  0x2048f89d
[00:00:00.023,000] <err> os:  xpsr:  0x41070000
[00:00:00.023,000] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000004  s[ 2]:  0x200023cc  s[ 3]:  0x00000004
[00:00:00.023,000] <err> os: s[ 4]:  0x2a10d44a  s[ 5]:  0x20002510  s[ 6]:  0x200024a0  s[ 7]:  0x20000738
[00:00:00.023,000] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x0801198d  s[10]:  0xd13f2a00  s[11]:  0x181b9a07
[00:00:00.023,000] <err> os: s[12]:  0x0101eb42  s[13]:  0x5f80f1b1  s[14]:  0xe9cdd247  s[15]:  0xf7ff0ff0
[00:00:00.023,000] <err> os: fpscr:  0x0f66f1b8
[00:00:00.023,000] <err> os: Faulting instruction address (r15/pc): 0x08011638
[00:00:00.023,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.023,000] <err> os: Current thread: 0x200010b0 (unknown)
[00:00:00.044,000] <err> os: Halting system
```

The FLASH_SIZE is internal value of the ../modules/hal/stm32/stm32cube/stm32l4xx/, and the  `#define CONFIG_FLASH_SIZE 1024 ` is correct.
The flash API get_size() function is implemented by he stm32L4 series

FRASTM, 2025-06-06:
after disabling the mpu and restoring the log out put is now :
```
Mounting fs
Mounting successfull
[00:00:00.001,000] <inf> littlefs: LittleFS version 2.9, disk version 2.1
[00:00:00.001,000] <inf> littlefs: FS at flash-controller@40022000:0xf8000 is 16 0x800-byte blocks with 512 cycle
[00:00:00.001,000] <inf> littlefs: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.003,000] <inf> littlefs: /lfs unmounted
```

erwango, 2025-06-06:
> It appears that the behavior changes between running the code on the board and running it on renode

Feel free to close if issue is not reproduced with actual board.

closed 2025-06-06 by Stanislas-MARTIN