zephyrproject-rtos/zephyr #89120: No cfg left for sc_save in bt gatt.c no mechanism to clear SC data once error is raised.

mstumpf585, 2025-04-25:
**Describe the bug**
Seeing this error after I've been disconnected from failed pairing attempts with L2 Security enabled `<err> bt_gatt: sc_save: unable to save SC: no cfg left` . I'm looking at my current gatt.c using zephyr 3.7.0 and I don't see where the SC ever gets cleared if find_sc_cfg fails. I just see the Error message. Is there a higher level API function to clear it? Am I doing something wrong?  I was doing some testing where I'd disconnect if pairing failed on an IOS device and this came up after doing that multiple times, and now I can't get it to go away :(.

**To Reproduce**

Steps to reproduce the behavior:
Using zephyr 3.7.0 with the following BLE configs 
```
CONFIG_SETTINGS=y
CONFIG_BT=y
CONFIG_BT_HCI=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_ATT_PREPARE_COUNT=1
CONFIG_BT_PRIVACY=y
CONFIG_BT_KEYS_OVERWRITE_OLDEST=y
CONFIG_BT_FILTER_ACCEPT_LIST=y
CONFIG_BT_GATT_DYNAMIC_DB=y
CONFIG_BT_CTLR_LE_PING=n
CONFIG_BT_EXT_ADV=y
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_MAX_PAIRED=1
```

I'm using a device without any input and very little output so I was using Security L2 just works pairing with my IOS device. I was doing some testing for the condition where a user presses "cancel" on the "Do you want to pair" pop-up. When Pairing fails I get notified in the bt_conn_cb connected callback and handle it like so: 

```
static void connected(struct bt_conn *conn, uint8_t err)
{
    // set this early we will need it to disconnect if an error occurs here.
    ble_context.conn = conn;

    if (err) {
        LOG_ERR("Connection failed (err 0x%02x)", err);
        ble_attempt_disconnect();
        return;
    }
    else {
        LOG_INF("Connected");
        int err = bt_conn_set_security(conn, BT_SECURITY_L2);
        if (err != 0)  // Request encryption
        {
            LOG_ERR("Error requesting security with controller,  %d. Disconnecting", err);
            ble_attempt_disconnect();
            return;
        }
        ble_cmd_svc_on_connect();
    }
}

int ble_attempt_disconnect()
{
    int result = -ECANCELED; // assume error.
    if (ble_context.conn != NULL)
    {
        result = bt_conn_disconnect(ble_context.conn, BT_HCI_ERR_REMOTE_USER_TERM_CONN);
        if (result == 0) {
                LOG_INF("Disconnect success");
        } else {
                LOG_WRN("Disconnect failed (err %d)", result);
        }
    }

    return result;
}
```

After doing multiple iterations of cancelling the pair request my device now logs `<err> bt_gatt: sc_save: unable to save SC: no cfg left`. I found this issue https://github.com/zephyrproject-rtos/zephyr/issues/22539 and saw it was resolved by clearing the SC based on the connection. I don't see a high level api that would allow for my code to call sc_clear. 

**Expected behavior**
I would expect the sc_clear to be called for the failed connection or for there at least be a higher level function to call it myself. 

**Impact**
Seems like the device can still pair and communicate with the services. Not a complete show stopper just worried about releasing with this issue. 

**Logs and console output**
My logs if I press "pair":
```
Advertising updated
Advertising successfully started
Updated MTU: TX: 23 RX: 23 bytes
Connected
Advertising successfully stopped
Updated MTU: TX: 527 RX: 498 bytes

bt_gatt: sc_save: unable to save SC: no cfg left
Pairing Complete bonded 1
```

Logs if I don't pair:
```
Advertising updated
Advertising successfully started
Updated MTU: TX: 23 RX: 23 bytes
Connected
Advertising successfully stopped
Updated MTU: TX: 527 RX: 498 bytes

bt_gatt: sc_save: unable to save SC: no cfg left
bt_smp: smp_pairing_failed: pairing failed (peer reason 0x8)
```

**Environment (please complete the following information):**

 - OS: Linux 
 - Toolchain using zephyr sdk 0.16.8
 - Commit SHA or Version Tag zephyr 3.7.0

**Additional context**
Message @jhedberg on Discord about the issue which led me to posting this here.

jhedberg, 2025-04-28:
It looks to me like this piece of code is making the false assumption that `BT_ADDR_LE_ANY` would find a free slot:
```c
		/* Find and initialize a free sc_cfg entry */
		cfg = find_sc_cfg(BT_ID_DEFAULT, BT_ADDR_LE_ANY);
		if (!cfg) {
			LOG_ERR("unable to save SC: no cfg left");
			return;
		}
```

When we look at the implementation of `find_Sc_cfg()` that's clearly not the case:
```c
static struct gatt_sc_cfg *find_sc_cfg(uint8_t id, const bt_addr_le_t *addr)
{
	LOG_DBG("id: %u, addr: %s", id, bt_addr_le_str(addr));

	for (size_t i = 0; i < ARRAY_SIZE(sc_cfg); i++) {
		if (id == sc_cfg[i].id &&
		    bt_addr_le_eq(&sc_cfg[i].peer, addr)) {
			return &sc_cfg[i];
		}
	}

	return NULL;
}
```

jhedberg, 2025-04-28:
> When we look at the implementation of `find_sc_cfg()` that's clearly not the case:

Sorry, actually it is the case, although implicitly, since `BT_ID_DEFAULT` and `BT_ADDR_LE_ANY` are the "all zeroes" values, so any such entry would be returned.

jhedberg, 2025-04-28:
@mstumpf585 from what I can see, the following code in the `gatt.c` `disconnected_cb()` function should be responsible for clearing the sc data when necessary, so the task at hand would be to figure out why it's not being called in your case:

https://github.com/zephyrproject-rtos/zephyr/blob/29197ac9633c2542fd0c9b63639af1f75462844e/subsys/bluetooth/host/gatt.c#L3434-L3441

mstumpf585, 2025-04-28:
@jhedberg I can try to  reproduce with main tonight. Sorry was away all weekend. Will the `disconnected_cb()` still be called if I assigned callback to to the bt_conn_cb struct? I'd assume so? 

```
static void disconnected(struct bt_conn *conn, uint8_t reason)
{
    ARG_UNUSED(conn);
    LOG_INF("Disconnected (reason 0x%02x)", reason);
    ble_cmd_svc_on_disconnect();
    k_timer_stop(&disconnect_timer);
    ble_context.conn = NULL;
}

static struct bt_conn_cb conn_callbacks = {
    .connected = connected,
    .disconnected = disconnected,
};
```

I just noticed I nulled out `ble_context.conn` could that be why? Not sure why I did that...

mstumpf585, 2025-04-28:
Seems like setting the conn to null might have had some bad side effects since I removed it, I don't see the error anymore. Still need to verify this on the latest.

mstumpf585, 2025-04-30:
[test_mods.patch](https://github.com/user-attachments/files/19982194/test_mods.patch)

Okay so I tried to replicate on main and in 3.7 by modifying the peripheral_accept_list sample in both. I've attached a patch file for the changes I did to mains peripheral_accept_list. I'm wondering if I just got unlucky and caught some weird edge case. Like I said not setting the conn to NULL in the disconnected callback seemed to clean this up for me. 

I suppose we can just close this ticket. I'll keep an eye out.

closed 2025-04-30 by mstumpf585