zephyrproject-rtos/zephyr #89132: Pico2 MPU Test Fail

RafaelRMachado, 2025-04-26:
I am new to zephyr and I am trying to understand the MPU test sample. I am using Raspberry Pi Pico 2. When I run the MPU test sample, the result is different than expected, according to https://docs.zephyrproject.org/latest/samples/arch/mpu/mpu_test/README.html#mpu

On raspberry pi pico 2, the mpu write function works without error. My question is why does this function work? I understand that mpu write, mpu run and mpu read should all give error. 

I will attach a print screen of the results I obtained using the Raspberry pi pico 2.
[image]

[image]

_Originally posted by @EnricoEng in https://github.com/zephyrproject-rtos/zephyr/discussions/89038_

henrikbrixandersen, 2025-04-26:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

RafaelRMachado, 2025-04-29:
Hi @henrikbrixandersen 
Description updated.
Let us now in case something is missing.

Thanks [@EnricoEng](https://github.com/EnricoEng) for providing me the bug report in particular