zephyrproject-rtos/zephyr #89133: ttgo_tbeam/esp32: system crashes on rtc_set_time

fmoessbauer, 2025-04-26:
**Describe the bug**

When calling `rtc_set_time` on the `ttgo_tbeam/esp32/procpu` board, the system crashes. Calling `rtc_get_time` properly works and reports a time ~ 1919.

**To Reproduce**

Add an alias for the RTC (in `samples/drivers/rtc/boards/ttgo_tbeam_esp32_procpu.overlay`:

```
/ {
	aliases {
			rtc = &rtc;
	};
};
```

Steps to reproduce the behavior:
1. `west build -b ttgo_tbeam/esp32/procpu samples/drivers/rtc/`
2. `west flash`
3. `west espressif monitor`

```
abort()
[00:00:38.246,000] <err> os:  ** FATAL EXCEPTION
[00:00:38.246,000] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[00:00:38.246,000] <err> os:  **  PC 0x40086413 VADDR 0
[00:00:38.246,000] <err> os:  **  PS 0x60220
[00:00:38.246,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:2 CALLINC:2)
[00:00:38.246,000] <err> os:  **  A0 0x800ebdd2  SP 0x3ffcd2b0  A2 0x4  A3 0x1
[00:00:38.246,000] <err> os:  **  A4 0x1  A5 0  A6 0x3ffc2c38  A7 0x3ffc3f68
[00:00:38.246,000] <err> os:  **  A8 0x8008abdc  A9 0x3ffcd240 A10 0x3f406460 A11 0x3ffcd2a0
[00:00:38.246,000] <err> os:  ** A12 0x3ffcd280 A13 0x4 A14 0x3ffc1d10 A15 0x3ffcd3d0
[00:00:38.246,000] <err> os:  ** LBEG 0x4000c2e0 LEND 0x4000c2f6 LCOUNT 0
[00:00:38.246,000] <err> os:  ** SAR 0xf
[00:00:38.246,000] <err> os:  **  THREADPTR 0x3ffcd2c0

Backtrace:0x40086410:0x3ffcd2b0 0x400ebdcf:0x3ffcd2c0 0x4008e45a:0x3ffcd2e0 0x40093c29:0x3ffcd300 0x401073d1:0x3ffcd320 0x400dc85c:0x3ffcd340 0x400dcc11:0x3ffcd3a0 0x400dce2b:0x3ffcd3d0 0x400d9737:0x3ffcd410 0x40087aff:0x3ffcd440 0x400ddd3b:0x3ffcd460 

[00:00:38.246,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:38.246,000] <err> os: Current thread: 0x3ffc3e48 (unknown)
[00:00:38.367,000] <err> os: Halting system
```

**Impact**

This is a potential showstopper, as it means I cannot use the RTC.

**Environment (please complete the following information):**

Zephyr commit: 02f85b253a16f71176d8e4995d801ac464a14633

I cannot go back to a stable release, as the 4.1.0 release does not have the board. CC @kartben

tuna-f1sh, 2025-05-15:
Same problem on xiao_esp32c3. Can reproduce on samples/drivers/rtc using overlay:

```
/ {
	aliases {
		rtc-dev = &rtc;
	};
};

&xiao_i2c {
	status = "okay";

	pcf8563_rtc: pcf8563@51 {
		status = "okay";
		compatible = "nxp,pcf8563";
		reg = <0x51>;
	};
};
```

Using the xiao extender board with PCF8563 and aliasing that in place of `rtc` the sample works but otherwise goes to fatal error handler. I did try stepping through and seems to be the assert here: https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/components/esp_hw_support/periph_ctrl.c#L65

sylvioalves, 2025-05-15:
Will take a look.

tuna-f1sh, 2025-05-19:
Had a bit of time to look into this more this morning. I think it's a bit of user error going on here but still some questions.

The `rtc` peripheral in the devicetree refers to the pure RTC counter.

The reason it was hanging at the assert linked above is because the dev->api is cast to the `clock_control_esp32_api` and it's calling [`clock_control_esp32_on`](https://github.com/zephyrproject-rtos/zephyr/blob/05c7fadf959619590fec89083b387580b05f06e2/drivers/clock_control/clock_control_esp32.c#L801) in place of what should be [`set_time`](https://github.com/zephyrproject-rtos/zephyr/blob/05c7fadf959619590fec89083b387580b05f06e2/include/zephyr/drivers/rtc.h#L168) in the RTC API (since they sit at the same offset). It's passing a `struct *rtc_time` as the second arg for what should be the `clock_control_subsys_t sys` so no surprise that assert faults!

My questions:

* Is there a ESP32 peripheral that can be used in Zephyr for _the_ [RTC](https://github.com/zephyrproject-rtos/zephyr/blob/05c7fadf959619590fec89083b387580b05f06e2/include/zephyr/drivers/rtc.h)? I came across [rtc_time.c](https://github.com/zephyrproject-rtos/hal_espressif/blob/fa60a7aaa1f834c66413bbaa771794cf8aa82946/components/esp_hw_support/port/esp32/rtc_time.c) but doesn't seem to define set time?
* There is time support using the RTC in Espressif: https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/system/system_time.html
* One could of course use the RTC as a persistent counter and set it with an epoch to track time but seems like there should be an API around this and I'm missing it.

Thank!

sylvioalves, 2025-05-19:
I am fixing this and will submit a simple RTC driver that keeps track of the time.
https://github.com/zephyrproject-rtos/zephyr/pull/90122 -> 1st re-organized clock subsystem.

Next PR will bring RTC functionality.

tuna-f1sh, 2025-05-19:
Ah perfect. Makes sense thanks - the RTC naming was confusing so first change is good move. Will watch for RTC PR and help if you need testing.

sylvioalves, 2025-06-26:
Closing this as ESP32 has no RTC calendar as expected by Zephyr. For more info:
Read https://docs.zephyrproject.org/latest/hardware/peripherals/rtc.html#overview
Check #92253 comments.

closed 2025-06-26 by sylvioalves