zephyrproject-rtos/zephyr #89147: nRF54L: uptime does not reset across reboots

JordanYates, 2025-04-27:
**Describe the bug**

On the nRF54L, resetting the application without removing power does not result in the internal counters starting again from 0.

**To Reproduce**

1. Update `zephyr/samples/hello_world` to:
```
#include <stdio.h>
#include <zephyr/kernel.h>

int main(void)
{
	while(1) {
		printf("Hello World! %s %6d\n", CONFIG_BOARD_TARGET, k_uptime_seconds());
		k_sleep(K_SECONDS(1));
	}
	return 0;
}
```
2. `west build -b nrf54l15dk/nrf54l15/cpuapp zephyr/samples/hello_world/`
3. `west flash`
4. Wait several seconds
5. `nrfutil device reset`
6. Observe the serial output

**Expected behavior**

Application should start counting from 0 on each reset.

**Impact**

Application thinks it's been running for much longer than it has been.
Absolute timestamps not constructed without reference to the current uptime will be broken.
i.e. `K_TIMEOUT_ABS_MS(5000)` will expire immediately upon reboot.

**Logs and console output**

[image]

**Environment (please complete the following information):**

 - Zephyr v4.1 and main

**Additional context**

`k_uptime_seconds()` is not special in any way, it simply retrieves the system tick counter and does a division.
This implies that the system tick counter is not being reset across reboots.

The nRF54L series does not use the same system tick source as previous nRF SoCs, it uses the GRTC (`nrf_grtc_timer`).

nordicjm, 2025-05-02:
Ping @nordic-krch

adamkondraciuk, 2025-05-08:
GRTC is not cleared on every reboot by default and we don't want to do that, however I agree that uptime should be cleared on every reboot - we're working on it. 
For now you can use this PR which adds configuration that clears the GRTC on driver initialization:
https://github.com/zephyrproject-rtos/zephyr/pull/89664
To use it just add `CONFIG_NRF_GRTC_CLEAR_AT_INIT=y` in your `.conf file`, or build it using command:
`west build -b nrf54l15dk/nrf54l15/cpuapp zephyr/samples/hello_world/ -- -DCONFIG_NRF_GRTC_CLEAR_AT_INIT=y`
`*** Booting Zephyr OS build v4.1.0-3665-g4c7866beb59a ***
Hello World! nrf54l15dk/nrf54l15/cpuapp      0
Hello World! nrf54l15dk/nrf54l15/cpuapp      1
Hello World! nrf54l15dk/nrf54l15/cpuapp      2
Hello World! nrf54l15dk/nrf54l15/cpuapp      3
Hello World! nrf54l15dk/nrf54l15/cpuapp      4
Hello World! nrf54l15dk/nrf54l15/cpuapp      5
Hello World! nrf54l15dk/nrf54l15/cpuapp      6
*** Booting Zephyr OS build v4.1.0-3665-g4c7866beb59a ***
Hello World! nrf54l15dk/nrf54l15/cpuapp      0
Hello World! nrf54l15dk/nrf54l15/cpuapp      1
Hello World! nrf54l15dk/nrf54l15/cpuapp      2
Hello World! nrf54l15dk/nrf54l15/cpuapp      3
Hello World! nrf54l15dk/nrf54l15/cpuapp      4
Hello World! nrf54l15dk/nrf54l15/cpuapp      5
Hello World! nrf54l15dk/nrf54l15/cpuapp      6
Hello World! nrf54l15dk/nrf54l15/cpuapp      7
`

hubertmis, 2025-05-09:
GRTC is not cleared to provide RTC functionality. When GRTC is used as timer, it can give values starting with 0 at the system initialization. It can be easily done by reading GRTC counter value at the driver init and offsetting returned values by the initial one.