zephyrproject-rtos/zephyr #89177: http_server/http_client samples not working with 802.15.4 support

EricSTMicro, 2025-04-28:
I'm trying to make the samples http_server and http_client work together with the support of IEEE 802.15.4.
Unfortunately, it fails during the TCP connection and doesn't go further (see wireshark capture in zip). I'm not sure what is expected after that, some data exchange I guess, the readme doesn't provide any information about it.

I am trying to make it work on 2 nRF52840 dongles (the result is the same on an other chip from other manufacturer). So for me it's not related to nRF board.

Samples are here: 
- zephyrproject\zephyr\samples\net\sockets\http_server
- zephyrproject\zephyr\samples\net\sockets\http_client

The only modification I made:
- for http_client, swap the values of CONFIG_NET_CONFIG_MY_IPV6_ADDR and CONFIG_NET_CONFIG_PEER_IPV6_ADDR, otherwise boards will have the same IPv6 address.

Here is the build command used (using Zephyr IDE in VSCode):
* http_server:
 west build -b nrf52840dongle c:\zephyrproject\zephyr\samples\net\sockets\http_server  --build-dir c:\zephyrproject\zephyr\samples\net\sockets\http_server\build\nrf52840dongle  -- -DBOARD_ROOT='c:\zephyrproject\zephyr'   -DEXTRA_CONF_FILE='c:\zephyrproject\zephyr\samples\net\sockets\http_server\ieee802154-overlay.conf;'  

* http_client:
west build -b nrf52840dongle c:\zephyrproject\zephyr\samples\net\sockets\http_client  --build-dir c:\zephyrproject\zephyr\samples\net\sockets\http_client\build\nrf52840dongle  -- -DBOARD_ROOT='c:\zephyrproject\zephyr'   -DEXTRA_CONF_FILE='c:\zephyrproject\zephyr\samples\net\sockets\http_client\ieee802154-overlay.conf;'

I'm using Windows, on Zephyr 4.1.0-rc1 using Zephyr IDE in VSCode.
CMake version: 3.28.1
zephyr-sdk-0.16.1
zephyrproject commit 50c21f159143e18c8a05cbb946a5e7610033355f (from 02/13/2025).

[http_server_client_nrf52840.zip](https://github.com/user-attachments/files/19936567/http_server_client_nrf52840.zip)

rlubos, 2025-04-28:
This looks like a simple port mismatch? Client tries to connect to port `8000`, yet the server by default listens on port `80`. This can be seen in the pcap, the server rejects connection on port `8000`. Server port can be specified with `CONFIG_NET_SAMPLE_HTTP_SERVER_SERVICE_PORT=8000`.

rlubos, 2025-04-28:
Also note, the `http_client` sample was not really designed with the `http_server` sample in mind (the latter was added later), some requests (POST ones) will fail.

rlubos, 2025-04-28:
> So it's clear for me, will something be done to make them work together?

If you have some spare time, I think we could update the addresses and port in the 15.4 overlay files, contributions are welcome. Optionally also add dynamic handlers for missing resources on the server side.

closed 2026-09-23 by github-actions[bot]