zephyrproject-rtos/zephyr #89180: west: runner: failed to flash remote image for mbox test

hakehuang, 2025-04-28:
**Describe the bug**

only flash the main core image and does not flash the remote board image.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt700_evk
 - mimxrt1180_evk_cm33
 - frdm_mcxn947_cpu0
 - mimxrt1170_evk
 - What have you tried to diagnose or workaround this issue?
 - No
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?

```
89721463022e9eb0ad02c712ad2f975b57594b44 is the first bad commit
commit 89721463022e9eb0ad02c712ad2f975b57594b44
Author: Vignesh Pandian <vignesh@aerlync.com>
Date:   Mon Mar 24 10:00:13 2025 +0530

    west: fix for context with sysbuild

    show `west flash --context` with sysbuild

    Signed-off-by: Vignesh Pandian <vignesh@aerlync.com>

 scripts/west_commands/run_common.py | 117 +++++++++++++++++++++++++-----------
 1 file changed, 83 insertions(+), 34 deletions(-)

```

**To Reproduce**

1. erase the board totally
2. build and flash the mbox image

```
west update
scripts/twister --device-testing --hardware-map ${HOME}/nxp/mimxrt1180_evk_cm33/map.yaml -T tests/drivers/mbox/ --west-flash
```

**Expected behavior**

run pass

**Impact**

functions with remote boards

**Logs and console output**

```

*** Booting Zephyr OS build v4.1.0-3062-g29197ac9633c ***
Running TESTSUITE mbox_data_tests
===================================================================
START - test_ping_pong_1
*** Booting Zephyr OS build v4.1.0-3062-g29197ac9633c ***
Running TESTSUITE mbox_data_tests
===================================================================
START - test_ping_pong_1

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-3062-g29197ac9633c

hakehuang, 2025-04-28:
[test_failure.txt](https://github.com/user-attachments/files/19937123/test_failure.txt)

[test_ok.txt](https://github.com/user-attachments/files/19937126/test_ok.txt)

I attached two log one is failure and one is ok.

nordic-piks, 2025-04-28:
Problem visible at any test for .nrf54l15dk/nrf54l15/cpuflpr.

PR #89179 (2025-04-28): Revert "west: fix for context with sysbuild"
This reverts commit 89721463022e9eb0ad02c712ad2f975b57594b44.

This commit breaks flashing multiple images @vignesh-aerlync

Fixes #89180
closed 2025-04-28 by kartben