zephyrproject-rtos/zephyr #89192: west: build: test-item support conditional configs

hakehuang, 2025-04-28:
**Is your enhancement proposal related to a problem? Please describe.**

in samples.yaml and testcase.yaml, we support conditional extra_configs, extra_args like below

<arch>: <arch name> : xxx
<platform>: <platform name>:xxx

which parsed by twister scripts, but this is not supported by west build. se when you run

```
west build -b mimxrt1060_evk -T drivers.uart.async_api.lpuart.rt_nocache tests/drivers/uart/uart_async_api
```  

it will run with

```
Running CMake: /home/ubuntu/.local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/shared/disk/zephyr_project/zephyr_test/zephyr/build -GNinja -Dplatform:mimxrt1010_evk/mimxrt1011:DTC_OVERLAY_FILE=nxp/dut_lpuart4_loopback.overlay -Dplatform:mimxrt1015_evk/mimxrt1015:DTC_OVERLAY_FILE=nxp/dut_lpuart4_loopback.overlay -Dplatform:mimxrt1020_evk/mimxrt1021:DTC_OVERLAY_FILE=nxp/dut_lpuart2_loopback.overlay -Dplatform:mimxrt1024_evk/mimxrt1024:DTC_OVERLAY_FILE=nxp/dut_lpuart2_loopback.overlay -Dplatform:mimxrt1160_evk/mimxrt1166/cm7:DTC_OVERLAY_FILE=nxp/dut_lpuart2_loopback.overlay -Dplatform:mimxrt1170_evk@A/mimxrt1176/cm7:DTC_OVERLAY_FILE=nxp/dut_lpuart2_loopback.overlay -Dplatform:mimxrt1170_evk@B/mimxrt1176/cm7:DTC_OVERLAY_FILE=nxp/dut_lpuart2_loopback.overlay -Dplatform:vmu_rt1170/mimxrt1176/cm7:DTC_OVERLAY_FILE=nxp/dut_lpuart4_loopback.overlay -Dplatform:mimxrt1180_evk/mimxrt1189/cm33:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1050_evk/mimxrt1052/hyperflash:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1060_evk/mimxrt1062/hyperflash:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1060_evk@A/mimxrt1062/qspi:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1060_evk@B/mimxrt1062/qspi:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1060_evk@C/mimxrt1062/qspi:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -Dplatform:mimxrt1064_evk/mimxrt1064:DTC_OVERLAY_FILE=nxp/dut_lpuart3_loopback.overlay -DCONFIG_DCACHE=y -DCONFIG_NOCACHE_MEMORY=y -DCONFIG_USERSPACE=n -DCONFIG_TEST_USERSPACE=n -S/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/drivers/uart/uart_async_api
```

which is wrong. and look at the west_command/build.py this implement is not supported

```

    def _parse_test_item(self, test_item):
        found_test_metadata = False
        for yp in ['sample.yaml', 'testcase.yaml']:
            yf = os.path.join(self.args.source_dir, yp)
            if not os.path.exists(yf):
                continue
            found_test_metadata = True
            with open(yf, 'r') as stream:
                try:
                    y = yaml.safe_load(stream)
                except yaml.YAMLError as exc:
                    self.die(exc)
            common = y.get('common')
            tests = y.get('tests')
            if not tests:
                self.die(f"No tests found in {yf}")
            if test_item not in tests:
                self.die(f"Test item {test_item} not found in {yf}")
            item = tests.get(test_item)

            sysbuild = False
            extra_dtc_overlay_files = []
            extra_overlay_confs = []
            extra_conf_files = []
            required_snippets = []
            for section in [common, item]:
                if not section:
                    continue
                sysbuild = section.get('sysbuild', sysbuild)
                for data in [
                        'extra_args',
                        'extra_configs',
                        'extra_conf_files',
                        'extra_overlay_confs',
                        'extra_dtc_overlay_files',
                        'required_snippets'
                        ]:
                    extra = section.get(data)
                    if not extra:
                        continue
                    if isinstance(extra, str):
                        arg_list = extra.split(" ")
                    else:
                        arg_list = extra

                    if data == 'extra_configs':
                        args = []
                        for arg in arg_list:
                            equals = arg.find('=')
                            colon = arg.rfind(':', 0, equals)
                            if colon != -1:
                                # conditional configs (xxx:yyy:CONFIG_FOO=bar)
                                # are not supported by 'west build'
                                self.wrn('"west build" does not support '
                                         'conditional config "{}". Add "-D{}" '
                                         'to the supplied CMake arguments if '
                                         'desired.'.format(arg, arg[colon+1:]))
                                continue
                            args.append("-D{}".format(arg.replace('"', '\"')))
                    elif data == 'extra_args':
                        # Retain quotes around config options
                        config_options = [arg for arg in arg_list if arg.startswith("CONFIG_")]
                        non_config_options = [arg for arg in arg_list if not arg.startswith("CONFIG_")]
                        args = ["-D{}".format(a.replace('"', '\"')) for a in config_options]
                        args.extend(["-D{}".format(arg.replace('"', '')) for arg in non_config_options])
                    elif data == 'extra_conf_files':
                        extra_conf_files.extend(arg_list)
                        continue
                    elif data == 'extra_overlay_confs':
                        extra_overlay_confs.extend(arg_list)
                        continue
                    elif data == 'extra_dtc_overlay_files':
                        extra_dtc_overlay_files.extend(arg_list)
                        continue
                    elif data == 'required_snippets':
                        required_snippets.extend(arg_list)
                        continue

                    if self.args.cmake_opts:
                        self.args.cmake_opts.extend(args)
                    else:
                        self.args.cmake_opts = args

            self.args.sysbuild = sysbuild

        if found_test_metadata:
            args = []
            if extra_conf_files:
                args.append(f"CONF_FILE=\"{';'.join(extra_conf_files)}\"")

            if extra_dtc_overlay_files:
                args.append(f"DTC_OVERLAY_FILE=\"{';'.join(extra_dtc_overlay_files)}\"")

            if extra_overlay_confs:
                args.append(f"OVERLAY_CONFIG=\"{';'.join(extra_overlay_confs)}\"")

            if required_snippets:
                args.append(f"SNIPPET=\"{';'.join(required_snippets)}\"")

            # Build the final argument list
            args_expanded = ["-D{}".format(a.replace('"', '')) for a in args]

            if self.args.cmake_opts:
                self.args.cmake_opts.extend(args_expanded)
            else:
                self.args.cmake_opts = args_expanded

        return found_test_metadata
```

**Describe the solution you'd like**

update the build.py to support such conditional build

**Describe alternatives you've considered**

use `west twister xxxx`

hakehuang, 2025-04-28:
@PerMac any comments on this request?

PerMac, 2025-05-08:
I had an RFC for exact same issue (west --test-item not aligned with twister -s) and it was resolved then by adding missing functionality to west build command #60502. I don't think further copying twister features to west will be productive. IMO we have to think and decide what is the main purpose of using west vs twister and what is the difference between these tools and which functionality we want to share. If we want the same functionality we have to start using the same "backend", i.e. some python module shared by both tools. 
@nashif did mention on the last testing wg meeting that maybe some functionality like twister-lite, striped from all twister preprocessing, filtering, extensive reporting etc. Maybe a solution can resolve around this idea? We could maybe try achieving this by abstracting and extracting the build and execution module in twister so it can be used by twister and by west. I don't think it should be usable by itself, as west  --test-item is already "light weight".