zephyrproject-rtos/zephyr #89193: websocket_client loses first messages if they are in the same network packet

dantti, 2025-04-28:
**Describe the bug**

When the server upgrade reply and the first N websocket messages are on the same network packet they are skipped by websocket_recv_msg(), in fact looking at the code I would believe the issue lies on the underground http code. But didn't do a deep investigation, only that in Wireshark the packet was well formed.
I'm using esp32.

**To Reproduce**

I believe the websocket client sample should be enough, tho I find it very weird that it has raw socket operations there, as servers should just drop those messages as they won't be well formed.
The important step is to send the http upgrade handshake and some websocket packet msg, ping/binary/text doesn't matter the kind.

**Expected behavior**

I need to process ping messages to keep the connection alive, and receive setup instructions but those are lost if I send the right away.

**Impact**

It's a critical issue due data-loss IMO.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux
 - Zephyr SDK (main)
 - f9220146e623a7a6794c53f1a641aaca055e6b16

**Additional context**

jukkar, 2025-04-28:
> I believe the websocket client sample should be enough, tho I find it very weird that it has raw socket operations there, as servers should just drop those messages as they won't be well formed.

I do not follow you here, what "raw socket operations" you mean?

dantti, 2025-04-28:
I mean this https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/websocket_client/src/main.c#L160 this is just writing the lorem ipsum bytes into the socket, the server won't be able to parse as it has no OPCODE, mask... so I was really confused why that sample was like that.

jukkar, 2025-04-28:
The `websocket_client` sample demostrates two APIs that zephyr provides for applications, the actual websocket API and also BSD socket API. With websocket API one can set various bits and pieces of websocket protocol. With BSD socket API, one can simply send data, the actual websocket API is run under the BSD socket and it does the encapsulation.

jukkar, 2025-04-28:
Please provide more information how to replicate this, I tried the `websocket_client` sample and I am not seeing any issues. So this is probably related to the server implementation, what server are you using here?
Also it would be useful if you could provide pcap file for the transferred data.

dantti, 2025-04-28:
[websocket_zephyr.pcapng.gz](https://github.com/user-attachments/files/19948014/websocket_zephyr.pcapng.gz)
It's my own websocket server implementation (cutelyst), you can look at the attached pcanp that packet 11 has many websocket messages, and I can't get any of them using websocket_recv_msg()

rlubos, 2025-04-29:
@dantti Can you check https://github.com/zephyrproject-rtos/zephyr/pull/89262 if it helps with the issue? I was not able to reproduce this behavior on my side, but I think I figured out where the problem could be. If the HTTP client part read out more data from the socket than it parsed (i.e. the websocket part), it would never be processed and simply lost. Hopefully the PR will fix that.

PR #89262 (2025-05-05): net: websocket: Make use of any leftover data after HTTP processing 
In case websocket data was piggybacked in the same packet as HTTP Switching Protocols reply, the websocket client would ignore that data, because the HTTP client library would read them already out from the socket. This PR attempts to fix this.

Fixes #89193

~~Marking as DNM for now until the issue is confirmed as resolved by the reporter~~
closed 2025-05-05 by kartben