zephyrproject-rtos/zephyr #89198: lib: uuid: libraries.uuid.base fails

katgiadla, 2025-04-28:
**Describe the bug**
The test `tests/lib/uuid/libraries.uuid.base` fails.

Observed for
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk@0.14.0/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -s libraries.uuid.base -p nrf5340dk/nrf5340/cpuapp/ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting nRF Connect SDK v3.0.99-d660578cb9d5 ***
*** Using Zephyr OS v4.0.99-128319d82200 ***
Running TESTSUITE clock_control
===================================================================
START - test_async_on
I: setup done

    Assertion failed at zephyr/tests/drivers/clock_control/clock_control_api/src/test_clock_control.c:210: test_async_on_instance: (executed is false)
clock@5000: Expected flag to be true
 FAIL - test_async_on in 0.024 seconds
===================================================================
START - test_async_on_stopped
I: setup done
I: setup done
 PASS - test_async_on_stopped in 0.727 seconds
===================================================================
START - test_double_start
I: setup done
I: setup done
 PASS - test_double_start in 0.458 seconds
===================================================================
START - test_double_stop
I: setup done
I: setup done
 PASS - test_double_stop in 0.455 seconds
===================================================================
START - test_on_off_status
I: setup done
I: setup done
 PASS - test_on_off_status in 0.709 seconds
===================================================================
TESTSUITE clock_control failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  80.00% [clock_control]: pass = 4, fail = 1, skip = 0, total = 5 duration = 2.373 seconds
 - FAIL - [clock_control.test_async_on] duration = 0.024 seconds
 - PASS - [clock_control.test_async_on_stopped] duration = 0.727 seconds
 - PASS - [clock_control.test_double_start] duration = 0.458 seconds
 - PASS - [clock_control.test_double_stop] duration = 0.455 seconds
 - PASS - [clock_control.test_on_off_status] duration = 0.709 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 863e5b664485f4023b69e2a303a610db
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: `v4.1.0-2843-g9c9c08`

**Additional information:**
Probably introducedy by commit dfd0dd04c1857d038107b78b325ae992016a66df

nordic-piks, 2025-05-08:
Possible fixed by https://github.com/zephyrproject-rtos/zephyr/commit/9bbccf39cb715201e0fcaba20debbd4638c22af5

carlescufi, 2025-05-08:
@dantti would you mind checking this failure?

dantti, 2025-05-08:
I can try next week, would testing on an ESP32 work? that's the only thing I have, still it would be really odd if my change break these tests, @katgiadla did you git bisect?

carlescufi, 2025-05-08:
> I can try next week, would testing on an ESP32 work?

Sure, that works. Hopefully we can repro this with another SoC. If not I will look at it myself, since I obviously have the board (I work for Nordic).

katgiadla, 2025-05-09:
> @katgiadla did you git bisect?

yes, I have done git bisect

carlescufi, 2025-05-12:
@tomi-font the [bisected SHA](https://github.com/zephyrproject-rtos/zephyr/commit/dfd0dd04c1857d038107b78b325ae992016a66df) points to a TF-M update, could this be causing the failure?

tomi-font, 2025-05-12:
> [@tomi-font](https://github.com/tomi-font) the [bisected SHA](https://github.com/zephyrproject-rtos/zephyr/commit/dfd0dd04c1857d038107b78b325ae992016a66df) points to a TF-M update, could this be causing the failure?

Interesting. Not impossible, though I'm wondering how could TF-M influence the UUID library. I can have a look at some point.

tomi-font, 2025-05-14:
I looked into this, and indeed the test starts failing after the TF-M update. The TF-M update updates Mbed TLS to 3.6.3 in TF-M, which brings some changes to the configuration of the MD module in Mbed TLS: https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/130/files#diff-f099753872f927fda4221b05411176ec1197002d8f2d4d58399de5ee02edb02bR54-R169

The failing test tests `uuid_generate_v5()`, which needs to use SHA-1.
With the changes coming from Mbed TLS it results in MD calling `psa_hash_*()` to perform the operations, which means the calls now go to TF-M.
But in TF-M SHA-1 is disabled in all the profiles, and I don't think we currently have a way to pass single configurations to TF-M (`PSA_WANT_ALG_SHA_1`), so it looks like there's currently no way to enable SHA-1 in TF-M.
Also, `CONFIG_MBEDTLS_PSA_CRYPTO_C` cannot be enabled when TF-M is enabled (and this would really just be a workaround).
Looks like the only, proper way to solve this would be to allow passing PSA Crypto configs to TF-M. (Or modifying the behavior that was just changed in Mbed TLS).

@valeriosetti, as it's your commit which introduced the behavior change in Mbed TLS and I think you have been in talks of improving the Zephyr => TF-M configuration, maybe there's something you could do here? Or if you have other solution suggestions?

In the meantime this failure is "expected".

valeriosetti, 2025-05-22:
> @valeriosetti, as it's your commit which introduced the behavior change in Mbed TLS and I think you have been in talks of improving the Zephyr => TF-M configuration, maybe there's something you could do here? Or if you have other solution suggestions?

Sorry for the delay, but I just looked at this now.
Yes, the problem is that Mbed TLS' MD module can now dispatch to PSA also when `MBEDTLS_PSA_CRYPTO_CLIENT` is set (which is in case of a TF-M build) and that in TF-M there is no support of SHA-1. One solution is surely to pass `PSA_WANT_xxx` from Zephyr down to TF-M at build time. This would be really cool and I would really like to look into that, but it will take some time unfortunately...

Side note: UUID library should not use `mbedtls_md_xxx()` functions, but it should rely directly on PSA Crypto. Not that this will solve the problem (again, no SHA-1 in TF-M), but it would have made the problem easier to identify (no need required on MD's internal magic).

valeriosetti, 2025-05-23:
Based on the discussion above I created https://github.com/zephyrproject-rtos/zephyr/pull/90372 to prevent this test from being executed on TF-M platforms.