zephyrproject-rtos/zephyr #89215: West patch lacks documentation

matt-wood-ct, 2025-04-28:
**Is your enhancement proposal related to a problem? Please describe.**
There is no documentation beyond the CLI help for `west patch`, my team have been actively using zephyr for years and none of us noticed this feature even existed until today.  
If we had known about it we would have had a much easier time handling WIP driver fixes, we previously took the whole drivers out of tree to tweak them while PRs were going through.  

IMO This seems like a very good and very important feature, it deserves to be fully explained to users. The linking to PRs seems especially  interesting.

**Describe the solution you'd like**
It would be nice to have a page or section in the west built-in commands documentation.

**Describe alternatives you've considered**
N/A

**Additional context**
Currently the best source on how to use this excellent feature is the discussion in the PR and the yaml schema file.

cfriedt, 2025-05-01:
I wish I understood the link-to-pr feature better. Perhaps the documentation can be updated collaboratively?

cfriedt, 2025-05-09:
I just went through `west patch gh-fetch`. It's ok.

It would be nice to be able to refer to a pull request inside the YAML directly as well, so that `west patch apply` would download, verify, and apply a patch series from a given PR.

PR #89748 (2025-05-14): doc: west: add documentation for west patch
Add documentation for the `west patch` command.

Closes #89215

[Doc Preview](https://builds.zephyrproject.io/zephyr/pr/89748/docs/develop/west/zephyr-cmds.html)
closed 2025-05-14 by kartben