zephyrproject-rtos/zephyr #89217: Add GPIO hardware interrupt sample for Seeed XIAO BLE Sense board

SterlingHofmann, 2025-04-28:
### Description
There is currently no sample in Zephyr specifically demonstrating how to configure and handle GPIO hardware interrupts on the Seeed XIAO BLE Sense board.

While I understand there is already a GPIO interrupt example under samples/basic/button, it still took me a significant amount of time and research to adapt that example for this particular board. Hardware interrupt behavior can vary depending on pin configurations, power management, and hardware quirks.
Because of that, I believe a board-specific sample would be very helpful for future developers using the Seeed XIAO BLE Sense.

When working with this board, I found that setting up hardware interrupts properly required significant research, especially when it came to configuring pins, callbacks, and ensuring reliable triggering.

I propose adding a board-specific sample under `samples/boards/seeed/xiao_ble/` that demonstrates:
- Setting up a GPIO pin with a pull-up resistor
- Configuring a falling edge interrupt specific for the board and desired interrupt type
- Handling interrupts via a callback function

This would serve as an educational resource for developers working with this board, making it easier to get started with GPIO interrupts.

### Motivation
To improve the developer experience by providing a working, minimal example for GPIO hardware interrupts on Seeed XIAO BLE Sense, which currently does not exist in the Zephyr codebase.

### Proposed Solution
Add a new sample application under `samples/boards/seeed/xiao_ble/hardware_interrupt/` including:
- Minimal C source code
- DeviceTree overlay to configure the necessary pins
- `prj.conf` with required kernel options
- A `README.md` file explaining how to build, flash, and test the example

### Additional Context
I have already developed a working version of this example and can submit it along with clear commit messages following Zephyr's guidelines.
**Note:** I also have additional hardware interrupt examples (such as data ready interrupts and inactivity interrupts) that could be submitted in the future, which involve setting specific registers on sensors.
If the community would find these examples valuable, I would be happy to contribute those as well.

SterlingHofmann, 2025-04-28:
I would like to work on this issue. I already have a working implementation ready for review. Please let me know if there are any specific requirements I should follow. Thank you!

henrikbrixandersen, 2025-04-28:
Having board-specific samples for common functionality like GPIO interrupts doesn't scale well (we support 600+ different boards). 

Have you considered if the documentation on the existing sample could be improved in a way, that would have helped your understanding of it?

closed 2026-09-23 by github-actions[bot]