zephyrproject-rtos/zephyr #89218: CONFIG_DCACHE=y causing issues in UART driver on STM32U5 devices

ABESTM, 2025-04-28:
**Describe the bug**

`CONFIG_DCACHE` is now enabled for STM32U5 and other devices, which have ST proprietary DCache.
This is related to PR #88585.
The DCache on these devices affects only external memories (internal SRAM is not cached). This can cause issues in some scenarios.

**To Reproduce**

Example `samples\boards\st\uart\circular_dma` on `nucleo_u575zi_q` doesn't work properly. The driver refuses the async API, since it checks the DMA buffer in `buf_in_nocache` function in `drivers\serial\uart_stm32.c` file.
To compile the example, I'm using:
`west build -p -b nucleo_u575zi_q samples\boards\st\uart\circular_dma`

**Expected behavior**

There are few ways how to handle this:
1. Don't enable `CONFIG_DCACHE` by default or give user the option to disable it. But I'm not sure what is feasible in current system.
2. Modify the `buf_in_nocache` to allow internal SRAM as non-cacheable.
  a. This function should be probably moved outside UART driver
  b. We need to check if there are STM32 devices that have ST DCache affecting internal memories as well

**Impact**

This issue is concerning UART driver and STM32U5, but it might affect all drivers where DMA is involved.
Additional devices that might be affected: STM32L5, STM32H5, STM32WBA.
Related issue #89098 

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain Zephyr SDK
 - Commit SHA: 9de3d6bf643455493dad32eda0c496cb97ee221a

**Additional context**

erwango, 2025-04-28:
@lindblandro ^^

lindblandro, 2025-04-28:
Hmm, that [code](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_stm32.c#L1383) probably shouldn't look at `CONFIG_DCACHE` directly as the actual cache peripheral is not enabled by default. That option only indicates that the cache is supported. Only ICACHE is enabled by default in the SOC initialization code. Or at least it should be like that. I'll check the sample you provided in more detail tomorrow.

lindblandro, 2025-04-29:
Ok so depending on the symbol is apparently equivalent to assuming the cache is enabled. The docs currently say:

```
* :kconfig:option:`CONFIG_DCACHE` / :kconfig:option:`CONFIG_ICACHE`: these
  options must be selected when support for data or instruction cache is
  present and working in zephyr. Note that if these options are disabled,
  caching may still be enabled depending on the hardware defaults.

  All the code paths related to cache control must be conditionally enabled
  depending on these symbols. When the symbol is set the cache is considered
  enabled and used.

  These symbols say nothing about the actual API interface exposed to the user.
  For example a platform using the data cache can enable the
  :kconfig:option:`CONFIG_DCACHE` symbol and use some HAL exported function in
  some platform-specific code to enable and manage the d-cache.
```

The second paragraph makes it clear that `#ifdef CONFIG_(I|D)CACHE` can be used instead of a dedicated API call (new API functions `sys_cache_*_enabled()` could wrap those options to remove a lot of ifdef clutter from the drivers). 

In any case it seems that either DCACHE should be disabled by default to keep existing behavior on U5 or the driver (and/or sample) code modified to make it work with with external DCACHE peripheral. Without touching the drivers the sample could be made to work by marking the buffers as `nocache`, but it will get messy pretty quickly.

There are several similar definitions for `buf_in_nocache()` in e.g. these files:

- drivers/adc/adc_stm32.c (behind `#if defined(CONFIG_ADC_STM32_DMA) && defined(CONFIG_SOC_SERIES_STM32H7X)`)
- drivers/i2c/i2c_ambiq.c (behind `#ifdef CONFIG_DCACHE`)
- drivers/i2c/i2c_ll_stm32_v2.c (behind `#if defined(CONFIG_DCACHE) && defined(CONFIG_I2C_STM32_V2_DMA)`)
- drivers/spi/spi/spi_ll_stm32.c (behind `#ifdef CONFIG_DCACHE`)
- drivers/spi/spi/spi_ambiq_spic.c (behind `#ifdef CONFIG_DCACHE`)
- drivers/serial/uart_stm32.c (behind `#ifdef CONFIG_DCACHE`)

Probably no need to worry about ambiq except for code duplication, but the other STM32 drivers most likely suffer from this same issue.

There is, however, one additional nuance: The U5 has two DCACHE peripherals and DCACHE2 can cache also internal memories. It's non-secure only and intended use is for GPU2D so it's maybe not as relevant for a serial driver. If somebody is using that I think they just need to be more careful with caching anyway (mark buffers correctly and so on) and DCACHE2 can be left out of this issue.

 For now it might be easiest to just disable DCACHE by default for the U5.

erwango, 2025-04-29:
@ajarmouni-st Are you in line ?

lindblandro, 2025-04-29:
> > Are you in line ?
> 
> `buf_in_nocache()` can be moved to a common file at the STM32 SoC level, then we can add a check guarded by `CONFIG_EXTERNAL_CACHE` to ignore buffers between `_image_ram_start` and `_image_ram_end`.

Is the convention to put common SOC code under `soc/st/stm32/common` (e.g. `stm32_cache.{c,h}`? There's also `soc/common` which currently contains only something related to RISC-V. The function doesn't use any STM32 specifics.

ABESTM, 2025-04-29:
Regarding DCACHE2, I even haven't realized have such feature. But it seems to be tightly tied to GPU2D, which controlled by graphics library. I assume, the DCACHE2 should be handled by graphics library as well. So, maybe we could remove the support for now.

I tried to play a bit with the KConfig, but I'm not much experienced and I'm not sure what is the right way to handle.
- In the current implementation the `CONFIG_DCACHE=y`, but I can't overwrite it in `prj.conf` of the example.
- I can't configure `CONFIG_NOCACHE_MEMORY=y` for STM32U5 so `__nocache_noinit` does nothing. I got following warning:
```
warning: NOCACHE_MEMORY (defined at
C:/ZephyrOS/zephyrproject/zephyr/soc/mediatek/mt8xxx/Kconfig.defconfig:43, arch/Kconfig:408) was
assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: (SOC_FAMILY_MTK
|| ARCH_HAS_NOCACHE_MEMORY_SUPPORT) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_NOCACHE_MEMORY and/or look up
NOCACHE_MEMORY in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
```
Maybe this is just issue with `CONFIG_NOCACHE_MEMORY` not being configured for STM32U5. One solution could be to enable DCache for STM32U5 only when the external memory is enabled in device tree.

Regarding the `buf_in_nocache()` implementation:
- I'm not sure if `_image_ram_start` and `_image_ram_end` can be redirected to external memory? If yes this could be an issue.
- We can check the internal RAM by absolute address (0x0 - 0x40000000). This should be common for all STM32, maybe even ARM Cortex-M devices.
- On some devices we can return `false` if the buffer is in DTCM (data tightly coupled memory), since those are not accessible by DMA. This raises question if devices with DTCM, but without data cache should be covered (I think this is only STM32F4).

Regarding the examples, I think those should be modified, since we want to support devices where cache is crucial for performance (STM32H7, STM32N6).

ABESTM, 2025-04-30:
@JarmouniA The ARM_MPU should be set, since Cortex-M33 has the support. However, `ARCH_HAS_NOCACHE_MEMORY_SUPPORT` is only selected when `CPU_HAS_DCACHE` is enabled (which is not the case for Cortex-M33):
https://github.com/zephyrproject-rtos/zephyr/blob/1596cc5667bfa5b72738b1c9ae50959c4e313bdb/arch/arm/core/Kconfig#L17
This setting could be overwritten by in STM32U5 Kconfig, but then `z_arm_mpu_init` initialization code needs to be updated, since cache functions are not present:
https://github.com/zephyrproject-rtos/zephyr/blob/1596cc5667bfa5b72738b1c9ae50959c4e313bdb/arch/arm/core/mpu/arm_mpu.c#L441C1-L456C35
I will try to prepare PR if I have a time.

closed 2025-06-06 by kartben