zephyrproject-rtos/zephyr #89233: doc: Documentation issue in 'boards/shields/arduino_giga_display_shield/doc/index'

dleach02, 2025-04-29:
**Describe the bug**

The instructions to build do not work.

`west build -b arduino_giga_r1_wifi --shield arduino_giga_display_shield samples/subsys/display/lvgl`

- there is no board target for arduino_giga_r1_wifi.
- if you correct the board name to arduino_giga_r1/stm32h747xx/m7 it errors on the shield name.
- if you set the shield name to giga_display_shield, you get a build error

`west build -p -b arduino_giga_r1/stm32h747xx/m7 --shield giga_display_shield samples/subsys/display/lvgl
`

Results with:

```
C:/work/zephyrproject/zephyr/drivers/display/display_stm32_ltdc.c: In function 'stm32_ltdc_init':
C:/work/zephyrproject/zephyr/drivers/display/display_stm32_ltdc.c:422:25: error: 'CONFIG_VIDEO_BUFFER_SMH_ATTRIBUTE' undeclared (first use in this function)
  422 |                         CONFIG_VIDEO_BUFFER_SMH_ATTRIBUTE,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/work/zephyrproject/zephyr/drivers/display/display_stm32_ltdc.c:422:25: note: each undeclared identifier is reported only once for each function it appears in
[136/468] Building C object modules/lvgl/CMakeFiles/module...phyrproject/modules/lib/gui/lvgl/src/core/lv_obj_pos.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Strawberry\c\bin\cmake.EXE' --build 'C:\work\zephyrproject\zephyr\build'

```
C

**Environment**

* Page: `boards/shields/arduino_giga_display_shield/doc/index`
* Version: main
* SHA-1: 9cea552250afd3a807eb021dfa0042576c93239d

dleach02, 2025-04-29:
@iabdalkader ping

iabdalkader, 2025-04-29:
Does this fix the issue? https://github.com/zephyrproject-rtos/zephyr/pull/89246

moosery, 2025-04-29:
I manually applied those changes in #89246 and while it now compiles, the arduino giga display isn't showing anything.  Also, the doc still needs to be updated to the right board name / shield name per @dleach02's comments.

Unfortunately I am too new to using zephyr and I don't know which port the error logs are being written to.  Once I figure that out I can provide more insight.

iabdalkader, 2025-04-29:
> and while it now compiles, the arduino giga display isn't showing anything.

Does any other sample work? Like `samples/drivers/display/` or maybe `samples/subsys/display/cfb`?

moosery, 2025-04-29:
Same thing, compiles but the display isn't showing anything for either of those samples.

iabdalkader, 2025-04-29:
So normally we use this from the Arduino Zephyr core, which does other initialization. One thing that jumps to mind is SMH init. Can you try adding the following to your main.c?

```C
#if defined(CONFIG_SHARED_MULTI_HEAP)
#include <zephyr/kernel.h>
#include <zephyr/devicetree.h>
#include <zephyr/multi_heap/shared_multi_heap.h>

struct memory_region_t {
	uintptr_t dt_addr;
	size_t dt_size;
	const char *dt_name;
};

#define _BUILD_MEM_REGION(node_id)      \
	{ .dt_addr = DT_REG_ADDR(node_id),  \
      .dt_size = DT_REG_SIZE(node_id),  \
      .dt_name = DT_PROP(node_id, zephyr_memory_region) \
    },

int smh_init(void)
{
    int ret = 0;
    ret = shared_multi_heap_pool_init();
    if (ret != 0) {
        return ret;
    }

    const struct memory_region_t regions[] = {
        DT_FOREACH_STATUS_OKAY(zephyr_memory_region, _BUILD_MEM_REGION)
    };

    for (size_t i=0; i<ARRAY_SIZE(regions); i++) {
        if (!strncmp("SDRAM", regions[i].dt_name, 5)) {
            struct shared_multi_heap_region smh_sdram = {
                .addr = regions[i].dt_addr,
                .size = regions[i].dt_size,
                .attr = SMH_REG_ATTR_EXTERNAL,
            };
            ret = shared_multi_heap_add(&smh_sdram, NULL);
            if (ret != 0) {
                return ret;
            }
        }
    }
	return 0;
}

SYS_INIT(smh_init, POST_KERNEL, CONFIG_CLOCK_CONTROL_PWM_INIT_PRIORITY);
#endif
```

moosery, 2025-04-29:
I added your code to the samples/drivers/display and it compiles but fails to link.

c:/users/moosery/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: Undefined initialization levels used.
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\zephyr\gigazephyr\zephyr\build'

moosery, 2025-04-29:
That works!  I get the white screen with a red block in upper left, green block in upper right, blue block in lower left and lower right block is cycling colors!

iabdalkader, 2025-04-29:
Okay, I guess it's working then. I'm not sure where the best place is for this SMH initialization. Perhaps we should disable it by default and enable it in the core CC @pillo79.

moosery, 2025-04-29:
@iabdalkader - are you going to add this to the #89246 request?  Do we always want that SMH initialized for the giga board or only if the shield is attached?  Otherwise it won't work.  Again, too new to zephyr, but the shield won't work without it in my opinion.  Also @danieldegrasse - not sure how priorities rank in zephyr, but again, the shield doesn't work without these changes, and the docs are incorrect.

iabdalkader, 2025-04-29:
>  Do we always want that SMH initialized for the giga board or only if the shield is attached

Only if you use the video/camera at the same time, and want more video buffers, then you have to switch both the camera and the display to use SMH. Ultimately, it's up to the user to configure the app properly, so I'd say we don't enable SMH by default. But I don't have any strong opinions about this, whatever works best for everyone.

moosery, 2025-04-29:
@iabdalkader Not sure I understand.  Basically, I have an arduino giga r1 wifi with and arduino giga display shield.  I cannot get any display samples to work without the #89246 change along with adding your init code to the main ... and using @dleach02's board/shield names.  So I would think it needs to be part of the standard initialization for the board?  I have no camera attached, just the shield and the board.

moosery, 2025-04-29:
I reset everything, re-applied #89246 changes manually, created a board.c in the boards\arduino\giga_r1 folder that contains your code above (completely with ifdef), added the required CMakeLists.txt to add that to the compile, and it works for everything.  It doesn't include that code if I don't put the shield on the build, so it appears this may be what is needed.  I don't know enough about this project, but I think we can add that as another patch to go along with your 89246?  Would it be a different change?  If so, I could make it.

moosery, 2025-04-29:
Also, most samples seem to work with that change.  However, the samples/subsys/display/cfb still doesn't do anything.  Basically the display turns on, but what is shown is old data (from previous screen from the previous run).  So I don't think everything works yet with that shield.

iabdalkader, 2025-04-30:
I've disable SMH by default, the application should enable and initialize it (if needed). PR https://github.com/zephyrproject-rtos/zephyr/pull/89246 should make the samples work out of the box.

moosery, 2025-04-30:
It doesn't though ... not without adding the initialization code to the samples, which isn't "out of the box".  Unless you made some additional change?

PR #89246 (2025-05-06): drivers: display: stm32_ltdc: Add SMH attribute for LTDC buffer.
Add a config option for the LTDC FB SMH attribute, and fix/update the board config.

Fixes #89233.
closed 2025-05-06 by kartben