zephyrproject-rtos/zephyr #89238: The Vulnerabilities Report Email is Unreachable (vulnerabilities@zephyrproject.org)

Shangzewen, 2025-04-29:
**Describe the bug**

Hello, 
I wants to report several recent found vulnerabilities but all my email sent to the listed email (vulnerabilities@zephyrproject.org) got rejected. 
Please let me know who should I contact for the bug report instead.

**Environment**

* Page: `security/reporting`
* Version: main
* SHA-1: da255c3e3dc433268a0c132a77dd935b318bfbbb

de-nordic, 2025-04-29:
Isn't this thing a proper way to report:

[image]

Shangzewen, 2025-04-29:
Dear @de-nordic 
Thanks for the reply, yes I have tired to rise a new issue to report but after selecting it will pointing the the reporting website again which ask to send a email.
However, I am able to find a report button on the Security Page at the github link which solved my problem. Thanks

closed 2025-04-29 by Shangzewen