zephyrproject-rtos/zephyr #89244: esp32c6 IRQ 00 stack usage always 100% test zephyr's sample

CHONG-GOGOGO, 2025-04-29:
**Describe the bug**

1. west build -b esp32c6_devkitc/esp32c6/hpcore -p always samples/subsys/shell/shell_module
2. west flash

3. uart:~$ kernel thread stacks
0x40810f10 shell_uart                       (real size 2048):   unused 1132     usage  916 / 2048 (44 %)
0x40811ae8 sysworkq                         (real size 1024):   unused  832     usage  192 / 1024 (18 %)
0x40810a00 logging                          (real size  768):   unused  576     usage  192 /  768 (25 %)
0x40811808 idle                             (real size  512):   unused  268     usage  244 /  512 (47 %)
0x4080df20 IRQ 00                           (real size 2048):   unused    0     usage 2048 / 2048 (100 %)

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (ubuntu 22.04)
 - Toolchain (Zephyr SDK 0.17.0)
 - zephyr 4.1.99

**Additional context**

sylvioalves, 2025-04-29:
Besides the full IRQ0 memory usage, do you see any other issues? Is it impacting somehow anywhere?

CHONG-GOGOGO, 2025-04-30:
> Besides the full IRQ0 memory usage, do you see any other issues? Is it impacting somehow anywhere?

[00:00:32.253,000] <err> os:
[00:00:32.253,000] <err> os:  mcause: 5, Load access fault
[00:00:32.253,000] <err> os:   mtval: 428f258f
[00:00:32.253,000] <err> os:      a0: 00000040    t0: 4080f9b8
[00:00:32.253,000] <err> os:      a1: 00000000    t1: 01a00000
[00:00:32.253,000] <err> os:      a2: 0000008c    t2: 00000000
[00:00:32.253,000] <err> os:      a3: 00000001    t3: aaaaaaaa
[00:00:32.253,000] <err> os:      a4: 00000001    t4: aaaaaaaa
[00:00:32.253,000] <err> os:      a5: 428f258b    t5: aaaaaaaa
[00:00:32.253,000] <err> os:      a6: ffffffff    t6: aaaaaaaa
[00:00:32.253,000] <err> os:      a7: ffffffff
[00:00:32.253,000] <err> os:      sp: 40814250
[00:00:32.253,000] <err> os:      ra: 4202ad2a
[00:00:32.253,000] <err> os:    mepc: 4202a864
[00:00:32.253,000] <err> os: mstatus: 00001880
[00:00:32.253,000] <err> os:
[00:00:32.253,000] <err> os: call trace:
[00:00:32.253,000] <err> os:       0: sp: 40814250 ra: 4202a864
[00:00:32.253,000] <err> os:       1: sp: 408142b0 ra: 4202ad2a
[00:00:32.253,000] <err> os:       2: sp: 408142d0 ra: 4201cefc
[00:00:32.253,000] <err> os:       3: sp: 408142d4 ra: 420469bc
[00:00:32.253,000] <err> os:       4: sp: 40814320 ra: 4201cf48
[00:00:32.253,000] <err> os:       5: sp: 40814390 ra: 420278b0
[00:00:32.253,000] <err> os:       6: sp: 408143e0 ra: 42023b2e
[00:00:32.253,000] <err> os:       7: sp: 40814460 ra: 42027a0c
[00:00:32.253,000] <err> os:
[00:00:32.253,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:32.253,000] <err> os: Current thread: 0x40836148 (main)
[00:00:32.496,000] <err> os: Halting system

When I add LVGL-related application code following the example routine, the program always crashes unexpectedly during operation. After gradually investigating the cause, I discovered that the issue might be with the code here.

sylvioalves, 2025-05-13:
@CHONG-GOGOGO, I know the root cause for the IRQ0 to be indicating "100%" of usage but I doubt it is the reason for the crash. Anyway, you can increase IRQ stack to test: `CONFIG_ISR_STACK_SIZE=8192`.
Can you provide a sample that crashes or anything so we can check it?

CHONG-GOGOGO, 2025-05-15:
> [@CHONG-GOGOGO](https://github.com/CHONG-GOGOGO), I know the root cause for the IRQ0 to be indicating "100%" of usage but I doubt it is the reason for the crash. Anyway, you can increase IRQ stack [...]

Thank you for your reply. I've tried setting CONFIG_ISR_STACK_SIZE very large, but no matter how large I set it, the interrupt stack still gets almost filled up, with only a few bytes occasionally remaining. Now I've switched to ESP32-S3 to continue my development using similar code, and I haven't encountered any unexpected crashes so far.

sylvioalves, 2025-05-15:
Can you provide a sample code or more info regarding esp32c6 LVGL issue?

CHONG-GOGOGO, 2025-05-15:
I am using the Waveshare ESP32-C6-LCD-1.47 development board and testing screen switching events simulated by the boot button. When the button is pressed for the 6th time, the system crashes. The exact timing of the crash may vary during development

[esp32c6.zip](https://github.com/user-attachments/files/20228020/esp32c6.zip)

> Can you provide a sample code or more info regarding esp32c6 LVGL issue?

sylvioalves, 2025-06-30:
> I am using the Waveshare ESP32-C6-LCD-1.47 development board and testing screen switching events simulated by the boot button. When the button is pressed for the 6th time, the system crashes. The [...]

@CHONG-GOGOGO does `input_report` calls in `knob.c` file trigger any callback?

CHONG-GOGOGO, 2025-07-01:
There won't be any other callbacks, because I haven't removed or hidden any content, the project is complete.