zephyrproject-rtos/zephyr #89251: New module creation: LoRa Basics Modem (transition from LoRaMAC-node)

JordanYates, 2025-04-29:
## Origin

https://github.com/Lora-net/SWL2001

## Purpose

The current backend for LoRa and LoRaWAN, https://github.com/Lora-net/LoRaMac-node (forked as https://github.com/zephyrproject-rtos/loramac-node), has been deprecated by Semtech for some time.

The replacement is https://github.com/Lora-net/SWL2001

The initial integration of this module is present in #89241

## Mode of integration

Module, the same as LoRaMAC-node. The upstream repository is already setup for integration with Zephyr as a module with external CMake and Kconfig logic.

## Maintainership

@JordanYates 
Additional proposals
@martinjaeger 
@mniestroj 

## Dependencies

None

## Revision

https://github.com/Lora-net/SWL2001/commit/9a14f6772c1d6e303bacb2d594c8063bb804b6ee (v4.8.0)

## License

BSD-3-Clause-Clear (https://github.com/Lora-net/SWL2001/blob/master/LICENSE.txt)
Same content as the existing LoRaMAC-node license (https://github.com/Lora-net/LoRaMac-node/blob/master/LICENSE) except for the title.

## Notes

Eventually the existing LoRaMAC-node backend will be deprecated and the module removed, following the deprecation process.

nashif, 2025-05-22:
I suggest to get this new module for TSC approval, create and move zephyr code to the new module, and obsolete/archive the old one.

nashif, 2025-06-17:
https://github.com/zephyrproject-rtos/lora-basics-modem created

closed 2025-06-17 by nashif