zephyrproject-rtos/zephyr #89279: updatehub: expose public API to report deployment status in manual mode

adrienmaillard03, 2025-04-29:
**Is your enhancement proposal related to a problem? Please describe.**
Yes. When using UpdateHub in manual mode, there is currently no public API to report the deployment status (e.g., success, failure, or rollback) to the UpdateHub server. The internal report(enum updatehub_state) function used by updatehub_autohandler() is static and not accessible from user applications.
As a result, manual-mode applications cannot notify the server of a failed update, which can cause the same faulty update to be retried repeatedly, leading to a loop of downgrade/upgrade attempts.

**Describe the solution you'd like**
Expose a public API such as:

```c
int updatehub_report(enum updatehub_state state);
```
This function would allow developers using manual mode to report success or failure of an update explicitly, just like the autohandler does internally

**Describe alternatives you've considered**
-

**Additional context**

This proposal complements a PR (#89265 ) that fixes the error-reporting issue in automatic mode by ensuring updatehub_autohandler() reports a failure before rollback. Manual mode needs a similar capability to be usable in robust update flows.

otavio, 2025-04-29:
Are you willing to contribute this enhancement?

adrienmaillard03, 2025-04-29:
Thank you for your reply!
I’m a student, and this is my first time using Zephyr, so I’m still learning the ins and outs.
I’m not sure how much time this enhancement would take, and I’m not certain I’ll have a lot of time available, but I’d be happy to help with the enhancement if I can. At least I will try.

adrienmaillard03, 2025-04-29:
I have made a proposal in the PR #89265, as it solves the same problem but for manual mode.

nandojve, 2026-09-19:
Hi @adrienmaillard03 ,

The #89265 was merge. Can we close this?