zephyrproject-rtos/zephyr #89281: Support gzip compression

mlohvynenko, 2025-04-29:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

sayooj-aerlync, 2025-05-14:
Are you expecting any existing zip library to be included, or write another minimal version of it? Existing ones may be bulky.

mlohvynenko, 2025-05-15:
Hi @sayooj-aerlync .
This isn't a direct request to add additional compression libraries to Zephyr. Rather, I'm looking for advice from the community on whether there are any existing solutions or recommended approaches. Specifically, I'm interested in insights on the best way to gzip a file from a Zephyr RTOS application.

closed 2026-09-23 by github-actions[bot]