zephyrproject-rtos/zephyr #89287: Loopback packets missing l2 headers

go2sh, 2025-04-29:
**Describe the bug**
While working on #89248, I stumbled on some failing test with my code. One of the test failing was `socket_packet.test_raw_packet_sockets_udp_send`.

From the test code its visible that loopback packets are missing the l2 headers. This leads to receiving only the IP + UDP header, which is wrong for an AF_PACKET + SOCK_RAW + ETH_P_ALL socket.

**To Reproduce**
Run test `socket_packet.test_raw_packet_sockets_udp_send`.

**Expected behavior**
L2 Header is always part of packets received through AF_PACKET + SOCK_RAW + ETH_P_ALL sockets.

**Impact**
Wrong AF_PACKET socket behavior.

**Logs and console output**
None

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Clang
 - Commit SHA: 6661952dd44f2296f549ac3fff031c3e283f7596

**Additional context**
To fix this, loopback, broadcast and multicast should go through l2 sending before routing back to receive process.

jukkar, 2025-04-30:
Lookback network interface is dummy L2 type which means that there is no L2 in it.

dkalowsk, 2025-05-06:
closing issue based upon comment from @jukkar

closed 2025-05-06 by dkalowsk