zephyrproject-rtos/zephyr #89306: Uninitialized parameters in functions

moudi599, 2025-04-30:
Hello, in file ull_adv_sync.c mainly in the following function, as u can see for example lazy parameter is only defined and not initialized, which will take any garbage value, but it must start from zero because it affects on the event_counter, so is this a bug or what?

static void mfy_sync_offset_get(void *param)
{
	struct ll_adv_set *adv = param;
	struct lll_adv_sync *lll_sync;
	struct ll_adv_sync_set *sync;
	struct pdu_adv_sync_info *si;
	uint32_t sync_remainder_us;
	uint32_t aux_remainder_us;
	uint32_t ticks_to_expire;
	uint32_t ticks_current;
	struct pdu_adv *pdu;
	uint32_t remainder;
	uint8_t chm_first;
	uint8_t ticker_id;
	uint16_t lazy;
	uint8_t retry;
	uint8_t id;
...
}

Thank you!

cvinayak, 2025-05-02:
`lazy` is assigned a value as being passed by reference to the function `ticker_next_slot_get_ext()`. Ensure that `CONFIG_BT_TICKER_LAZY_GET=y` in your implementation/sample/application.

moudi599, 2025-05-02:
@cvinayak , so you mean that enabling this config is mandatory but what if it is disabled. Do we really need to enable it? Also when enabling it, it assigns a certain pointer to lazy pointer, meaning that the address of lazy will not be change.

cvinayak, 2025-05-02:
It is mandatory `select` already in Kconfig file.  If you still see errors/warning, please provide details to reproduce the issue you face. (Please use the github issue template when reporting issues, https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md).

moudi599, 2025-05-02:
Thank you, the issue is resolved, but if this config is mandatory why it is not by default enabled.

cvinayak, 2025-05-02:
It is already mandatory as `select` -ed here: https://github.com/zephyrproject-rtos/zephyr/blob/593716d501d3087a30d136ef598264657e79a3ac/subsys/bluetooth/controller/Kconfig.ll_sw_split#L69