zephyrproject-rtos/zephyr #89343: Create guidelines around AI assisgted code changes and code review

nashif, 2025-04-30:
Create a guidelines around AI assissted code changes and code reviews using Copilot in github.

keith-zephyr, 2025-04-30:
One agreement from Process WG - Copilot reviews are suggestions only.  Copilot requested changes are non-blocking and require a human reviewer to make a change request blocking.

As a project - Zephyr needs to establish that using Copilot for reviews is allowed.

@kartben - Does the process WG think copilot is adding noise or providing value?
* summary can be seen as noise or helpful depending on the size of the PR.
* @fabiobaltieri - doesn't like the summary feature
* @nashif - there are other tools that can identify complexity issues better than copilot
* @kartben - copilot is finding real issues
* @nashif - we could use the hide feature when copilot doesn't add anything useful to reduce noise
* @keith-zephyr - copilot is provided value and noise
* @kartben - hopefully it continues to improve
* @nashif - would like to see the project move to always enabling once we agree there is value
* @keith-zephyr - if low-value comments are hidden can we track this to measure signal to noise?
* @kartben - for well formed PRs, copilot will only provide a summar

Reviewer accounts do need to enable copilot individually before the option to add copilot as a reviewer shows up.

Action Items
* Need to review costs
* Continue gathering data to assess value to the project.  @kartben to investigate the costs as Github rolls this out

nashif, 2025-05-07:
about AI assisted reviews, they actually fall under AI generated code, i.e. if the Copilot bot suggests a code change and the suggestion is appled, it is a AI generated code and we need some guidelines around this.

kartben, 2025-05-07:
I think the topic of code provenance (copyrighted code / license incompatibility) is actually orthogonal. As pointed out by @nashif recently, the project needs some code scanning tools to help us look for matches in existing code bases, and flag any potential issue in contributions (and in code that's already in-tree... 🫣).

We've seen many contributions where people just "naively" copy-paste Linux code, or just change SPDX license in copyright headers of files they "borrowed" somewhere else, long before AI was the new cool kid on the block.

nashif, 2025-05-07:
> I think the topic of code provenance (copyrighted code / license incompatibility) is actually orthogonal. As pointed out by [@nashif](https://github.com/nashif) recently, the project needs some code [...]

it does not help that we now have a bot adding more fuel to the fire.

kartben, 2025-05-07:
> it does not help that we now have a bot adding more fuel to the fire.

Obligatory "IANAL" but FWIW all the suggestions I've seen from said bot all fall in the "unoriginal" bucket, I would say.
https://softwarefreedom.org/resources/2007/originality-requirements.html

henrikbrixandersen, 2025-05-07:
An interesting LWN article on how other projects have reacted to GenAI: https://lwn.net/Articles/972331/

franc0is, 2025-05-07:
I agree with @kartben's point that provenance is orthogonal. I also think we should acknowledge that regardless of policy folks will submit PRs with AI generated code in them, whether or not they acknowledge it. We're going to need some way to manage the risk no matter what we decide here.

teburd, 2025-05-21:
I'd propose that even if we do use AI for review, we *do not* use it automatically. That any usage of AI for review should report an estimated carbon and monetary cost and a question about whether the submitter felt the review was worth the cost every time its used to dissuade unreasonable usages and waste.

teburd, 2025-05-21:
This is exactly what I mean by wasteful

https://github.com/zephyrproject-rtos/zephyr/pull/90182#discussion_r2100472161

Notably we're talking about running an AI tool, a highly wasteful tool, to catch what an existing static analyzer would have found had we simply ran them.

decsny, 2025-05-21:
I am not against AI in general so much, but there is one observation I do have to share which I don't know if it has been discussed yet.

One thing I have noticed from having people (and myself) running AI reviewers on my PRs, is that it is creating a lot of unnecessary work for me to go and try to understand a suggestion which is sometimes subtly incomprehensible at worst and flat out didn't make sense at best (yes, subtle is worse). And since sometimes the AI has valid points, it feels like there is an obligation to check all of them which may not be so easy to judge and require deciphering of what it means or squinting at the code to try to prove that there is something there or not. It's like a guilty until proven innocent situation. And I suppose this is the case with human reviewers to, but it seems a lot more rare for human reviewers to mess up so badly yet convincingly. Maybe it's common to have disagreements with human reviewers but that is actually productive because there really is an understanding for one or both people to be had there, whereas this AI does not benefit from being wrong, and it does not feel like as good of a use of time to validate or dispel it's suggestions as it is when coming from a human because it is often hallucinating and nobody is actually benefiting from that time sink.

Also, I am pretty sure I read something lately about a similar situation happening on like bug bounty websites. Apparently maintainers of open source projects are even now banning people who are submitting AI generated bug bounties because they are such detailed and convincing bug reports that people are submitting to try to claim some easy money, but they turn out to be completely hallucinated and wasting a monumental amount of maintainer time. This is obviously the fault of the people who submit the reports without even confirming what the AI said but it seems this general issue of wasting time due to something an AI generated is becoming an industry wide problem.

kartben, 2025-05-21:
> This is exactly what I mean by wasteful
> 
> [#90182 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/90182#discussion_r2100472161)
> 
> Notably we're talking about running an AI tool, a highly wasteful tool, to catch what an existing static analyzer would have found had we simply ran them.

As I think it was stated multiple time, I don't think anyone is looking at replacing good ol' static analysis with AI -- in fact, and to maybe alleviate the "wasteful" aspect a bit, we should probably explicitly tell Copilot/whichever tool we use, to not try and report about issues an SCA tool would find, and focus on everything else. 
The thing is, there are many types of issues that no static code analysis tool would ever spot. See for example: https://github.com/zephyrproject-rtos/zephyr/pull/90182#pullrequestreview-2858997991. It's basically finding issues with the driver not following the datasheet.

Other examples of things that I am not quite sure a static code analysis tool would spot:

- https://github.com/zephyrproject-rtos/zephyr/pull/89108#pullrequestreview-2838075174 (but I guess some would, it's a pretty common "typo" so I'm sure there are rules to flag this)
- https://github.com/zephyrproject-rtos/zephyr/pull/89662#discussion_r2081499197
- https://github.com/zephyrproject-rtos/zephyr/pull/87001#pullrequestreview-2823494033 -- not sure this sneaky static buffer causing potential thread-safety issues would have easily been flagged either.

teburd, 2025-05-22:
> > This is exactly what I mean by wasteful
> > [#90182 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/90182#discussion_r2100472161)
> > Notably we're talking about running an AI tool, a highly wasteful tool, to catch what an existing static analyzer would have found had we simply ran them.
> 
> As I think it was stated multiple time, I don't think anyone is looking at replacing good ol' static analysis with AI -- in fact, and to maybe alleviate the "wasteful" aspect a bit, we should probably explicitly tell Copilot/whichever tool we use, to not try and report about issues an SCA tool would find, and focus on everything else. The thing is, there are many types of issues that no static code analysis tool would ever spot. See for example: [#90182 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/90182#pullrequestreview-2858997991). It's basically finding issues with the driver not following the datasheet.
> 
> Other examples of things that I am not quite sure a static code analysis tool would spot:

Well so C static analysis tools you may very well be right, but I'm going to cheat just a little and point at rustc because it too is a static analyzer and actually... maybe not so shockingly... solves all of these issues without needing to pay anyone extra or burn forests down to train a bot. Its rules are also well understood so unlike heuristic or neural net based rule checkers its much harder to trick though still feasible if you are very very intentionally crafty.

> 
>     * [drivers: syscon: bflb efuses #89108 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/89108#pullrequestreview-2838075174) (but I guess some would, it's a pretty common "typo" so I'm sure there are rules to flag this)

Some tools appear to support flagging this sort of thing for C/C++ https://wiki.sei.cmu.edu/confluence/display/c/EXP46-C.+Do+not+use+a+bitwise+operator+with+a+Boolean-like+operand

rust and rustc fail to compile this sort of thing noting its not a boolean expression [rust playground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2024&gist=4c087a190286f8def179023626782c95)

ultimately this is because C allows for implicitly interpreting integers as booleans, and we will forever be cursed with this in C or C++

> 
>     * [samples: boards: espressif: ulp: Add interrupt sample #89662 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/89662#discussion_r2081499197)

Looking at the code its mutated then printed in the interrupt directly after the mutation, volatile does nothing here really and the suggestion should have been to use atomics for the problem that the AI suggests is there in my opinion. Carnagie Melon seems to agree with me here and notes the tools that check for this.. https://wiki.sei.cmu.edu/confluence/display/c/CON43-C.+Do+not+allow+data+races+in+multithreaded+code

Rust and rustc would have failed to compile this and complained about a mutable global without synchronization or unsafe which atomics provide https://play.rust-lang.org/?version=stable&mode=debug&edition=2024&gist=4c087a190286f8def179023626782c95

> 
>     * [drivers: display: Re-introduce SSD1327 #87001 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/87001#pullrequestreview-2823494033) -- not sure this sneaky static buffer causing potential thread-safety issues would have easily been flagged either.

This is maybe the only one where I'll say a C static analyzer may not have caught it but surely a static variable in a function is going to be a code smell in Zephyr land, though perhaps some of the same rules would apply https://wiki.sei.cmu.edu/confluence/display/c/CON43-C.+Do+not+allow+data+races+in+multithreaded+code

rust and rustc would have caught this as static/global mutables are unsafe without synchronization for exactly the same reason as the above global volatile would fail to compile.

**Summary**

Static tools can solve many more problems than we look for today given the nice tables at https://wiki.sei.cmu.edu/ suggest.

Rust does almost all of this without any additional tooling.

Static analysis is absolutely possible, we just aren't using the right things to do it well.

AI tools can, seemingly, be confidently incorrect and fool us all, just like the rest of us.

jfischer-no, 2025-05-22:
>     * [samples: boards: espressif: ulp: Add interrupt sample #89662 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/89662#discussion_r2081499197)

This is another good example of how human review has failed. LLM did what it was designed to do. Obviously, this is not how volatile should be used; it just doesn't make sense. Who is going to fix it? How many of these are in the tree now?

> 
>     * [drivers: display: Re-introduce SSD1327 #87001 (review)](https://github.com/zephyrproject-rtos/zephyr/pull/87001#pullrequestreview-2823494033) -- not sure this sneaky static buffer causing potential thread-safety issues would have easily been flagged either.

That is simply the author's/reviewer's level of experience/knowledge and does not justify the problems that LLM may cause.

kartben, 2025-05-22:
> That is simply the author's/reviewer's level of experience/knowledge and does not justify the problems that LLM may cause.

This applies to all reviews though, right? Surely if we had an infinite amount of reviewers at hand, all with decades of experience and unlimited knowledge about everything we would have human catching all bugs and code smells and we wouldn't need to look at _any_ tool to help us with reviews and code analysis?

jfischer-no, 2025-05-22:
> > That is simply the author's/reviewer's level of experience/knowledge and does not justify the problems that LLM may cause.
> 
> This applies to all reviews though, right? Surely if we had an infinite amount of reviewers at hand, all with decades of experience and unlimited knowledge about everything we would have humans catching all bugs and code smells and we wouldn't need to look at _any_ tool to help us with reviews and code analysis?

Yes, but as I said "does not justify the problems that LLM may cause.".
Humans are teachable, especially from their mistakes. LLMs are not (and I do not want to train them). Also, people will likely question human/SCA reviews but blindly follow LLM. I remember a discussion in a review where someone just pulled arguments from an AI platform instead of looking at the C standard. These arguments were obviously completely wrong, yet no one questioned them. No one needs "unlimited" knowledge to contribute, we are not building on a fusion reactor (That would be the right project to spend money on instead of AI). Also, the SCA tool report may cause more harm than good (MISRA does not make it better). However, there is usually an analysis trace that a reviewer or author can follow and understand. The copilot output is just "Prisencolinensinainciusol".

fabiobaltieri, 2025-05-22:
Picking from @teburd and @decsny comments, I think that dumping AI generated comments on someone's PR is a bad approach, imagine a new contributor getting a bunch of generated comment and nothing else, should they do it? Is the PR going be AI approved.

If an author wants to use AI tools for authoring code, or if a review wants to use AI tools as a reviewer tool so be it, it's not like the project an enforce that anyway. But then I'd expect the author/reviewer to stay ahead of their tools, in the same way as one is expected to stay ahead of their editor config, git workflow etc.

To me it all comes down to: if I can tell that your code or review comments are generated without you telling me, we are doing things wrong.

keith-zephyr, 2026-03-06:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/104903