zephyrproject-rtos/zephyr #89350: doc: Documentation issue in 'connectivity/bluetooth/bluetooth-le-host'

PicoBoy2017, 2025-04-30:
**Describe the bug**

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `connectivity/bluetooth/bluetooth-le-host`
* Version: main
* SHA-1: bf95ad47be6f98f14c8f654b17da7ef1afde456d

alwa-nordic, 2025-05-02:
There is no use of "CONFIG_BT_SECURITY_L3" anywhere in the documentation as far as I can see. Can you point to a place in the documentation and explain why you think it is inconsistent? Thanks!

jhedberg, 2025-05-02:
The left-over `bt_conn_security()` reference seems like a valid bug though:
```
❯ git grep bt_conn_security
doc/connectivity/bluetooth/bluetooth-le-host.rst:the :c:func:`bt_conn_security` API on a connection object.
doc/releases/release-notes-2.5.rst:  * The deprecated bt_conn_security function has been removed,
```

alwa-nordic, 2025-05-02:
I created a PR now to fix the use of `bt_conn_security`: https://github.com/zephyrproject-rtos/zephyr/pull/89396

closed 2025-07-03 by jhedberg