zephyrproject-rtos/zephyr #89362: net: Problem using SO_BINDTODEVICE for sockets

javad123javad, 2025-05-01:
**Describe the bug**
When I want to bind a UDP socket to the interface in the following code, a meaningless error occurs that is not in the list of `zsock_setsockopt` error codes:
```c
struct ifreq ifr;
memset(&ifr, 0, sizeof(struct ifreq));
size_t ifr_name_len = sizeof(ifr.ifr_name);
strncpy(ifr.ifr_name, interfaceName, ifr_name_len);
ifr.ifr_name[ifr_name_len - 1] = '\0';
setOptStatus = zsock_setsockopt(socketImpl->sock, SOL_SOCKET, SO_BINDTODEVICE, (void*) &ifr, sizeof(struct ifreq));
```c
if (setOptStatus < 0)
        {
                printk("ERROR: zsock_setsockopt Error code: %d--> %s\n", -errno, strerror(errno));
                printk("ifr_name: %s\n", ifr.ifr_name);
            status = SOPC_STATUS_NOK;
        }
```

The reported error is : 

```shell
ERROR: zsock_setsockopt Error code: -2--> No such file or directory
`ifr_name: ethernet`
```
However, the interface `ethernet` does exist and the TCP side is working on it.

**Environment (please complete the following information):**

 - OS: Zephyr 4.1(e.g. Linux, MacOS, Windows)
 - Toolchain : Zephyr SDK,-0.17

jukkar, 2025-05-01:
Couple of questions:
* Do you have `CONFIG_NET_INTERFACE_NAME` enabled?
* If you have net-shell enabled, what is the output of `net iface`

javad123javad, 2025-05-02:
> Couple of questions:
> 
> * Do you have `CONFIG_NET_INTERFACE_NAME` enabled?
> * If you have net-shell enabled, what is the output of `net iface`

Yes,  it is set. 
When I changed the interface name from `ethernet` to `eth0` the error disappeared and I was able to bind the socket to `eth0`.

```c
 strcpy(ifr.ifr_name, "eth0");
        ifr.ifr_name[4] = '\0';

        setOptStatus = zsock_setsockopt(socketImpl->sock, SOL_SOCKET, SO_BINDTODEVICE, (void*) &ifr, sizeof(struct ifreq));

```
The output of `net iface` command:
```shell
uart:~$ net iface
Hostname: zephyr

Interface eth0 (0x80003700) (Ethernet) [1]
===================================
Link addr : 02:04:9F:55:D1:3A
MTU       : 1500
Flags     : AUTO_START,IPv4,IPv6
Device    : ethernet (0x3007f0ec)
Status    : oper=UP, admin=UP, carrier=ON
Ethernet capabilities supported:
	TX checksum offload
	RX checksum offload
	10 Mbits
	100 Mbits
	MAC address filtering
Ethernet PHY device: phy@0 (0x3007f0d4)
Ethernet link speed: 100 Mbits full-duplex
IPv6 unicast addresses (max 3):
	fe80::4:9fff:fe55:d13a autoconf preferred infinite
IPv6 multicast addresses (max 3):
	ff02::1
	ff02::1:ff55:d13a
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 22994
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 2):
	192.0.2.10/255.255.255.0 manual preferred infinite
	192.168.42.21/255.255.255.0 manual preferred infinite
IPv4 multicast addresses (max 8):
	224.0.0.1
IPv4 gateway : 192.168.42.2

Interface dummy0 (0x80003744) (Dummy) [2]
================================
Link addr : 00:00:5E:00:53:FF
MTU       : 576
Flags     : AUTO_START,IPv4,IPv6
Device    : lo (0x3007f0a4)
Status    : oper=UP, admin=UP, carrier=ON
IPv6 unicast addresses (max 3):
	::1 autoconf preferred infinite
	fe80::5eff:fe00:53ff autoconf preferred infinite
IPv6 multicast addresses (max 3):
	ff02::1
	ff02::1:ff00:1
	ff02::1:ff00:53ff
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 37139
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 2):
	127.0.0.1/255.0.0.0 autoconf preferred infinite
IPv4 multicast addresses (max 8):
	224.0.0.1
IPv4 gateway : 0.0.0.0

```

jukkar, 2025-05-02:
> Yes, it is set.
> When I changed the interface name from `ethernet` to `eth0` the error disappeared and I was able to bind the socket to `eth0`.

If `CONFIG_NET_INTERFACE_NAME` is set, then the network interface has a real name which can be set/get at run time time. In your case the name for the Ethernet interface is `eth0`.
If `CONFIG_NET_INTERFACE_NAME` is **not** set, then the device name `ethernet` is used as a fallback instead for a interface name.

jukkar, 2025-05-02:
I noticed from your log that there was an issue with loopback interface naming, it should be set to `lo`.
This is fixed by #89385

javad123javad, 2025-05-02:
> If `CONFIG_NET_INTERFACE_NAME` is set, then the network interface has a real name which can be set/get at run time time. In your case the name for the Ethernet interface is `eth0`. If [...]

I still believe there's an issue. When I assign the network interface name as `eth0`, I can bind a socket to it without any problems. However, when no name is explicitly assigned, it should fall back to the default name, which, as you mentioned, is `ethernet`.

When I query the available interfaces using `net_if_get_by_index()`, it returns the fallback name (`ethernet`). But when I try to bind a socket to that fallback name, it fails. Strangely, binding to `eth0` still works—even though I never assigned that name to the interface.

jukkar, 2025-05-02:
> I still believe there's an issue. When I assign the network interface name as `eth0`, I can bind a socket to it without any problems. However, when no name is explicitly assigned, it should fall back to the default name, which, as you mentioned, is `ethernet`.

No, that is not the case. The device name "ethernet" in this case, is only used with `SO_BINDTODEVICE` if `CONFIG_NET_INTERFACE_NAME` is **not** set. If on the other hand, if `CONFIG_NET_INTERFACE_NAME` is set, then the interface is assigned a default name, and for the first Ethernet interface it is set to `eth0`.

> 
> When I query the available interfaces using `net_if_get_by_index()`, it returns the fallback name (`ethernet`). But when I try to bind a socket to that fallback name, it fails. Strangely, binding to `eth0` still works—even though I never assigned that name to the interface.

As I said, the `ethernet` is the name of the **device**, and not an network interface. It is just used as a fallback if `CONFIG_NET_INTERFACE_NAME` is **not** set. Otherwise it would not be possible to use the `SO_BINDTODEVICE` in a meaningful way. If you have only one network interface, then it is not really needed to enable `CONFIG_NET_INTERFACE_NAME` or use `SO_BINDTODEVICE`. Also loopback device is not very useful normally, and is mostly used in testing in zephyr.

jukkar, 2025-05-02:
The network interface name was introduced in 3.5, some information about it can be found in migration guide (in first bullet point)
https://docs.zephyrproject.org/latest/releases/migration-guide-3.5.html#networking

javad123javad, 2025-05-02:
Thanks, Jukkar, just got it. I will close this debug.

closed 2025-05-02 by javad123javad