zephyrproject-rtos/zephyr #89432: MQTT-SN client issues

mpecar, 2025-05-03:
With NCS 3.0.0 I got the updated MQTT-SN client.

As far as I can tell, it implements the gateway search procedure, but it introduces issue 1. 
Issues 2. and 3. are old, but 3. is also linked to 1.

1. mqtt_sn_transport_udp.c, tp_udp_init()
- The part that deals with broadcasts (IP_ADD_MEMBERSHIP, IP_MULTICAST_TTL) seems to require CONFIG_NET_IPV4_IGMP=y, but selecting CONFIG_MQTT_SN_xxx does not select it.
- The code does not work in my case - ESP32 WiFi AT WiFi board, but I cannot disable it.

2. mqtt_sn_transport_udp.c, get_ip_str()
- should use zsock_inet_ntop, not inet_ntop - introduces dependency for no good reason.

3. mqtt_sn.c, mqtt_sn_client_init()
- IMO this function should fail if call to "transport->init(transport)" fails.

rlubos, 2025-05-06:
2 and 3 seem pretty trivial to fix, feel free to send a PR, I can do it as well if you prefer. Let's focus on 1.

> The code does not work in my case - ESP32 WiFi AT WiFi board

I assume this used the offloaded ESP Wi-Fi driver? IGMP library doesn't really work with offloaded interfaces as it usees lower-level APIs internally to send IGMP messages. Perhaps we could modify the IGMP APIs to skip the actual report sending if the interface is offloaded, could you check if the following patch helps in your case?
```diff
diff --git a/subsys/net/ip/igmp.c b/subsys/net/ip/igmp.c
index c9551962a26..3c3d33057b5 100644
--- a/subsys/net/ip/igmp.c
+++ b/subsys/net/ip/igmp.c
@@ -622,6 +622,10 @@ int net_ipv4_igmp_join(struct net_if *iface, const struct in_addr *addr,
 
        net_if_ipv4_maddr_join(iface, maddr);
 
+       if (net_if_is_offloaded(iface)) {
+               goto out;
+       }
+
 #if defined(CONFIG_NET_IPV4_IGMPV3)
        ret = igmpv3_send_generic(iface, maddr);
 #else
@@ -640,6 +644,7 @@ int net_ipv4_igmp_join(struct net_if *iface, const struct in_addr *addr,
        }
 #endif
 
+out:
        net_if_mcast_monitor(iface, &maddr->address, true);
 
        net_mgmt_event_notify_with_info(NET_EVENT_IPV4_MCAST_JOIN, iface, &maddr->address.in_addr,
@@ -658,6 +663,10 @@ int net_ipv4_igmp_leave(struct net_if *iface, const struct in_addr *addr)
                return -ENOENT;
        }
 
+       if (net_if_is_offloaded(iface)) {
+               goto out;
+       }
+
 #if defined(CONFIG_NET_IPV4_IGMPV3)
        maddr->record_type = IGMPV3_CHANGE_TO_INCLUDE_MODE;
        maddr->sources_len = 0;
@@ -670,6 +679,7 @@ int net_ipv4_igmp_leave(struct net_if *iface, const struct in_addr *addr)
                return ret;
        }
 
+out:
        if (!net_if_ipv4_maddr_rm(iface, addr)) {
                return -EINVAL;
        }
```

mpecar, 2025-05-06:
I can confirm that the above patch works if I manually enable CONFIG_NET_IPV4_IGMP.

Please also implement 2. and 3., 2. is minor and 3. is connected to 1. - without it, one does not even notice that something in transport init failed.

rlubos, 2025-05-07:
Thanks for confirming, I'll wrap everything up then and send a PR.

mpecar, 2025-05-07:
Ok, thanks. 

I need to say this one more time, just to make sure you understand what I wanted to warn you about:
Without enabling CONFIG_NET_IPV4_IGMP and with 3. implemented, **mqtt_sn_client_init() will always fail with errno = 88.**

rlubos, 2025-05-07:
@mpecar Please take a look: https://github.com/zephyrproject-rtos/zephyr/pull/89606

PR #89606 (2025-05-09): net: mqtt_sn: Fix several minor issues
A set of minor fixes for the MQTT SN library and dependencies. For details, please check commit messages.

Fixes #89432