zephyrproject-rtos/zephyr #89434: west flash: fails to program Arduino GIGA board when run on a windows install

KurtE, 2025-05-03:
**Describe the bug**
I am trying build some of the zephyr examples and run them on an Arduino GIGA.  In particular some of the
samples within samples/subsystem/input. 

The west build completes:

```
(.venv) D:\zephyrproject\zephyr\samples\subsys\input\input_dump>west build -p -b  arduino_giga_r1//m7 --shield giga_display_shield
-- west build: making build dir D:\zephyrproject\zephyr\samples\subsys\input\input_dump\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/zephyrproject/zephyr/samples/subsys/input/input_dump
-- CMake version: 4.0.1
-- Found Python3: D:/Users/kurte/zephyrproject/.venv/Scripts/python.exe (found suitable version "3.11.9", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: D:/zephyrproject/zephyr/.cache
-- Zephyr version: 4.1.99 (D:/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: arduino_giga_r1, qualifiers: stm32h747xx/m7
-- Shield(s): giga_display_shield
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (D:/Users/kurte/Downloads/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (D:/Users/kurte/Downloads/zephyr-sdk-0.16.8)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: D:/Users/kurte/zephyrproject/zephyr/boards/arduino/giga_r1/arduino_giga_r1_stm32h747xx_m7.dts                                    -- Found devicetree overlay: D:/zephyrproject/zephyr/boards/shields/arduino_giga_display_shield/giga_display_shield.overlay                          -- Found devicetree overlay: D:/zephyrproject/zephyr/boards/shields/arduino_giga_display_shield/boards/arduino_giga_r1_m7.overlay                    -- Generated zephyr.dts: D:/zephyrproject/zephyr/samples/subsys/input/input_dump/build/zephyr/zephyr.dts                                             -- Generated pickled edt: D:/zephyrproject/zephyr/samples/subsys/input/input_dump/build/zephyr/edt.pickle                                            -- Generated devicetree_generated.h: D:/zephyrproject/zephyr/samples/subsys/input/input_dump/build/zephyr/include/generated/zephyr/devicetree_generated.h                                                                                                                                                 -- Including generated dts.cmake file: D:/zephyrproject/zephyr/samples/subsys/input/input_dump/build/zephyr/dts.cmake                                D:/zephyrproject/zephyr/samples/subsys/input/input_dump/build/zephyr/zephyr.dts:1330.25-1359.5: Warning (spi_bus_bridge): /soc/spi@52005000: incorrect #size-cells for SPI bus                                                                                                                            <stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'                                                                                                                                                                                                                                     warning: DISPLAY_INIT_PRIORITY (defined at drivers/display/Kconfig:13) was assigned the value '87'                                                   but got the value ''. Check these unsatisfied dependencies: DISPLAY (=n). See                                                                        http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_DISPLAY_INIT_PRIORITY and/or look up                                                        DISPLAY_INIT_PRIORITY in the menuconfig/guiconfig interface. The Application Development Primer,                                                     Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be                                                  helpful too.                                                                                                                                                                                                                                                                                                                                                                                                                                                   warning: STM32_LTDC_FB_USE_SHARED_MULTI_HEAP (defined at drivers/display/Kconfig.stm32_ltdc:56) was                                                  assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: STM32_LTDC (=n),                                                 DISPLAY (=n). See                                                                                                                                    http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_STM32_LTDC_FB_USE_SHARED_MULTI_HEAP and/or                                                  look up STM32_LTDC_FB_USE_SHARED_MULTI_HEAP in the menuconfig/guiconfig interface. The Application                                                   Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of                                                  the manual might be helpful too.                                                                                                                                                                                                                                                                                                                                                                                                                               warning: STM32_LTDC_DISABLE_FMC_BANK1 (defined at drivers/display/Kconfig.stm32_ltdc:59) was                                                         assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: STM32_LTDC (=n),                                                 DISPLAY (=n). See                                                                                                                                    http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_STM32_LTDC_DISABLE_FMC_BANK1 and/or look up                                                 STM32_LTDC_DISABLE_FMC_BANK1 in the menuconfig/guiconfig interface. The Application Development                                                      Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual                                                   might be helpful too.                                                                                                                                                                                                                                                                                                                                                                                                                                          warning: The choice symbol STM32_LTDC_RGB565 (defined at drivers/display/Kconfig.stm32_ltdc:38) was                                                  selected (set =y), but no symbol ended up as the choice selection. See                                                                               http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_STM32_LTDC_RGB565 and/or look up                                                            STM32_LTDC_RGB565 in the menuconfig/guiconfig interface. The Application Development Primer, Setting                                                 Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful                                                  too.                                                                                                                                                                                                                                                                                                      Parsing D:/zephyrproject/zephyr/samples/subsys/input/input_dump/Kconfig                                                                              Loaded configuration 'D:/Users/kurte/zephyrproject/zephyr/boards/arduino/giga_r1/arduino_giga_r1_stm32h747xx_m7_defconfig'                           Merged configuration 'D:/zephyrproject/zephyr/samples/subsys/input/input_dump/prj.conf'         [... cut ...]
setup to flash...  (Might be able to get that command to work for these programs by copying the built 
hex file into the place that the Arduino command for burn bootloader is expecting the file... May try)

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain installed using instructions: https://docs.zephyrproject.org/latest/develop/getting_started/index.html
 - zephyr is pretty close to up to data:  currently using my PR branch: https://github.com/zephyrproject-rtos/zephyr/pull/88502
 - sdk: 16.8 I think

**Additional context**

KurtE, 2025-05-03:
Quick note: I was able to use the dfu-util from Arduino 2.x install:

```
(.venv) D:\zephyrproject\zephyr_test_sketches\print_touch_events>"C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1/dfu-util" --device 0x2341:0x0366 -D "build\zephyr\zephyr.bin" -a0 --dfuse-address=0x08040000:leave
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 2341:0366
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash  2MB   "
Downloading element to address = 0x08040000, size = 38280
Erase           [=========================] 100%        38280 bytes
Erase    done.
Download        [=========================] 100%        38280 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

josuah, 2025-05-06:
I am not very familiar with Windows deployments, however:

> `C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1/`

Is this directory in the current [binary search path](https://www.shellhacks.com/windows-cmd-path-variable-add-to-path-echo-path/)?

If adding it fixes the bug, this might be a documentation bug, or unhelpful error message. Suggesting the user to check the presence for the `dfu-util` program in the current `PATH` could address that maybe.

KurtE, 2025-05-06:
> Is this directory in the current [binary search path](https://www.shellhacks.com/windows-cmd-path-variable-add-to-path-echo-path/)?

It turns out is not in the search path on my Windows machine.  Arduino  installs it's over version(s) depending on
what each board/platform says they require.

**It now works**, that I copied a version of it to my user bin file (c:\users\kurte\bin) which is on my search path

What confused me, is that it just worked fine on my Ubuntu machine, but it turns out I already had a copy
of it in my /usr/bin/ directory.  I probably copied or downloaded it there a while ago, to probably flash some other
board.

Not sure if I  just close this out, or somewhere there should be documentation about this?
If not feel free to close it.

josuah, 2025-05-06:
Very glad to hear that!
All Arduino users trying Zephyr might face this same error.

It looks like a bug as the error messages should have been slightly different:

https://github.com/zephyrproject-rtos/zephyr/blob/d26ed348ef21eeb9559fd45447e51686ead0fe14/scripts/west_commands/runners/core.py#L795-L812

P.S.: It seems possible to specify a `--dfu-util=...path\to\dfu-util.exe`:
https://github.com/zephyrproject-rtos/zephyr/blob/d26ed348ef21eeb9559fd45447e51686ead0fe14/scripts/west_commands/runners/dfu.py#L76-L77

josuah, 2025-05-06:
Let me know if the new name for the issue is not fitting!

KurtE, 2025-05-06:
Sounds good... Thanks

pdgendt, 2025-05-14:
Sorry, I don't have experience with this. Maybe @carlescufi can help?

closed 2025-08-03 by KurtE