zephyrproject-rtos/zephyr #89444: NXP ENET driver: timestamp IRQ handler not called

duerrfk, 2025-05-04:
**Describe the bug**

When compiled with gPTP support, the NXP ENET driver does not call the timestamp IRQ handler (ENET_TimeStampIRQHandler()) handling timer interrupt events (counting seconds of ENET PTP clock) and timestamp available events (tx packet timestamp available). Consequently, the event flags are never reset. The system will not respond anymore (does not execute other threads anymore). Moreover, the seconds of the ENET PTP clock are not updated correctly since timer events are not handled. (The timestamp available events are less critical since they are also written to the packet data structure after transmission.) Obviously, this leads to a broken gPTP implementation for NXP ENET since seconds are not tracked correctly.

The problem is easy to fix (I will make pull request): call the existing ENET_TimeStampIRQHandler() from eth_nxp_enet_isr() when timer interrupt events or timestamp available events are set, as follows:

```
static void eth_nxp_enet_isr(const struct device *dev)
{
        const struct nxp_enet_mac_config *config = dev->config;
        struct nxp_enet_mac_data *data = dev->data;
        unsigned int irq_lock_key = irq_lock();

        uint32_t eir = ENET_GetInterruptStatus(data->base);

        ...

        // THIS IS THE FIX
        if ((eir & kENET_TsAvailInterrupt) || (eir & kENET_TsTimerInterrupt)) {
                ENET_TimeStampIRQHandler(ENET_IRQ_HANDLER_ARGS(data->base, &data->enet_handle));
        }
}
```

There is another severe bug in the NXP ENET/PTP driver implementation, which I will address in another issue/pull request.
  
**To Reproduce**

Platform: Teensy 4.1 (same issue applies at least to all NXP i.MX RT106x platforms).

Compile with gPTP:

CONFIG_NET_GPTP=y

Run gPTP sample (samples/net/gptp), and wait (system will stop responding within a few seconds when first timer interrupt from ENET driver is available). 

**Expected behavior**

System should keep responding (executing threads) instead of being stuck. 

Seconds value of PTP clock should be updated with each timer interrupt event (as done in ENET_TimeStampIRQHandler(), which is never invoked currently). 

**Impact**

Showstopper for running gPTP on platforms using NXP ENET, e.g., NXP i.MX RT106x.

**Logs and console output**

Since system stops responding, no further output is sent to log.

**Environment (please complete the following information):**

OS: Linux (Debian);
Target platform: Teensy 4.1 / NXP i.MX RT106x
Toolchain: SDK 0.17.0
Head May 4, 2025 (021d90b)

**Additional context**
n/a

theadib, 2025-05-04:
hi @duerrfk  have you checked maybe similar PR #88214 
I wrote that PR with the background of using PTP 1588 (not gPTP)
That one uses an frdm_k64f board. I do not have a board with i.MX RT106x.
Regards, Adib

duerrfk, 2025-05-04:
While I wanted to submit another pull request for another bug (enet_handle of NXP PTP driver data should be a pointer to the enet_handle of the MAC driver, not a new instance of the data structure), I have seen that this bug has already been fixed concurrently and gPTP now works as expected,. Maybe, the PTP driver now calls the TimeStampIRQHandler, or the PR #88214 mentioned above by theadib already fixed the issue. Therefore, my reported issue and corresponding pull request seem to be obsolete already. At least I cannot reproduce the bug anymore. 

Sorry if this caused confusion. I am happy to see that several people are actively working on and using PTP/gPTP on this platform!